espressif/esp-at #418: Can I use SYSFLASH to write to an app partition?

espressifelse, 2020-05-21:
Here is my secondary partition table.

```
AT+SYSFLASH?
+SYSFLASH:"ble_data",64,1,0x21000,0x3000
+SYSFLASH:"server_cert",64,2,0x24000,0x2000
+SYSFLASH:"server_key",64,3,0x26000,0x2000
+SYSFLASH:"server_ca",64,4,0x28000,0x2000
+SYSFLASH:"client_cert",64,5,0x2a000,0x2000
+SYSFLASH:"client_key",64,6,0x2c000,0x2000
+SYSFLASH:"client_ca",64,7,0x2e000,0x2000
+SYSFLASH:"factory_param",64,8,0x30000,0x1000
+SYSFLASH:"wpa2_cert",64,9,0x31000,0x2000
+SYSFLASH:"wpa2_key",64,10,0x33000,0x2000
+SYSFLASH:"wpa2_ca",64,11,0x35000,0x2000
+SYSFLASH:"mqtt_cert",64,12,0x37000,0x2000
+SYSFLASH:"mqtt_key",64,13,0x39000,0x2000
+SYSFLASH:"mqtt_ca",64,14,0x3b000,0x2000
+SYSFLASH:"ota_2",0,18,0x40000,0x30000
+SYSFLASH:"fatfs",1,129,0x70000,0x90000

OK
```

I want to read and write to ota_2. This command doesn't work.

```
AT+SYSFLASH=2,"ota_2",0,4096

ERROR
```

What am I doing wrong?

ustccw, 2020-05-25:
@espressifelse Please take a look: https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Customize_Partitions.md

espressifelse, 2020-05-25:
Yes, this limitation is blocking me.

> Users should not change the "name" and "type" of the user partitions which we defined in the "at_customize.csv". But "offset" and "size" can be changed, if necessary.

closed 2020-05-25 by espressifelse