espressif/esp-at #421: A fatal error occurred: Found 2 segments that could be irom0. Bad ELF file?

PontusO, 2020-05-27:
When trying to build for 1MB ESP8266 i get the following problem:

Generating esp8266.project.ld
LD build/esp-at.elf
esptool.py v2.8
A fatal error occurred: Found 2 segments that could be irom0. Bad ELF file?

Moving back to V2.0.0 I don't see this problem.

ustccw, 2020-06-01:
@PontusO Hi, it looks like you firmware has exceeded the limit.
The maximum size allowed for firmware is 896KB (0xE0000), which defined in https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_1mb/partitions_at.csv

Maybe you can try to disable some AT features you would not use, to reduce firmware size.
- `make menuconfig` -> `Component config` -> `AT` - `mdns/wps/smartconfig/ping/mqtt/etc..`

PontusO, 2020-06-03:
Hi, I tried that but I still have the same problem.  I tried disabling most of the features and still have the same issue.
Although, woudn't the linker already have thrown an error if the image was to big ?

PontusO, 2020-06-04:
I cloned a new tree and now it works right out of the box. Not sure what I did wrong previously but all is good now.  I am closing this.

closed 2020-06-04 by PontusO