espressif/esp-at #422: AT+PING causes watchdog to reset

aheberer, 2020-05-29:
Hi all,

I just checked out and built the current versions of RTOS-SDK and esp-at. I noticed that a ping causes the module to reset. Can anyone else confirm this? Am I doing something wrong?

AT+PING="www.google.de"<CR><LF>
+PING:30<CR><LF>
<CR><LF>
OK<CR><LF>
<CR><LF>
 ets Jan  8 2013,rst cause:4, boot mode:(3,2)<CR><LF>
<CR><LF>
wdt reset<CR><LF>
load 0x40100000, len 7288, room 16 <CR><LF>
tail 8<CR><LF>
chksum 0x91<CR><LF>
load 0x3ffe8408, len 24, room 0 <CR><LF>
tail 8<CR><LF>
chksum 0x18<CR><LF>
load 0x3ffe8420, len 3532, room 0 <CR><LF>
tail 12<CR><LF>
chksum 0x41<CR><LF>
csum 0x41<CR><LF>
<CR><LF>
ready<CR><LF>

Best regards
Andreas

--------------------------
Hardware: 
ESP8266EX_Demo_Board

Software:
AT+GMR<CR><LF>
AT version:2.1.0.0-dev(f598046 - May 26 2020 08:25:03)<CR><LF>
SDK version:v3.2-479-gcf83d22a<CR><LF>
compile time(fe6f262):May 29 2020 09:41:01<CR><LF>
Bin version:2.0.0(WROOM-02)<CR><LF>

SDK: SHA-1: cf83d22a9b237c9e5933058455b725b531e8f86f
AT: SHA-1: fe6f26238d37db746e9eda681af5b7ce4249bbb1

ustccw, 2020-06-01:
@aheberer It looks work well.

[image]

From the version information, maybe you have a wrong commitID of SDK version.
As the [IDF VERSION](https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_default/IDF_VERSION) shows,
SDK has already updated to release/v3.3 and commitID came to `c47938e9`.

you can try with the following steps:
```
git reset --hard && git clean -xdff
git pull
export ESP_AT_PROJECT_PLATFORM=PLATFORM_ESP8266
export ESP_AT_MODULE_NAME=WROOM-02
export ESP_AT_PROJECT_PATH=$(pwd)
./esp-idf/tools/idf.py build
```

aheberer, 2020-06-02:
Hey ustccw,

thank you very much! That helped. 
Now the SDK version is also v3.3-rc1 and the ping works fine.

Best regards
Andreas

closed 2020-06-02 by aheberer