espressif/esp-at #424: AT+BLEHIDINIT=1无法初始化，显示ERROR

aithinker2222, 2020-06-01:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6896
ho 0 tail 12 room 4
load:0x40078000,len:14368
load:0x40080400,len:4248
entry 0x400806e0
[0;32mI (74) boot: Chip Revision: 1[0m
[0;32mI (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (40) boot: ESP-IDF v4.0-327-gfe67bed 2nd stage bootloader[0m
[0;32mI (40) boot: compile time 18:42:37[0m
[0;32mI (40) boot: Enabling RNG early entropy source...[0m
[0;32mI (46) boot: SPI Speed      : 40MHz[0m
[0;32mI (50) boot: SPI Mode       : DIO[0m
[0;32mI (54) boot: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (99) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (114) boot: End of partition table[0m
[0;32mI (118) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (125) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x29680 (169600) map[0m
[0;32mI (194) esp_image: segment 1: paddr=0x001296a8 vaddr=0x3ffbdb60 size=0x03a18 ( 14872) load[0m
[0;32mI (200) esp_image: segment 2: paddr=0x0012d0c8 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (201) esp_image: segment 3: paddr=0x0012d4d0 vaddr=0x40080400 size=0x02b40 ( 11072) load[0m
[0;32mI (215) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x108334 (1082164) map[0m
[0;32mI (602) esp_image: segment 5: paddr=0x00238354 vaddr=0x40082f40 size=0x195cc (103884) load[0m
[0;32mI (646) esp_image: segment 6: paddr=0x00251928 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (663) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (664) boot: Disabling RNG early entropy source...[0m
BLUFI BLE is not connected yet

ready
AT+BLEHIDINIT=1

ERROR
AT+GMR
AT version:2.1.0.0-dev(f598046 - May 26 2020 08:34:01)
SDK version:v4.0-327-gfe67bed
compile time(fe6f262):Jun  1 2020 18:42:25
Bin version:2.0.0(WROOM-32)

OK

[image]

xiewenxiang, 2020-06-04:
我复测了一下，是可以正常初始化的
麻烦提供下你指令的输入顺序呢

aithinker2222, 2020-06-04:
你好
能否提供一下你所测试的固件供我进行测试，我个人感觉是menuconfig里面某一个参数设置不对，还有一个问题   该指令是否跟GATT系列的指令不兼容导致它不可使用

> 我复测了一下，是可以正常
> 初始化的麻烦提供下你指令的输入顺序呢

xiewenxiang, 2020-06-04:
HID和其他的GATT命令是不能同时使用的

aithinker2222, 2020-06-04:
> HID和其他的GATT命令是不能同时使用的

你好
     HID和GATT指令能否在同一个固件下共存  比如说我先使用GATT系列的指令功能，操作完成后通过发送AT+RESTORE恢复出厂设置然后再使用HID系列的指令功能

xiewenxiang, 2020-06-05:
你说的可以

aithinker2222, 2020-06-05:
> 你说的可以

那我再编译固件的时候menuconfig中如何进行设置才能实现我所说的情况

aithinker2222, 2020-06-08:
+STA_DISCONNECTED:"c2:51:e1:99:53:e5"
AT+RESTORE

OK
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6932
load:0x40078000,len:14368
load:0x40080400,len:4248
entry 0x400806e0
[0;32mI (71) boot: Chip Revision: 1[0m
[0;32mI (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (39) boot: ESP-IDF v4.0-327-gfe67bed 2nd stage bootloader[0m
[0;32mI (39) boot: compile time 15:01:20[0m
[0;32mI (49) boot: Enabling RNG early entropy source...[0m
[0;32mI (49) boot: SPI Speed      : 40MHz[0m
[0;32mI (49) boot: SPI Mode       : DIO[0m
[0;32mI (53) boot: SPI Flash Size : 4MB[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (83) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (98) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (113) boot: End of partition table[0m
[0;32mI (118) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (125) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x28480 (164992) map[0m
[0;32mI (192) esp_image: segment 1: paddr=0x001284a8 vaddr=0x3ffbdb60 size=0x03970 ( 14704) load[0m
[0;32mI (198) esp_image: segment 2: paddr=0x0012be20 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (199) esp_image: segment 3: paddr=0x0012c228 vaddr=0x40080400 size=0x03de8 ( 15848) load[0m
[0;32mI (214) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x100344 (1049412) map[0m
[0;32mI (589) esp_image: segment 5: paddr=0x00230364 vaddr=0x400841e8 size=0x18324 ( 99108) load[0m
[0;32mI (630) esp_image: segment 6: paddr=0x00248690 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (648) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (648) boot: Disabling RNG early entropy source...[0m
BLUFI BLE is not connected yet

ready
AT+BLEINIT=2

OK
AT+BLEHIDINIT=0

OK
AT+BLEHIDINIT=1

ERROR
AT+GMR
AT version:2.1.0.0-dev(f598046 - May 26 2020 08:34:01)
SDK version:v4.0-327-gfe67bed
compile time(56018c1):Jun  8 2020 15:01:22
Bin version:2.0.0(WROOM-32)

OK
我这边重新编译固件发送AT+BLEHIDINIT=1还是返回ERROR

xiewenxiang, 2020-06-08:
直接初始化HID，不要调用任何其他的蓝牙命令

aithinker2222, 2020-06-09:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6932
load:0x40078000,len:14368
load:0x40080400,len:4248
entry 0x400806e0
[0;32mI (72) boot: Chip Revision: 1[0m
[0;32mI (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (40) boot: ESP-IDF v4.0-327-gfe67bed 2nd stage bootloader[0m
[0;32mI (40) boot: compile time 08:36:01[0m
[0;32mI (49) boot: Enabling RNG early entropy source...[0m
[0;32mI (49) boot: SPI Speed      : 40MHz[0m
[0;32mI (49) boot: SPI Mode       : DIO[0m
[0;32mI (53) boot: SPI Flash Size : 4MB[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (83) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (98) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (113) boot: End of partition table[0m
[0;32mI (118) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (125) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x35320 (217888) map[0m
[0;32mI (211) esp_image: segment 1: paddr=0x00135348 vaddr=0x3ffbdb60 size=0x03970 ( 14704) load[0m
[0;32mI (217) esp_image: segment 2: paddr=0x00138cc0 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (218) esp_image: segment 3: paddr=0x001390c8 vaddr=0x40080400 size=0x06f48 ( 28488) load[0m
[0;32mI (238) esp_image: segment 4: paddr=0x00140018 vaddr=0x400d0018 size=0x1090f4 (1085684) map[0m
[0;32mI (624) esp_image: segment 5: paddr=0x00249114 vaddr=0x40087348 size=0x151c4 ( 86468) load[0m
[0;32mI (660) esp_image: segment 6: paddr=0x0025e2e0 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (677) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (677) boot: Disabling RNG early entropy source...[0m
BLUFI BLE is not connected ye?
ready
AT+RESTORE

OK
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6932
load:0x40078000,len:14368
load:0x40080400,len:4248
entry 0x400806e0
[0;32mI (71) boot: Chip Revision: 1[0m
[0;32mI (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (39) boot: ESP-IDF v4.0-327-gfe67bed 2nd stage bootloader[0m
[0;32mI (39) boot: compile time 08:36:01[0m
[0;32mI (49) boot: Enabling RNG early entropy source...[0m
[0;32mI (49) boot: SPI Speed      : 40MHz[0m
[0;32mI (49) boot: SPI Mode       : DIO[0m
[0;32mI (53) boot: SPI Flash Size : 4MB[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (83) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (98) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (113) boot: End of partition table[0m
[0;32mI (118) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (125) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x35320 (217888) map[0m
[0;32mI (211) esp_image: segment 1: paddr=0x00135348 vaddr=0x3ffbdb60 size=0x03970 ( 14704) load[0m
[0;32mI (217) esp_image: segment 2: paddr=0x00138cc0 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (218) esp_image: segment 3: paddr=0x001390c8 vaddr=0x40080400 size=0x06f48 ( 28488) load[0m
[0;32mI (238) esp_image: segment 4: paddr=0x00140018 vaddr=0x400d0018 size=0x1090f4 (1085684) map[0m
[0;32mI (624) esp_image: segment 5: paddr=0x00249114 vaddr=0x40087348 size=0x151c4 ( 86468) load[0m
[0;32mI (660) esp_image: segment 6: paddr=0x0025e2e0 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (677) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (677) boot: Disabling RNG early entropy source...[0m
BLUFI BLE is not connected yet

ready
AT+BLEHIDINIT=1

ERROR
AT+GMR
AT version:2.1.0.0-dev(f598046 - May 26 2020 08:34:01)
SDK version:v4.0-327-gfe67bed
compile time(56018c1):Jun  9 2020 08:36:12
Bin version:2.0.0(WROOM-32)

OK
直接初始化HID也是返回ERROR

aithinker2222, 2020-06-10:
今天有最新进展吗

closed 2020-06-10 by igrr

ustccw, 2020-06-10:
@aithinker2222 please update the `esp-at` and test.  
feel free to reopen if not fixed.