espressif/esp-at #426: BLE蓝牙如果知道服务已经开启

woshi520laoxie, 2020-06-02:
1、BLE操作方式有:
      AT+BLEGATTSSRVCRE—GATTS 创建服务
      服务是否创建成功可以使用指令 发现服务进行查看   AT+BLEGATTSSRV?
      AT+BLEGATTSSRV—GATTS 发现服务

2、那么服务的开启指令是  AT+BLEGATTSSRVSTART
      那么怎么知道这个服务已经正常开启了呢？直接发送指令回复的是OK，假设我想知道这个服务是否已经开启，那么是否可以使用某些指令获取开启状态。

woshi520laoxie, 2020-06-02:
1、可以知道服务已经创建，但是不能准确的确定服务已经正常开启。因为创建服务之后，查询服务、查询特征值都是显示正常数据的。

xiewenxiang, 2020-06-04:
只是返回时OK，那么就可以认为已经开启了

woshi520laoxie, 2020-06-04:
那么这个可靠度高吗？1、还有因为我目前遇到了一个问题
&nbsp; &nbsp; &nbsp; 使用的是ble，使用经典蓝牙bt的时候却效果好很多。
&nbsp; &nbsp; &nbsp;手机可以搜索到蓝牙，但是有些时候非常难连接，好连接的时候也很好连接。
&nbsp; &nbsp; &nbsp;问题在于很难连接的时候，用户体验非常的差。怎么都连接不上，我想了解一下，有没有哪些指令可以影响到蓝牙的连接，
&nbsp; &nbsp; &nbsp;比如蓝牙可以搜索到广播，我这个时候发送某些指令可以使得蓝牙连接不上。复位除外。
&nbsp; &nbsp; &nbsp;-》我怀疑是我的操作上有误。

xiewenxiang, 2020-06-04:
上面你提到，发送某些指令可以使得蓝牙连接不上，这是指的哪些指令呢

woshi520laoxie, 2020-06-04:
是这样的，因为某些时候还能搜索到蓝牙的广播，但是却无法连接，我猜测是不是我发送了某些指令导致的，
&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;比如说在蓝牙发送广播的时候，有些指令是不能发送的，发送会导致蓝牙无法建立连接，我不确定是否有这些指令。所以提出一个疑问。

问:是否在蓝牙发送广播的时候不能发送某些指令？

xiewenxiang, 2020-06-04:
不会有这些影响的，除非把广播停了

woshi520laoxie, 2020-06-04:
目前，连接都还很顺利，只能等连接不顺利的时候截取这个数据了，不然数据太多都是成功的数据，就没有价值了。

xiewenxiang, 2020-08-10:
如果有新的问题，可以随时打开或者重新创建issues

closed 2020-08-10 by xiewenxiang