espressif/esp-at #427: AT MQTT 连接腾讯云失败

thisisausername-rgb, 2020-06-05:
AT+MQTT连接腾讯云返回+MQTTDISCONNECTED:0
**参照**
```
AT+CWMODE=1
AT+CWJAP="ssid","password"
AT+CIPSNTPCFG=1,8,"ntp1.aliyun.com"
AT+CIPSNTPTIME?

AT+MQTTUSERCFG=0,1,"6H55XGVIWXdev001","6H55XGVIWXdev001;21010406;cCrBd;9223372036854775807","b007455a219557822dff54623830733363e49dce;hmacsha1",0,0,""

AT+MQTTCONN=0,"6H55XGVIWX.iotcloud.tencentdevices.com",1883,1

AT+MQTTSUB=0,"6H55XGVIWX/dev001/control",1
AT+MQTTPUB=0,"6H55XGVIWX/dev001/event","Hello\, this is esp-at!",1,0
```
[https://blog.csdn.net/espressif/article/details/104714464/](url)
**我的log**
```
» AT+SYSLOG=1
« AT+SYSLOG=1

OK
» AT+CIPSNTPCFG=1,8,"ntp1.aliyun.com"
« AT+CIPSNTPCFG=1,8,"ntp1.aliyun.com"

OK
» AT+CIPSNTPTIME?
« AT+CIPSNTPTIME?
+CIPSNTPTIME:Fri Jun  5 11:29:50 2020
OK
» AT+MQTTUSERCFG=0,1,"23SI6PNSS9esp32","23SI6PNSS9esp32","hmacsha256",0,0,""
« AT+MQTTUSERCFG=0,1,"23SI6PNSS9esp32","23SI6PNSS9esp32","hmacsha256",0,0,""

OK
» AT+MQTTCONN=0,"23SI6PNSS9.iotcloud.tencentdevices.com",1883,1
« AT+MQTTCONN=0,"23SI6PNSS9.iotcloud.tencentdevices.com",1883,1
+MQTTDISCONNECTED:0

ERROR
» AT+MQTTCONN?
« AT+MQTTCONN?
+MQTTCONN:0,2,1,"","","",0

OK
```
**固件来源**
`https://github.com/espressif/esp-at/issues/309`

thisisausername-rgb, 2020-06-06:
问题解决
```
WIFI GOT IP
» AT+MQTTUSERCFG=0,1,"NULL","NULL","NULL",0,0,""
« AT+MQTTUSERCFG=0,1,"NULL","NULL","NULL",0,0,""

OK
» AT+MQTTCLIENTID=0,"23SI6PNSS9esp32"
« AT+MQTTCLIENTID=0,"23SI6PNSS9esp32"

OK
» AT+MQTTUSERNAME=0,"23SI6PNSS9esp32;12010126;3SYGK;1627412020"
« AT+MQTTUSERNAME=0,"23SI6PNSS9esp32;12010126;3SYGK;1627412020"

OK
» AT+MQTTPASSWORD=0,"7d986dc59927b552578b2209bdff035e6f6b834b7abd600e35023069d13517f5;hmacsha256"
« AT+MQTTPASSWORD=0,"7d986dc59927b552578b2209bdff035e6f6b834b7abd600e35023069d13517f5;hmacsha256"

OK
» AT+MQTTCONN=0,"23SI6PNSS9.iotcloud.tencentdevices.com",1883,1
« AT+MQTTCONN=0,"23SI6PNSS9.iotcloud.tencentdevices.com",1883,1
+MQTTCONNECTED:0,1,"23SI6PNSS9.iotcloud.tencentdevices.com","1883","",1

OK
» AT+MQTTSUB=0,"23SI6PNSS9/esp32/control",1
« AT+MQTTSUB=0,"23SI6PNSS9/esp32/control",1

OK
» AT+MQTTPUB=0,"23SI6PNSS9/esp32/event","Hello\, this is esp-at!",1,0
« AT+MQTTPUB=0,"23SI6PNSS9/esp32/event","Hello\, this is esp-at!",1,0

OK
```

closed 2020-06-06 by thisisausername-rgb

edieruby, 2021-09-16:
你好
 需要分开设置才能连接成功? 单独MQTTUSERCFG无法连接成功?

ustccw, 2021-09-17:
@edieruby The maximum length of an AT command is 256 bytes. It should be configured separately if the total length of MQTT username, password, clientid, ..., is too long.

edieruby, 2021-09-17:
@ustccw  need help in https://github.com/espressif/esp-at/issues/566#issue-998191423