espressif/esp-at #428: Flashing with Download Tool vs make flash

joeservo, 2020-06-12:
having trouble flashing with download tool. i followed issue #411 with some progress but using the Download Tool the device is not working correctly. using make flash works fine every time.

this is for an ESP32WROOM

make print_flash_cmd shows:
--flash_mode dio --flash_freq 40m --flash_size detect 
0x10000 ota_data_initial.bin 
0x1000 bootloader/bootloader.bin 
0x20000 at_customize.bin 
0xf000 phy_init_data.bin 
0x100000 esp-at.bin 
0x8000 partitions_at.bin

so i set up Download Tool for the correct files, in order above. does the order matter?

the FLASH_SIZE setting makes no sense and i can not figure out this setting. the output from make flash shows auto-detect 4MB which is not even an option.

[image]

Detected Info does not show the flash vendor.

this tool does not ask you to reset the device in boot mode. is this requried? it seems to flash and "finish" regardless.

output from make flash:
App "esp-at" version: ESP-AT
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x100000)...
esptool.py v2.8-dev
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:c5:93:fc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB

anybody see anything that looks incorrect?

ustccw, 2020-06-13:
@joeservo As the https://github.com/espressif/esp-at/issues/411 description,
```
make print_flash_cmd
make | tail -n 1 > build/download.config && cat build/download.config
make factory_bin
```
the real download file actually is more than ten, I am afraid you screenshot is lack of some flashing parameter. 

` make factory_bin` will print the real download file and address, and combine all the files into `one factory firmware`. you can send this to the factory for massive production.

---

As far as i know,  `Download Tool` on windows will not care about the file order.
What i recommend is that download the 4MB bin, which generated by `make factory_bin`, to 0x0 address by  `Download Tool` on windows

joeservo, 2020-06-13:
Thank you. 

The issue was not related to the tool, but in case it helps others i will post my findings.

The command window that opens with Download Tool was hidden and its showed status and the .....___..... that i was looking for. the order does not matter. i left the flash size at 32M, not sure what it does. also used a programming board i designed myself based on some eval boards.

closed 2020-06-15 by joeservo