espressif/esp-at #429: LIGHT SLEEP mode problem

Honzik321, 2020-06-13:
Hi,

AT+SLEEP=2 (light sleep mode) still doesn't work. Could you please test it?

////////////////////////////////////////////////////////////////////////////////////////
My AT commands:

ready
AT+SLEEPWKCFG=2,4,1

OK

AT+CWJAP="XXXX","XXXXXX","XX:XX:XX:XX:XX:XX",1,0,9
WIFI CONNECTED
WIFI GOT IP

OK

-> about 27mA average current here
AT+SLEEP=2

OK

-> about 10mA average current here
AT+CIPSTATUS

-> 1. NO RESPONSE from UART here
-> 2. aplying HIGH level on PIN4
-> 3. about 27mA average current
AT+CIPSTATUS
Ě�÷`�řń

-> This is response from UART after aplying high level on PIN4 (wake-up)
////////////////////////////////////////////////////////////////////////////////////////

I have also tested different wake-up sources with same result.

Thank you very much to solve this problem. I trying to solve it last month still without success.

John

ustccw, 2020-06-28:
@xcguang Please take a look if you are available

Honzik321, 2020-06-28:
> @xcguang Please take a look if you are available

Hello, I'm available, I have interrupted work on esp32, because of the light-sleep problem with AT commands. Do you have any advice ?

Thank you :-)