espressif/esp-at #431: Iot·Espressif 中API的使用

LiThirteen, 2020-06-18:
需求：可以自定义返回数据的内容，返回recommended_rom_version和recommended_rom_filename

API：http://iot.espressif.cn/v1/device/rom/?is_format_simple=true
返回了简单的数据，该数据中不带filename，无法使用动态的文件名，只能固定，并不方便
{
    "productRoms": [
        {
            "life_cycle": "release",
            "recommended": true,
            "version": "123331"
        }
    ],
    "recommended_rom_version": "1.0.4",
    "status": 200
}

API：http://iot.espressif.cn/v1/device/rom/ 
 返回了所有rom信息，不方便使用，很多不需要的参数
{
    "productRoms": [
        {
            "id": 9323,
            "created": "2020-06-18 10:07:13",
            "updated": "2020-06-18 10:07:13",
            "visibly": 1,
            "product_id": 9132,
            "version": "123331",
            "life_cycle": "release",
            "codename": "新固件测试",
            "changelog": "",
            "recommended": 1,
            "files": [
                {
                    "id": 16030,
                    "created": "2020-06-18 10:07:13",
                    "updated": "2020-06-18 10:07:13",
                    "visibly": true,
                    "rom_id": 9323,
                    "name": "partitions.bin",
                    "path": "478cf1790bc89fe08bd2413e82d86ea2.bin",
                    "size": 3072,
                    "download_count": 0
                }
            ]
        }
    ],
    "recommended_rom_version": "1.0.4",
    "status": 200
}

xcguang, 2020-06-18:
服务器内容无法修改

LiThirteen, 2020-06-18:
@xcguang 好的，感谢您的答复

closed 2020-06-18 by LiThirteen