espressif/esp-at #432: The certificate validity starts in the future on first connection

ubichsel, 2020-06-18:
Dear Espressif team

I make a TLS connection to our server right after start-up and WIFI is connected.
I use client and server certificate and have expiry check enabled. 
The first time I open the connection it fails always with the following message on monitor console:
"
E (13560) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (13560) esp-tls: Failed to verify peer certificate!
I (13561) esp-tls: verification info:   ! The certificate validity starts in the future

E (13570) esp-tls: Failed to open new connection
"
When I open the connection 15 seconds later again it works.
It looks like the ESP doesn't have the actual time when executing the first time the tls connection. Do you have any advice if I can somehow force the ESP to request the current time before making the tls connection? Or how long will it take before time is synced?

Thank you in advance!

ustccw, 2020-06-28:
@ubichsel the log said `The certificate validity starts in the future`.
that NOT means certificate check expiry, BUT the certificate issue date is ahead of system time. 

As far as my guess, maybe esp-at has not initialize the SNTP by `AT+CIPSNTPCFG=...`
you could see the system time by `AT+CIPSNTPTIME?`  

please know.

ubichsel, 2020-07-08:
When I set the time in advance with AT+SYSTIMESTAMP it works. If I send AT+CIPSNTPTIME it doesn't help. I was just wondering where and when ssl library takes current time. But again, when I set AT+SYSTIMESTAMP right after initializing it works

closed 2020-07-08 by ubichsel

ustccw, 2020-07-08:
`AT+CIPSNTPTIME` is a query command, other than a set command.  

`AT+CIPSNTPCFG` can enable SNTP server, and get a unix timestamp from sntp server.  

You can enable mbedtls log at `make menuconfig` to look into the certificate check if interested in it.