espressif/esp-at #433: ESP32-WROOM-32 - AT HTTP get block with null char (0x00)

AdriPapa, 2020-06-19:
Hello,

I have an ESP32-WROOM-32 with AT firmware of the branch *release/v2.0.0.0_esp32* (last commit 3c9e097ca).
```
AT version:1.3.0.0-dev(de4a90c - Jun 14 2019 07:28:10)
SDK version:v3.3-beta1
```
The problem:
When I get a binary file with the next HTTP command, the transmission of the 512B-block cuts the messages to the first 0x00 and ignores the rest.

Command:
`AT+HTTPCLIENT=2,0,"http://test.xx/file.test",,,1`

Example return :
```
+HTTPCLIENT:512,    2B 48 54 54 50 43 4C 49 45 4E 54 3A 35 31 32 2C
CRLF                0D 0A
+HTTPCLIENT:512,    2B 48 54 54 50 43 4C 49 45 4E 54 3A 35 31 32 2C
..CRLF              B5 FA 0D 0A
+HTTPCLIENT:512,    2B 48 54 54 50 43 4C 49 45 4E 54 3A 35 31 32 2C
.1.F..\..K.hO.zq    FF 31 18 46 12 F0 5C F8 14 4B 1B 68 4F F4 7A 71
.F..U..F.+...K.h    18 46 12 F0 55 F8 03 46 01 2B 0F D1 10 4B 1A 68
.K.....             10 4B 07 F1 08 0D 0A 2B 48
```

I guess it's something related to strings and its termination, but I'm not familiarised with the AT firmware.

Thank you very much in advance !

ustccw, 2020-06-23:
@AdriPapa Hi, it seems that has already fixed in the lastest version.  
I tried `AT+HTTPCLIENT=2,0,"http://download.espressif.com/esp_share/customer/test/xaa.bin",,,1`, it works well on my side.

Please `git pull` the latest code and compile again.  
please provide the HTTP GET `URL` if possible when test failed.

AdriPapa, 2020-06-23:
The lastets version works, thank you @ustccw  !

closed 2020-06-23 by AdriPapa