espressif/esp-at #434: Can't connect to Azure Iot Hub with AT-Comm

gipobbe, 2020-06-24:
Hi all,
I'm experiencing a bit of troubles trying to connect an esp 32-s2 to Azure via At commands. (I can't use azure SDK)
Like I read here, [https://blog.csdn.net/espressif/article/details/103736743](url) (with google translation) I did the following:

After connecting ESP to internet and setting the correct time I do the following:

sprintf(Buff, "AT+MQTTUSERCFG=0,2,%s,%s,%s,0,0,%s", "\"NULL\"", "\"NULL\"", "\"NULL\"","\"\"");
sprintf(Buff, "AT+MQTTCLIENTID=0,%s","\"Device_ID\"");
sprintf(Buff, "AT+MQTTUSERNAME=0,%s","\"{iot_hub}.azure-devices.net/Device_ID/?api-version=2018-06-30\"");
sprintf(Buff, "AT+MQTTPASSWORD=0,%s","\"SharedAccessSignature sr={iot_hub}-devices.net%2Fdevices%2FDevice_Name&sig=sig&se=se\"");  (I omitted private info) 

Then when I do AT+MQTTCONN I get error. 
From what I've read from Azure documentation one can access to IoT-Hub through specific username and password (SAS) without x.509 certificate. 
By the way I'm a bit confused about a thing.. (I'm a newbie in this regard)
Since I can't connect through AT-Commands I did some tests with some MQTT client and the result I got is the thing that confuse me. 
If I use MQTTfx to access to my Iot-Hub, I can do it only with the above username and password (SAS).
If I use Mosquitto, I need also the Baltimore certificate to access to the Iot Hub.

From what I understood in espressif blog post, I should be able to access to Iot Hub without certificate if I set the second parameter of AT+MQTTUSERCFG to 2, but in reality I can't. Right now I can't try to flash a certificate to see if I'm able to connect to Azure with it, so I ask you. Do I really need to flash the Baltimore cert to be able to connect to Azure or it's ok to only use Username + SAS password? Any suggestions?
Thank you in andvance.

ustccw, 2020-06-28:
@gipobbe please providing the AT uart0 log(see what command input) and AT uart1 log(see what internal log).

As far as i know, azure has the different register way for device. such as set the certificate requirement, or just username and password, clientid, please check it at azure iot hub.

We tested the connection without certificate verify before.
emmm, if the connection works by MQTTfx, esp-at should be okey in same.

gipobbe, 2020-06-29:
Hi ustccw, thanks for the reply.
Since my Esp32 was connected to a device which my company is developing I took out a Devboard I had here and checked if I was able to connect to Azure. Like you said, my devboard connects flawlessly to Azure, no problem at all with AT commands. Seems like my problem is elsewhere. 
I think we can close the issue.
If someone in the future has the same problem, check here [https://blog.csdn.net/espressif/article/details/103736743](url) (Google translate helps here), this is how to connect an esp with esp-at firmware to Azure. It has to work.

ustccw, 2020-06-30:
glad to hear that!

closed 2020-07-13 by gipobbe