espressif/esp-at #435: AT MQTT指令中的topic长度有没有办法改大一些？

ze9hyr, 2020-06-29:
我现在使用的是esp32的2.0.0.0版的固件， 自己使能了MQTT，但是topic的长度限制在64字节， 无法满足业务需求，请问有没有绕过这个限制的？

ustccw, 2020-06-29:
@ze9hyr Hi, what i would like to suggest is to reduce and optimise the topic length.  

**MQTT** as the lightweight protocol is applied for resource constrained IOT devices,  
Memory and throughput should be considered first.  
Please know.

ze9hyr, 2020-06-29:
> @ze9hyr Hi, what i would like to suggest is to reduce and optimise the topic length.
> 
> **MQTT** as the lightweight protocol is applied for resource constrained IOT devices,
> Memory and throughput should be considered first.
> Please know.

我这边是想连接到阿里云， topic长度上优化的可能性不大，而且阿里的topic限制是128字节。所以想问一下固件上是否有什么方式把这个长度限制改大一些

ustccw, 2020-06-29:
@ze9hyr As far as i know, maximum topic length: 64 bytes at `aliyun`  is enough for product.

[image]

ze9hyr, 2020-06-29:
@ustccw 我们需要使用到物模型， 会使用到的topic包含下图中的一些。
[image]
图中的打码部分有11字节， 而deviceName范围是4-32字节， ${tsl.event.identifer}表示的事件ID长度会由阿里生成，按不同的设备生成的长度不一致， 我这边遇到最长的是26字节， 所以确实会遇到不够的情况。

ustccw, 2020-06-29:
@xcguang Could we consider different MQTT configurations for ESP8266 and ESP32?

closed 2021-06-21 by FayeY