espressif/esp-at #439: ESP32 AT2.0 WiFi自动重连机制

RomeliFeng, 2020-06-30:
你好：
        在使用AT+CWJAP指令连接AP时，bssid为空 reconn=1的时候。模块实际进行重连时使用的是SSID还是BSSID。
实际问题是：如果把设备从第一个AP区域移动到第二个AP区域（两个AP SSID一致 BSSID不同），会不会自动连接第二个AP（第一个AP完全没信号）。或当第一个AP还有微弱信号会连接那个AP。

ustccw, 2020-06-30:
@RomeliFeng As far as i know, `802.11 probe request` packet will be sent at the beginning of WiFi connection,   
and the packet will include SSID item and BSSID item.  
SSID or BSSID item will adopt NULL or 0xFF as the auto fill if user does not assign the valid value for them.
> Normally,  SSID and BSSID should have at least one valid configuration for WiFi connection.

`esp-at` will do auto reconnection if WiFi disconnected, scan all the channel, and should be connected to second AP when first AP disappeared.
> Which AP to be connecting depends on the AP's reply(probe response)

RomeliFeng, 2020-06-30:
谢谢。和我自己测试的结果一致。客户现场的问题看来和客户自己的设备有关系。

closed 2020-06-30 by RomeliFeng