espressif/esp-at #440: ESP32 AT2.0 AT+CWJAP参数问题

RomeliFeng, 2020-06-30:
我是用的时AT v2.0.0.0_esp32 的固件
我阅读了文档 https://github.com/espressif/esp-at/blob/master/docs/en/get-started/ESP_AT_Commands_Set.md#cmd-JAP 3.2节CWJAP指令。
当使用
AT+CWJAP="test","password",,1,1,0
时可以运行。

当使用
AT+CWJAP="test","password",,1,1,0,1
添加了scan_mode参数时返回ERROR

[<scan_mode>]:
0: Do fast scan, scan will end after find SSID match AP, Wi-Fi will connect the first scanned AP.
1: All channel scan, scan will end after scan all the channel, Wi-Fi will connect the AP with the strongest signal scanned.

这个参数是在release 2.0 之后添加的吗？为什么我无法正常调用

ustccw, 2020-06-30:
pelase update the esp-at, and compile it.

RomeliFeng, 2020-06-30:
为了稳定，我需要使用发行版。我担心现有版本引入未知bug。这个参数是在2.0之后添加的吗，因为文档里我没有看到有关版本的描述。也没有2.0版本的pdf版本文档

ustccw, 2020-06-30:
@RomeliFeng please take a look https://github.com/espressif/esp-at/blob/release/v2.0.0.0_esp32/docs/ESP_AT_Commands_Set.md

RomeliFeng, 2020-06-30:
谢谢，这正是我所需要的。

closed 2020-06-30 by RomeliFeng