espressif/esp-at #441: [Question] _DEF and _CUR postfixes

pstolarz-prox, 2020-07-02:
During work on some ESP-AT WiFi driver i"ve found usage of `_DEF` and `_CUR` postfixes (e.g. `AT+CWMODE_CUR` and `AT+CWMODE_DEF`) which seems to be absent in this `esp-at` project. On the other hand the commands are described on some Espressif docs like [this one](https://cdn.sparkfun.com/assets/learn_tutorials/4/0/3/4A-ESP8266__AT_Instruction_Set__EN_v0.30.pdf). `esp-at` provides `AT+SYSSTORE` command which seems to have similar functionality along with specific command (e.g. `AT+CWMODE`).

What is the reason of such discrepancy?

Thanks,
Piotrek

ustccw, 2020-07-03:
@pstolarz-prox yeah, `AT+XXX_DEF` and `AT+XXX_CUR` serial command is deprecated now.
`AT+SYSSTORE` is recommended instead of the `_DEF` and `_CUR` feature.

About the `AT+SYSSTORE` command detail, please refer to [AT+SYSSTORE](https://docs.espressif.com/projects/esp-at/en/latest/get-started/ESP_AT_Commands_Set.html#at-sysstore-config-parameter-store-mode) doc.

> newest command as the onoff button to decide AT command save configuration into flash or not, will optimise internal code.
> At the same time, it is easier to save configuration or not for AT user.

pstolarz-prox, 2020-07-03:
@ustccw thanks for the clarification.
The problem with `AT+SYSSTORE` command is - it's not supported by any official release yet. It's present on `master` development branch only. Are you gonna release a new version of the firmware supporting with this command in the near future?

ustccw, 2020-07-03:
@pstolarz-prox We are conducting AT command testing and system testing internally, there will be no more change in the overall functionality.  
It will be released soon, please know.

closed 2020-07-03 by pstolarz-prox