espressif/esp-at #444: bytes loss in TCP transparent mode

flobreton, 2020-07-03:
Hi,

We are encountering some troubles with the transparent mode of the ESP32-WROOM-32 AT firmware (v2.0). We are trying to use such a module to build an UART<->Wifi bridge through a TCP socket. 

What we do is opening the socket, entering in transparent mode and sending a 450 bytes frame with a periodicity of around 40ms. Everything is OK except that we lose some bytes in a random maner => bytes sent to the ESP are not present in the wifi sent frame ... The lost bytes are contiguous in the frame, but the number of lost bytes isn't constant ... Also it doesn't happen in all frames, the frequency is around one incomplete frame per second.

Also we sniffed our UART link to be sure the loss is not due to our host MCU, but data are correctly sent to the ESP... and not present in the TCP frame sent by the ESP (Wireshark observed).

So my questions are : 
 - How is the transparent mode handled ? 
 - What can cause such a behavior (buffer overflow ?) ? 
 - What are your advices in order to get a reliable transparent link ?

Thank you in advance

ustccw, 2020-07-06:
@flobreton Please try with the latest firmware.

[factory_WROOM-32.zip](https://github.com/espressif/esp-at/files/4876445/factory_WROOM-32.zip)

ustccw, 2020-07-06:
@flobreton have you enabled UART flow-control? how is the baud rate of UART.

flobreton, 2020-07-06:
Baudrate is 460800, and I use CTS/RTS. But I think you're pointing the problem. We had to change the pins fot CTS (IO21) and RTS (IO22). We changed the factory_param_data.csv file but it looks like it is not taken into account by the firmware after rebuild and flashing.

ustccw, 2020-07-07:
how about the firmware performance I provided above.

ustccw, 2020-07-07:
Moreover, you could try to add some log at [uart_task]( https://github.com/espressif/esp-at/blob/master/main/interface/uart/at_uart_task.c#L172) to make sure that if uart drops the data or not.

ldsheng, 2020-07-23:
> @flobreton Please try with the latest firmware.
> 
> [factory_WROOM-32.zip](https://github.com/espressif/esp-at/files/4876445/factory_WROOM-32.zip)

Is this bin the same as the factory_param_WROOM-32.bin ?

flobreton, 2020-07-24:
No change with this new firmware. You can reproduce the problem in a simple way : 
 1) Flash an ESP32-WROOM-32 with the firmware.
 2) Connect to a TCP server in tranparent mode (for tests we use Hercule and it's log in a file functionality). UART config = 921600 bps with CTS / RTS.
 3) Send a large file through the UART (for example 20Mo).
 4) Compare your original file with what you get on your TCP server. They are differents.

ustccw, 2020-08-05:
@flobreton Thanks for your feedback.  

We reproduced it when UART baud rate is high, and will fix it.

ustccw, 2020-08-10:
@flobreton we fixed it at master branch, please update esp-at and test.

ldsheng, 2020-08-11:
I have updated to the newest version of master branch and test it over
SDIO, unfortunately, the problem still there.
My  test is sending data over the SSL@TCP to a TCP server. The speed is
about 1.5M bit/s, the server may close the connection as soon as a data
error occurs, and it does.
I have tested several times, and it happens irregularly.

Best Regards,