espressif/esp-at #445: ESP8266 partial boot-up

ctrlgeek, 2020-07-08:
After I flash my ESP8266 board with latest version of esp-at, the board not able to boot up fully:

`.[0;32mI (44) qio_mode: Enabling default flash chip QIO.[0m
.[0;32mI (48) boot: SPI Speed      : 40MHz.[0m
.[0;32mI (52) boot: SPI Mode       : QOUT.[0m
.[0;32mI (56) boot: SPI Flash Size : 2MB.[0m
.[0;32mI (60) boot: Partition Table:.[0m
.[0;32mI (64) boot: ## Label            Usage          Type ST Offset   Length.[0m
.[0;32mI (71) boot:  0 otadata          OTA data         01 00 00009000 00002000.[0m
.[0;32mI (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000.[0m
.[0;32mI (86) boot:  2 ota_0            OTA app          00 10 00010000 000e0000.[0m
.[0;32mI (93) boot:  3 at_customize     unknown          40 00 000f0000 00020000.[0m
.[0;32mI (101) boot:  4 ota_1            OTA app          00 11 00110000 000e0000.[0m
.[0;32mI (108) boot:  5 nvs              WiFi data        01 02 001f0000 00010000.[0m
.[0;32mI (116) boot: End of partition table.[0m
.[0;32mI (120) boot: No factory image, trying OTA 0.[0m
.[0;32mI (125) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0xa7ef0 (687856) map.[0m
.[0;32mI (365) esp_image: segment 1: paddr=0x000b7f08 vaddr=0x402b7f00 size=0x17848 ( 96328) map.[0m
.[0;32mI (399) esp_image: segment 2: paddr=0x000cf758 vaddr=0x3ffe8000 size=0x00688 (  1672) load.[0m
.[0;32mI (400) esp_image: segment 3: paddr=0x000cfde8 vaddr=0x40100000 size=0x00a50 (  2640) load.[0m
.[0;32mI (406) esp_image: segment 4: paddr=0x000d0840 vaddr=0x40100a50 size=0x05980 ( 22912) load.[0m
.[0;32mI (422) boot: Loaded app from partition at offset 0x10000.[0m
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
.` 
The version number, max tx power and ready were missing.

I had flash the same board with older version of ESP-AT (version 2.0) without any issue.

I am using esptool.py version 2.4.0 and latest version of ESP8266_RTOS_SDK.

Please let me know if I've missed out anything.

Thanks in advance for the help.
 
[sdkconfig_v2_1.txt](https://github.com/espressif/esp-at/files/4889322/sdkconfig_v2_1.txt)
[sdkconfig_v2.0.txt](https://github.com/espressif/esp-at/files/4889329/sdkconfig_v2.0.txt)

ustccw, 2020-07-09:
@ctrlgeek Sorry that i can not get your point.
What does  `the board not able to boot up fully` mean?

Could you please providing your firmware here to check?

ctrlgeek, 2020-07-09:
The boot-up stopped right after the "phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new". It does not shows the version, max tx power and "ready".

The older version of the firmware works fine on same hardware.

The firmware is too large to upload. I got the firmware from this repository two days ago.

ustccw, 2020-07-09:
you can combine a 2MB size bin by the following shell command, zip it and attach here:
```
export ESP_AT_PROJECT_PLATFORM=PLATFORM_ESP8266
export ESP_AT_MODULE_NAME=WROOM-02
export ESP_AT_PROJECT_PATH=$(pwd)
./esp-idf/tools/idf.py build
cp build/flash_project_args build/download.config
make factory_bin
```

ustccw, 2020-07-09:
have you disabled some features at `make menuconfig`?
please do `rm -rf build sdkconfig` and re-compile it.

ctrlgeek, 2020-07-09:
Get a fresh copy from github:
`git clone --recursive https://github.com/espressif/esp-at.git`

Edit the Makefile for ESP8266 board:
_export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266 
export ESP_AT_MODULE_NAME ?= WROOM-02_

`make menuconfig` to change the parameters:
_
- spi speed:  80MHz -> 40MHz
- change serial port
- UART for console o/p: custom -> default:UART 0
_

then `make flash` to download to board.

The board still hang after "phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new"

[factory_WROOM-02.bin.zip](https://github.com/espressif/esp-at/files/4894409/factory_WROOM-02.bin.zip)

ustccw, 2020-07-10:
@ctrlgeek Please do not change `UART for console` to `UART0`, keep `UART1` by default.

gipobbe, 2020-07-14:
> Get a fresh copy from github:
> `git clone --recursive https://github.com/espressif/esp-at.git`
>
> Edit the Makefile for ESP8266 board:
> _export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266 export [...]

You missed `make defconfig` before `make menuconfig` ! If my experience this step is necessary!
Btw, changing UART for console is a bit messy to do and the result is not that good. Working with serial cable is much better than redirect the console to the debug USB (I suppose you're working with a devkit If you're trying to change the UART for console).

Maintraxx, 2020-08-26:
Did you solve the issue? I have exactly the same problem. (with a ESP8285EX)

Maintraxx, 2020-08-27:
My workaround is to disable the following line: 

//    uart_intr_config(esp_at_uart_port, &intr_config);

closed 2021-04-14 by FayeY