espressif/esp-at #446: AT Command parser issue?

gipobbe, 2020-07-08:
Here I'm again,
I think I found an unwanted behaviour about esp-at firmware. 
Basically what I found is that the payload can't contain any comma, so if I want to send a JSON with more than 1 parameter like for example: 
"{"state":{"reported":{"Speed": 2, "Lights":"ON"}}}"
it can't be done since AT+MQTTPUB always return error and the message is not sent to AWS.
If I delete the comma, the message is sent but AWS can't recognize the message as JSON (since comma is missing AWS display the message with 2 parameters as plain text). 
The last alternative I thought to continue working with JSON is to send 1 parameter per MQTT message to AWS. I'm not an IOT expert but I think this is really a sub optimal solution (or is it ok doing this way?).
BTW, I'm working with a firmware written in "C" language so the payload is full of escape characters for the ( -> " must be written as -> \\\" <-). Maybe I'm missing some escape character for the comma? Anyone has suggestions? I tried lot of combination, even using ASCII code but nothing work as I'm expecting.

ustccw, 2020-07-09:
Maybe you can try one of the following:
1. escape by `\`
2. try with `AT+MQTTPUBRAW` command

ctrlgeek, 2020-07-09:
I am having same issue.

This string works fine:
String pub = "AT+MQTTPUB=0,\"/v1.6/devices/new-device\",`\"{sensor:50}\"`,1,0";

But if I insert a set of inverted commas "" within the curly brackets {} as below:
String pub = "AT+MQTTPUB=0,\"/v1.6/devices/new-device\",`\"{\"sensor\":012}\"`,1,0";

the firmware returns error:
_AT+MQTTPUB=0,"/v1.6/devices/new-device","{"sensor":012}",1,0
[at-mqtt]error code:0x6044_

Tried with both AT+MQTTPUB and AT+MQTTPUBRAW

gipobbe, 2020-07-09:
> Maybe you can try one of the following:
> 
> 1. escape by `\`
> 2. try with `AT+MQTTPUBRAW` command

Problem solved with AT+MQTTPUBRAW, I can now finally send a valid JSON with multiple parameter correctly separated by commas. Honestly I always used MQTTPUB but MQTTPUBRAW seems a lot nicer to use. I can get rid of all the escape sequences (at some point there were more escape sequences that real data on my payloads).
Thank you again ustccw for your help. You know your stuff ;)

ustccw, 2020-07-09:
@gipobbe @ctrlgeek escape is necessary to sensitive, related to delimiters in AT command.
- `\\`: escape backslash itself
- `\,`: escape comma which used to separate each parameter
- `\"`: eacape double quotation marks which used to mark string input

ustccw, 2020-07-09:
> I am having same issue.
>
> This string works fine:
> String pub = "AT+MQTTPUB=0,"/v1.6/devices/new-device",`\"{sensor:50}\"`,1,0";
>
> But if I insert a set of inverted commas "" within the curly brackets [...]

@ctrlgeek Please try:
```c
AT+MQTTPUB=0,"/v1.6/devices/new-device","\"{sensor:50}\"",1,0
AT+MQTTPUB=0,"/v1.6/devices/new-device","{\"sensor\":012}",1,0
```

gipobbe, 2020-07-09:
> @gipobbe @ctrlgeek escape is necessary to sensitive, related to delimiters in AT command.
> 
> * `\\`: escape backslash itself
> * `\,`: escape comma which used to separate each parameter
> * `\"`: eacape double quotation marks which used to mark string input

Escape comma didn't work for me earlier with MQTTPUB. But since MQTTPUBRAW works really nice I'm fine. 
@ctrlgeek 
If you're working with C firmware, try this "\"{\\\"state\\\":{\\\"reported\\\":{\\\"S\\\":0}\"" if you want to send a correct JSON.

ustccw, 2020-07-09:
@gipobbe comma should be okey with  
`AT+MQTTPUB=0,"/v1.6/devices/new-device","{\"state\":{\"reported\":{\"Speed\": 2\, \"Lights\":\"ON\"}}}",1,0`

[image]

ctrlgeek, 2020-07-09:
> @ctrlgeek Please try:
>
> ```c
> AT+MQTTPUB=0,"/v1.6/devices/new-device","\"{sensor:50}\"",1,0
> AT+MQTTPUB=0,"/v1.6/devices/new-device","{\"sensor\":012}",1,0
> ```

Tried your suggestion, does not solve the problem.

ustccw, 2020-07-09:
@ctrlgeek I tried, it looks good. please try again. What is the error number?

[image]

ustccw, 2020-07-09:
`AT+MQTTPUB=0,"/v1.6/devices/new-device","\"{\"sensor\":012}\"",1,0` maybe is a completed JSON format in peer.

ctrlgeek, 2020-07-09:
@ustccw @gipobbe 
Thanks for the help.

I am programming in Arduino to talk to an ESP8266 with ESP-AT firmware. My arduino code:
```
SoftwareSerial mySerial(7, 6); // RX, TX
mySerial.begin(19200);
......
String pub = "AT+MQTTPUBRAW=0,\"/v1.6/devices/new-device\",\"{\"sensor\":50}\",1,0";
mySerial.println(pub);
```
I've also tried:
`String pub = "AT+MQTTPUBRAW=0,\"/v1.6/devices/new-device\",{\"sensor\":50},1,0";
`
Both return error `[at-mqtt]error code:0x604B`

ustccw, 2020-07-09:
@ctrlgeek wow, maybe you had confused the parameter of MQTT command `AT+MQTTPUB` and `AT+MQTTPUBRAW`

please know.

ctrlgeek, 2020-07-09:
> @ctrlgeek I tried, it looks good. please try again. What is the error number?
> 
> [image]

```
AT+MQTTPUB=0,"/v1.6/devices/new-device",{"sensor":50},1,0
[at-mqtt]error code:0x6044

ERROR
```
My arduino code:
```
String pub = "AT+MQTTPUB=0,\"/v1.6/devices/new-device\",{\"sensor\":50},1,0";
mySerial.println(pub);
```

ustccw, 2020-07-09:
it is **NOT** `AT+MQTTPUB=0,"/v1.6/devices/new-device",{"sensor":50},1,0`,   
**BUT** `AT+MQTTPUB=0,"/v1.6/devices/new-device","\"{\"sensor\":50}\"",1,0`

ctrlgeek, 2020-07-10:
> it is **NOT** `AT+MQTTPUB=0,"/v1.6/devices/new-device",{"sensor":50},1,0`,
> **BUT** `AT+MQTTPUB=0,"/v1.6/devices/new-device","\"{\"sensor\":50}\"",1,0`

Thanks for the help

my code:
```
String pub = "AT+MQTTPUB=0,\"/v1.6/devices/new-device\",\"\"{\"sensor\":50}\"\",1,0";
mySerial.println(pub);
```

Result:
[image]

My code:
```
String pub = "AT+MQTTPUB=0,\"/v1.6/devices/new-device\",\"{\"sensor\":50}\",1,0";
 mySerial.println(pub);
```

Result:
[image]

ustccw, 2020-07-10:
@ctrlgeek  It looks like a issue about `arduino code`,  cause to the AT wrong command input.

What i would like to suggest is to test `AT+MQTTPUB` command with serial port tool, such as secureCRT, cutecom, minicom, etc

Then find out what the difference between the serial port tool and your mcu.

closed 2020-07-13 by gipobbe