espressif/esp-at #447: HTTPGETSIZE answer add a NULL character after size (and before <CR> <LF> OK <CR> <LF>)

mougz, 2020-07-09:
Hi,

I'm currently using a ESP32-WROOM-32D with AT commands ( built from Github source, latest version with HTTP and MQTT commands)

I'm writing a C function "wrapping" the AT command  **AT+HTTPGETSIZE** and I'm testing it on 4 HTTP resources from our test server. The 4 files have very different size (from 24bytes to 200k bytes) , in order to test the behavior and the rapidity of the command depending on the size of the file. 

I have 2 different behavior depending on the size of the file. See examples below.

A 686 bytes file (screen from oscilloscope and printing of the string buffer parsing the answer)

[image]
[image]
[image]

A 204 792 bytes file 

[image]
[image]
[image]

For the "small" file, there is a' \r\n' (CR LF) just after the size in digits , and before and after the "OK".

But for the large file, there is '\0 'and 1 after the size. I've tested this command several times and the pattern is always '\0' instead of the '\r' (CR) and a random character instead of the '\n' (LF) (it is not necessary '1' )

Not only that this behavior is not really "stable", it is also interfering with the way my firmware is parsing the answer : to be sure that the AT command succeed, I always check for the "OK" answer. But the NULL character is misleading my parser which always read the "+HTTPGETSIZE" part but not the "OK" part, **because a string in C always finished with a NULL char**. I don't understand why this happens only for "large" file but it might be a clue. I did not try to find a "boundary size" delimiting the triggering of the bug.

ustccw, 2020-07-10:
@mougz This is a very good description of the issue!  
AT fixed it internally, will auto-sync to GitHub next week.

Please test the following firmware if you need it urgently. 
[factory_WROOM-32.zip](https://github.com/espressif/esp-at/files/4903114/factory_WROOM-32.zip)

closed 2020-07-13 by igrr

mougz, 2020-07-15:
Thanks for your reactivity !