espressif/esp-at #451: UDP in passive receive mode sums in +IPD the sizes of pending messages.

jandrassy, 2020-07-25:
## Environment

- Development Kit:      Wemos D1 mini 4MB flash
- Module or chip used:  ESP-12
- esp-at version:  v2.1.0.0-rc1_esp8266-4-g51262c8

## Problem Description

UDP in passive receive mode sums in +IPD the sizes of pending messages. This makes it impossible to read the individual messages. 

### Expected Behavior

+IPD in passive mode should report the size only of the next pending UDP message.

### Steps to repropduce

1. setup the AT firmware with AT+CIPRECVMODE=1
2. start UDP listener AT+CIPSTART="UDP","0.0.0.0",2390,2390,2
3. send 3 different UDP messages to the IP of the AT firmware, (best from different IP addresses)
4. read with with AT+CIPRECVDATA the messages with sizes reported by +IPD 

## Debug Logs

```
WIFI CONNECTED
WIFI GOT IP
AT+CIPRECVMODE=1

OK
AT+CIPDINFO=1

OK
AT+CIPSTART="UDP","0.0.0.0",2390,2390,2
local port:2390

change type:2

CONNECT

OK

+IPD,41
AT+CIPRECVDATA=41
to read 41 bytes

+CIPRECVDATA:41,"192.168.1.113",40406,this is a test message from 192.168.1.113
OK

+IPD,41
AT+CIPRECVDATA=41
to read 41 bytes

+CIPRECVDATA:41,"192.168.1.109",46298,this is a test message from 192.168.1.109
OK

+IPD,82
AT+CIPRECVDATA=82
to read 82 bytes

+CIPRECVDATA:82,"192.168.1.109",46007,this is a test message from 192.168.1.113this is a test message from 192.168.1.109
OK
```

jandrassy, 2020-08-02:
I think a simplest fix would be to return only one UDP message even if +CIPRECVDATA requests larger size (because the host doesn't know the size of the next message if +IPD reported a sum of pending messages size)

jandrassy, 2021-07-10:
this is almost fixed in 2.2. 
a +IPD for a message is now reported only after the previous message was read. an AT+CIPRECVDATA  gets the content of one message only. this is OK.

**but if the requested length is smaller then the size of the message, the rest of the message is lost.**. this can happen for a large message if the host can't allocate one large buffer.

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

JAndrassy, 2022-05-27:
so it is same as it was in 2.2. a message must be read at once in passive mode
```
+IPD,4,81

>>Send to /dev/ttyACM0: "AT+CIPRECVLEN?"<<
AT+CIPRECVLEN?
+CIPRECVLEN:-1,-1,-1,-1,81

OK

>>Send to /dev/ttyACM0: "AT+CIPRECVDATA=4,32"<<
AT+CIPRECVDATA=4,32 
+CIPRECVDATA:32,12345678901234567890123456789012
OK

>>Send to /dev/ttyACM0: "AT+CIPRECVLEN?"<<
AT+CIPRECVLEN?
+CIPRECVLEN:-1,-1,-1,-1,0
 
OK
```

ds-c, 2022-07-06:
I confirm the problems reported by JAndrassy. Currently, if multiple UDP messages (datagrams) are received there is no way of separating them from each other. Separating UDP datagrams is a strict requirement for any UDP application, unlike TCP (which is a stream protocol).

The best solution is:
1) +IPD and +CIPRECVLEN in passive mode must report the size only of the next pending UDP message.
2) +CIPRECVDATA must return only one UDP message if +CIPRECVDATA requests a size that is larger or equal to the size of the next pending UDP message.

JAndrassy, 2022-07-07:
@ds-c 1) a 2) are true since version 2.2. for passive mode

ds-c, 2022-07-07:
@JAndrassy, unfortunately my tests show that 1) and 2) are not implemented.

- Module or chip used: ESP-12
- esp-at version: ESP8266-IDF-AT_V2.2.1.0

```
ATE0
> OK

AT+CIPMUX=0
> OK

AT+CIPDINFO=0
> OK

AT+CIPRECVMODE=1
> OK

AT+CIPSTART="UDP","0.0.0.0",2390,2390,2
> CONNECT
> OK
```

Send UDP message "0123" to the module:
`> +IPD,4`

Send UDP message "456" to the module.

Send UDP message "78" to the module.

Check the size of the next pending UDP message:

```
AT+CIPRECVLEN?
> +CIPRECVLEN:9
> OK
```

Expected: the reported size is the size of the next pending message (4)
Actual: the reported size is the size of all pending messages (9)

Receive the next pending message:

```
AT+CIPRECVDATA=4
> +CIPRECVDATA:4,0123
> OK
> +IPD,5
```

Expected: the size reported by IPD is the size of the next pending message (3)
Actual: the size reported by IPD is the size of all pending messages (5)

Receive the next pending message:

```
AT+CIPRECVDATA=100
> +CIPRECVDATA:5,45678
> OK
```

Expected: only the next pending message is read ("456")
Actual: all pending messages are read ("45678")

So it is impossible to check the size of the next pending UDP message and receive messages separately one by one, as **required** by UDP applications.