espressif/esp-at #452: Could AT+CIPRECVMODE be changed to allow separate setting of passive mode for UDP?

jandrassy, 2020-07-25:
Could AT+CIPRECVMODE be changed to allow separate setting of passive mode for UDP?
The passive receive mode now applies to TCP, SSL and UDP. If I want to use the passive receive mode for TCP I am forced to implement passive receive mode for UDP too.

MaJerle, 2020-07-25:
Your software is anyway capable to do manual receive, why would you need different for UDP?

Usually it is arch specific use case.

jandrassy, 2020-07-25:
> Your software is anyway capable to do manual receive, why would you need different for UDP?
> 
> Usually it is arch specific use case.
> 
> Otherwise executive answer is no. Not possible.

The UDP passive received mode works different then TCP passive receive mode. At least now. I don't know if it changes after fixes.
The UDP message now must be read with one AT+CIPRECVDATA with the exact size. If not the complete message is read, the rest is lost. Read for the rest returns ERROR.
If larger length is provided in AT+CIPRECVDATA, then it will read data from next pending message.

closed 2022-01-06 by jandrassy