espressif/esp-at #453: Safety check of "double subscribe" or "double unsubscribe" does not work

mougz, 2020-07-30:
Hardware : ESP32-WROOM-32D
Firmware : AT Firmware built from GitHub source with MQTT and HTTP AT command

Hi,

I'm currently testing MQTT AT commands. I've noticed a little issue with `AT+MQTTSUB` and` AT+MQTTUNSUB` functions.

The documentation says , for `AT+MQTTSUB` : **If the topic has been subscribed before, it will prompt:ALREADY SUBSCRIBE**
And for `AT+MQTTUNSUB` : **If the topic has not been subscribed, then the AT log will prompt NO UNSUBSCRIBE**

However, when I test these safety checks (by calling the same AT command twice ie `AT+MQTTSUB:0,"something/test",0`x2 & `AT+MQTTUNSUB:0,"something/test"`x2) , the AT will always responds  "OK". 
Note that for the `AT+MQTTUNSUB` function , the second call answers " OK" but the answer is way slower (~160ms VS ~25ms for the first call) 

Is that a bug or I'm missing something here ?

Thanks in advance,
Hugo

ustccw, 2020-08-03:
@mougz Hi, prompt information will print by `UART0 port`, other than `UART1 port`.

mougz, 2020-09-10:
Duly noted ! I have discover the UART debug port ^^
I close the issue !

closed 2020-09-10 by mougz