espressif/esp-at #455: uart failed at 2mbits.

jerrytry1, 2020-08-04:
# Environment

- Development Kit:      ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version:  v4.0.1-193-ge7ac221b4
  at firmware version: v2.1.0.0-rc2_esp32-17-g1a32eb3   
- Build System:         [Make|CMake]
- Compiler version : (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Power Supply:         external 3.3V

## Problem Description

It won't work with packet more than 120 bytes if baudrate of uart is 5mbps.
In esp32 uart hardware rx buffer is only 128 bytes.  when you do stress test, you will miss the packet. Eventually it breaks.

I think it is the same issue as below:

**https://www.esp32.com/viewtopic.php?t=6096
https://github.com/espressif/esp-idf/issues/4537**

Please update your at firmware as soon as possible. 

Thanks.
Jerry

costaud, 2020-08-18:
I don't think this is a same issue as to #4537 by the way.

HW flow control is recommended in such high baud applications. 
One approach is to use DMA in UHCI which is under development and will be merged soon.

For now, you can try setting UART_FULL_THRESH_DEFAULT to a smaller value(i.e. 90) and set UART_ISR_IN_IRAM in menuconfig under 5mbps speed.

jerrytry1, 2020-08-18:
True, I am not using HW flow control. not enough pin.   I will wait for DMA in UHCI.
I will try to set UART_FULL_THRESH_DEFAULT to a smaller value as well. 

Also, I tried to update esp-idf firmware to 4.2 with at lib, after that, the program crash at at_interface_init();
4.2 has a important feature modbus.  

Thanks.

ubichsel, 2020-08-24:
I use 5 MBaud and with hardware flow control it works fine (without I sometimes got problems)

Maintraxx, 2020-08-27:
> Also, I tried to update esp-idf firmware to 4.2 with at lib, after that, the program crash at at_interface_init();
> 4.2 has a important feature modbus.

You can disable `uart_enable_swap();` in at_user_task.c Line 379. This solved my problems.

Maintraxx, 2020-08-28:
I have the same issue. The UART Ringbuffer lost data. Even, if i decrease the Baudrate to 115.200!

**My Version**

> ESP8285
> AT version:2.2.0.0-dev(7963ea8 - ESP8266 - Aug 25 2020 07:31:48)
> SDK version:v3.3-2-gb6e861a7
> compile time(cf63f23):Aug 28 2020 11:08:11
> Bin version:2.0.0(Unknown)

I tried to see, where the data is gone and i make an interesting observation.

I printed the data length in the following functions:

- at_port_read_data
- at_port_get_data_length
- uart_task

**Here is, what happend (@115.200 Baud!):**

> 
>   27.019 [ESP8266 ] [TX] DATA[AT+CIPSEND=0,2048]
>   27.025 [ESP8266 ] [RX] DATA[uart_task event[0] size[19]]
>   27.028 [ESP8266 ] [RX] DATA[at_port_get_data_length[19]]
>   27.030 [ESP8266 ] [RX] DATA[at_port_read_data[19]]
>   27.033 [ESP8266 ] [RX] DATA[OK]
>   27.034 [ESP8266 ] [RX] DATA[>]
>   27.035 [ESP8266 ] [TX] LEN[2048]
>   27.050 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.053 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.056 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.059 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.062 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.065 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.069 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.072 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.074 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.078 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.081 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.084 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.090 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.093 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.096 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.100 [ESP8266 ] [RX] DATA[uart_task event[0] size[128]]
>   27.103 [ESP8266 ] [RX] DATA[at_port_get_data_length[128]]
>   27.105 [ESP8266 ] [RX] DATA[at_port_read_data[128]]
>   27.108 [ESP8266 ] [RX] DATA[uart_task event[2] size[128]]  >> Event UART_FIFO_OVF
>   27.111 [ESP8266 ] [RX] DATA[at_port_get_data_length[0]]    >> Received data discarded?!
>   27.113 [ESP8266 ] [RX] DATA[at_port_get_data_length[0]]
>   27.116 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.119 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.121 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.124 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.126 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.129 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.132 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.134 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.137 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.139 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.142 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.145 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.148 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.150 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.153 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.157 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.160 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.162 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.167 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.169 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.172 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.176 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.179 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.181 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.186 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.188 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.191 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.195 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.198 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.201 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.205 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.208 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.210 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.214 [ESP8266 ] [RX] DATA[uart_task event[0] size[110]]
>   27.217 [ESP8266 ] [RX] DATA[at_port_get_data_length[110]]
>   27.220 [ESP8266 ] [RX] DATA[at_port_read_data[110]]
>   27.222 [ESP8266 ] [RX] DATA[uart_task event[0] size[44]]
>   27.225 [ESP8266 ] [RX] DATA[at_port_get_data_length[44]]
>   27.227 [ESP8266 ] [RX] DATA[at_port_read_data[44]]
>   27.537 [ESP8266 ] AT-Timeout! (no xxxx bytes received)
>   27.539 [ESP8266 ] [TX] LEN[AT]
>   27.544 [ESP8266 ] [RX] DATA[uart_task event[0] size[3]]
>   27.546 [ESP8266 ] [RX] DATA[at_port_get_data_length[3]]
>   27.549 [ESP8266 ] [RX] DATA[at_port_read_data[3]]
> 
> No ok.... Latest AT is used as payload

**Here same facts:**

- If i only transmit packages smaller than 120 bytes (instead of 2048) the problem seems to be gone.
- if i increase the ring buffer to 8k, the problem is still there
- when i try to transmit a 10k file in blocks a 2k, i got the problem every time!
- i also encounted this problems with AT V1.7.4 SDK 3.0.5 (NonOS)

Maintraxx, 2020-09-01:
My Workaround for it: My Host splits the 2048 byte package into 128 byte chunks and added a delay of 3ms between all chunks.

teknikal2, 2021-04-14:
At 3M baud, I was able to workaround this issue by adding just 30uS after each 128 byte chunk.  The threshold for working or not was around 21uS and used 30uS for safety.

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

teknikal2, 2022-05-27:
I was working with AT firmware "ESP32-WROOM-32_AT_Bin_V2.1.0.0".  I don't know if there's been any improvement.