espressif/esp-at #456: Feature Request: Expect the throughput of ESP-AT over “SDIO-SPI” solution to reach 1MBytes/s

caowent, 2020-08-05:
## Environment

-Development Kit: [ESP32-DevKitC]

-Kit version: [v2]

-Module or chip used: [ESP32-WROOM-32U]

-IDF version:

    v4.0.1-193-ge7ac221b4

-Build System: [Make]

-Compiler version:

    (crosstool-NG esp-2019r2) 8.2.0

-Operating System: [Windows 64]

-Power Supply: [USB]

## Problem Description

I use the following block diagram scheme for high-speed wifi application.
[image]

[Espressif's official SDIOSPI host Demo address](https://github.com/espressif/esp-at/tree/master/examples/at_sdspi_host)

[LwESP AT parsing library project](https://github.com/MaJerle/lwesp)

ESP32 is used as a station mode to connect to the PC's AP, and use iperf to build a TCP server on the PC side.

STM32 establishes a connection with the iperf TCP server through the AT+CIPSTART command, and sends data to the server in the form of 2kBytes per packet (non-transparent) for upstream throughput testing.

### Expected Behavior

The official document WiFi TCP upstream throughput exceeds 6Mbytes/s, I compiled the iperf routine, and the measured speed can also be reached.
[image]

The description of the Sdiospi solution also mentioned that the throughput of the transferred TCP is close to 1.2MB/s (I think it should be 1.2MBytes/s).
[image]

So I expect the final TCP throughput of this solution to reach 1MBytes/s.

### Actual Behavior

The actual throughput measured by iperf is only about 250KBytes/s.
[image]

### Steps to repropduce

1. ESP AT over SDIO firmware compile and generate according to sdkconfig

2. The host uses SDIOSPI project

3. Create iperf server on PC

4. Connect to PC AP via AT command

5. AT+CIPSTART connect to iperf server

6. Turn on multi-connection mode AT+CIPMUX=1

7. You can write tcp flow test cases without porting LwESP, refer to freertos.c

8. AT+CIPSEND=ID,2048

9. Wait for "OK"and ">"

10. Send 2048 byte data packet

11. Wait for “SEND OK”

12. Loop to step 8

13. Get upstream throughput on PC iperf terminal

14. During the period, you can turn on wireshark to monitor the test process

### Code to reproduce this issue

[esp32 sdiospi solution throughput test (freertos.c)](https://gist.github.com/caowent/e18d6849dbc16b54394559ddce4a1673)

## Debug Logs

Systemview
[image]

wireshark
[image]

[image]

## Other items if possible

[File package.zip](https://github.com/espressif/esp-at/files/5027572/File.package.zip)

ESP-Coco, 2020-08-11:
Hi @caowent  The default sdkconfig of ESP32 can't achieve such a high rate because we need to balance memory and performance.  You need to modify the relevant configuration parameters in the sdkconfig file, which will result in an increase in memory consumption. The main configuration items modified are as follows:

```
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240

CONFIG_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y

CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=32
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=32

CONFIG_FREERTOS_UNICORE=y

CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64
```

caowent, 2020-08-11:
> Hi @caowent The default sdkconfig of ESP32 can't achieve such a high rate because we need to balance memory and performance. You need to modify the relevant configuration parameters in the sdkconfig [...]

Thank you for your reply, I will try it and give you feedback.

caowent, 2020-08-11:
1. Sorry, the effect of this configuration is not very obvious.

2. Earlier, I used the configuration of the [iperf routine](https://github.com/espressif/esp-idf/blob/master/examples/wifi/iperf/sdkconfig.defaults), and the throughput was about 250kBytes/s.

3. According to your recommended configuration this time, the throughput is also around 250kBytes/s.

4. However, I found that the configuration "CONFIG_FLASHMODE_QIO=y
CONFIG_FREERTOS_UNICORE=n” can increase the throughput to about 280kBytes/s. [sdkconfig](https://gist.github.com/caowent/bd8341e995c86098477ff0e352034c63)

[image]

5. This is the time stamp of the TCP sending process captured by my debugging trace. From the figure, it takes about 8ms to send 2kBytes data in each sending cycle, and the throughput is 250kBytes/s.

[image]

If the data length sent in each cycle can be extended to 4kBytes, and then the transmission time consumption can be reduced to 4ms, the throughput can be close to 1Mbytes/s.

ESP-Coco, 2020-08-11:
Hi @caowent ,
What is the MCU SPI clock you tested?
In addtion, can you test the rate using pass-through mode?  The normal transmission mode takes a lot of time to wait for AT command response. This is a my test code 
[app_main.log](https://github.com/espressif/esp-at/files/5055287/app_main.log)
1. Enable the transparent transmission mode, AT+CIPMODE=1
2. Send data    AT+CIPSEND
3. Test SDIO send throughput         AT+TESTSEND    This will send the send data until the request is met

caowent, 2020-08-11:
Hi, @ESP-Coco 

1. My SPI clock is 21MHz.

2. The transparent transmission mode is an alternative, because it only supports a single link. Nevertheless, I tried the transparent transmission mode, but I had to face [a new problem](https://github.com/espressif/esp-at/issues/457).

alireza3137, 2020-12-01:
hi @caowent 
Can you find any solution for increase speed transmission to 1MBytes/s?

caowent, 2020-12-02:
> hi @caowent
> Can you find any solution for increase speed transmission to 1MBytes/s?

Here is another solution for mcu high speed wifi, which is worth exploring.
https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/MCU_based_readme.md

This is another MCU high-speed wifi solution based on ESP8266.
[https://github.com/Anylinkin/ALK8266WIFI](url)
[https://item.taobao.com/item.htm?id=576141575067](url)

alireza3137, 2020-12-02:
@caowent  
Many thanks for your reply.
In esp32 project you mention what is the transmission speed?

caowent, 2020-12-03:
> @caowent
> Many thanks for your reply.
> In esp32 project you mention what is the transmission speed?

I haven't tested the esp-hosted project. It requires a certain amount of work to be used normally, but I think its speed can reach 1MBytes/s, even better than the ESP-AT solution.

alireza3137, 2020-12-03:
@caowent 
many Thanks