espressif/esp-at #457: Bug Report: The TCP stream in the wifi transparent mode is partially wrong and failed checksum

caowent, 2020-08-06:
## Environment

-Development Kit: [ESP32-DevKitC]

-Kit version: [v2]

-Module or chip used: [ESP32-WROOM-32U]

-IDF version:

v4.0.1-193-ge7ac221b4
-Build System: [Make]

-Compiler version:

(crosstool-NG esp-2019r2) 8.2.0
-Operating System: [Windows 64]

-Power Supply: [USB]

## Problem Description

My application requires high speed and data accuracy, please refer to [the specific plan](https://github.com/espressif/esp-at/issues/456).

At first, I hope to meet the requirements through the non-transparent transmission mode, but unfortunately, although the data in the non-transparent transmission mode is not problematic, the speed is very limited. So I chose the transparent transmission mode and tested the speed with iperf. As the SDIOSPI solution said, the throughput reached 1.2MBytes/s. 

[image]

But when I went to verify the correctness of the data, the problem was exposed.

[This problem seems to be the same problem, see.](https://github.com/espressif/esp-at/issues/444)

After establishing a connection with the iperf TCP server, STM32 sends a picture (407KB) to the iperf TCP server through the transparent transmission mode. During the period, wireshark is turned on for monitoring, and the picture data is restored from Wireshark's Analyze -> Follow -> TCP Stream at the end of the transmission.

### Expected Behavior

The received picture is correct, and the CRC32 check code calculated by 7zip is consistent with the original picture check code.
[image]

### Actual Behavior

The data length is correct, some garbled codes appear in the picture, and the CRC check codes are inconsistent.
[image]

### Steps to repropduce

1. Turn on wireshark monitoring on the PC side

2. Establish a TCP connection to the iperf server on the PC side

3. Transmit a picture through transparent transmission mode

4. Restore picture from wireshark

5. Compare and verify with the original picture

### Try to fix

1. Sending  every 512 bytes data packet delay 20ms, there is still a bit of garbled.
[image]

2. Change to UDP transmission, no delay, the data length will be 3KB or more less.

3.UDP transmission, delay 1ms after sending a 512-byte data packet, the picture is correct, and the verification passed.

4. STM32 sends data to ESP with a synchronization signal (using SDIO general-purpose registers). Waitting for ESP32 to read the buffer before sending the next packet of data, and the garbled distribution is relatively concentrated.
[image]

5. When ESP32 reads the buffer, the buffer data is sent back to STM32 via SPI. Use the semi-host method to write to the file on the STM32 side, and checked with the original picture at the end of the transmission. The verification is passed, indicating that the previous communication link is normal, and the problem occurred after being sent to AT CORE.

### Code to reproduce this issue

Increase the synchronization mechanism code (using SDIO general registers), based on the sdiospi project, including modifications on both sides of STM32 and ESP32
[code](https://gist.github.com/caowent/ad50c02adc3cf160d6a468a46d06d7ab)

## Debug Logs

wireshark TCP flow passthrough_tcp_without_delay
[image]

## Other items if possible
[passthrough.zip](https://github.com/espressif/esp-at/files/5033531/passthrough.zip)

ldsheng, 2020-08-06:
I'm facing similar problems. 
I'm using the SDIO-AT too to perform the mbedtls communication over TCP. The connection may be closed during the transmission and it happens randomly. However, when I use the TCP without mbedtls, it never happens. So, I think there may be some bit errors occurred in the data which cause the TCP server to close the TCP connection. 
Looking for a resolution here. 
thx

ESP-Coco, 2020-08-11:
Hi @caowent ,
Thanks,  can you test with the latest master version? We have just recently solved the problem of incorrect data in high throughput, somewhat like what you described (https://github.com/espressif/esp-at/commit/ef77d4066d4e7cd76397268d29192762c30552dd)

Hi @ldsheng ,
I'm not sure this fix applies to your problem, but I still recommend that you test with the latest master branch as well.

If the problem is still not resolved, we will try to add your replicating steps to the test case, replicating and analyzing the problem.

caowent, 2020-08-11:
1. I noticed an update for [this issue](https://github.com/espressif/esp-at/issues/444), and I just retested it with the latest update.
2. The problem has been improved a lot, but it has not been completely resolved.
3. I still have to use the cache synchronization mechanism to ensure that I transfer the data to the ESP32 correctly. [reference](https://gist.github.com/caowent/ad50c02adc3cf160d6a468a46d06d7ab)
4. The speed can reach 1.4MBytes/s (11Mbits/s).

[image]

5. I sent the same 400kB picture 10 times, and selected one of them for verification.
6. The following is a comparison.
The previous AT-CORE version with cache synchronization

[image]

The latest AT-CORE version with cache synchronization

[image]

ESP-Coco, 2020-08-11:
OK, Thanks for your test.
We will write a test case to analyze this problem.

ldsheng, 2020-08-11:
> you described (ef77d40

HI

I have tested the newest version 784682f at master branch. It does
been improved a lot. However, the problem is still there.
[image: image.png]
In my test, the speed is about 75KByte/s.

caowent, 2020-08-25:
How is the progress in fixing this issue?  @ESP-Coco

caowent, 2020-09-02:
The latest version(a9e1ac6) has been tested and improved a lot, but there are still errors.
[image]

ldsheng, 2020-11-13:
Is there any improvement on this bug?

caowent, 2020-11-13:
> Is there any improvement on this bug?

 Here is another solution for mcu wifi, which is worth exploring.
[https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/MCU_based_readme.md](url)

ldsheng, 2020-11-13:
> > Is there any improvement on this bug?
> 
> Here is another solution for mcu high speed wifi, which is worth exploring.
> https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/MCU_based_readme.md

My project is highly similar to the solution you mentioned. I'm using a stm32f427 as the MCU which communicates with esp32 via SPI port( initially using the serial port). The problem is that under heavy load transferring, there are random transmission errors on the ESP32. I'm waiting for you guys to solve this problem. thx

caowent, 2020-11-13:
> My project is highly similar to the solution you mentioned. I'm using a stm32f427 as the MCU which communicates with esp32 via SPI port( initially using the serial port). The problem is that under [...]

The non-transparent mode works normally, but the speed is relatively slow, only 250kBytes/s.
[https://github.com/espressif/esp-at/issues/456](https://github.com/espressif/esp-at/issues/456)

ESP-Coco, 2021-02-25:
HI @caowent @ldsheng , so sorry for the late reply, we have fixed the problem, please follow this [commit ](https://github.com/espressif/esp-at/commit/764f672b4316fb0532f3dc5a02fe56a4af9459ac) and test again

ldsheng, 2021-02-25:
HI, thx for letting me know.

I have tried to compile the new commits. However, it failed.

After updating the esp_idf and declare the new ESP_AT_PORJECT_PATH system variable, I finished the "make menuconfig" job successfully. Then, when I do "make" it responses :
[image]

ESP-Coco, 2021-02-25:
The latest version needs to have upgraded the IDF version to V4.2 and needs to be compiled with Cmake. we recommend using the follow compilation method in the AT document. https://github.com/espressif/esp-at/blob/master/docs/en/Compile_and_Develop/How_to_clone_project_and_compile_it.rst#compiling-and-flashing-the-project

ldsheng, 2021-02-26:
Still fails to compile.

I have update the compile environment according to instructions of here:
![image](https://user-images.githubusercontent.com/3348857/109255799-aa3f4000-782f-11eb-8a2c-8b81a9d8de4a.png
And then execute subsequent instructions as shown below.

esp@esp:~/project$ **ls**
esp-at  esp-at-esp32  esp-at-esp8266
esp@esp:~/project$ rm -rf esp-at
esp@esp:~/project$ ls
esp-at-esp32  esp-at-esp8266
esp@esp:~/project$ git clone --recursive https://github.com/espressif/esp-at.git
Cloning into 'esp-at'...
remote: Enumerating objects: 956, done.
remote: Counting objects: 100% (956/956), done.
remote: Compressing objects: 100% (508/508), done.
remote: Total 7951 (delta 595), reused 760 (delta 434), pack-reused 6995
Receiving objects: 100% (7951/7951), 87.14 MiB | 1.16 MiB/s, done.
Resolving deltas: 100% (4406/4406), done.
esp@esp:~/project$ cd esp-at
esp@esp:~/project/esp-at$ rm -rf build sdconfig
esp@esp:~/project/esp-at$ rm -rf esp-idf
esp@esp:~/project/esp-at$ ./build.py menuconfig
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP8266
3. PLATFORM_ESP32S2
choose(range[1,3]):1    

Module name:
1. WROOM-32
2. WROVER-32
3. PICO-D4
4. SOLO-1
5. MINI-1 (description: ESP32-U4WDH chip inside)
6. ESP32-D2WD (description: 2MB flash, No OTA)
choose(range[1,6]):1

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):0
platform_name=ESP32,module_name=WROOM-32
Please wait for the SDK download to finish...
Cloning into 'esp-idf'...
remote: Enumerating objects: 968, done.
remote: Counting objects: 100% (968/968), done.
remote: Compressing objects: 100% (524/524), done.
remote: Total 220538 (delta 544), reused 666 (delta 430), pack-reused 219570
Receiving objects: 100% (220538/220538), 125.00 MiB | 1.70 MiB/s, done.
Resolving deltas: 100% (163337/163337), done.
old commit:af8bc4c43615b0500f1b8df52a22726d77b6f0af
checkout commit:c40f2590bf759ff60ef122afa79b4ec04e7633d2
Please wait for the update to complete, which will take some time
Already up to date.
Note: checking out 'c40f2590bf759ff60ef122afa79b4ec04e7633d2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at c40f2590b Merge branch 'featrue/remove_esp32-S2_preview_v4.2' into 'release/v4.2'
Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Submodule 'components/bt/controller/lib' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib'
[... 401 lines cut ...]
cd /home/esp/project/esp-at/build && /home/esp/.espressif/python_env/idf4.2_py3.8_env/bin/python /home/esp/project/esp-at/tools/esp_at_factory_bin_combine.py --module_name WROOM-32 --bin_directory /home/esp/project/esp-at/build --flash_mode dio --flash_size 4MB --flash_speed 40m --parameter_file /home/esp/project/esp-at/build/factory/factory_parameter.log --download_config /home/esp/project/esp-at/build/flash_project_args
0x8000,partition_table/partition-table.bin
0x10000,ota_data_initial.bin
0xf000,phy_init_data.bin
0x1000,bootloader/bootloader.bin
0x100000,esp-at.bin
0x20000,at_customize.bin
0x21000,customized_partitions/ble_data.bin
Traceback (most recent call last):
  File "/home/esp/project/esp-at/tools/esp_at_factory_bin_combine.py", line 98, in <module>
    main()
  File "/home/esp/project/esp-at/tools/esp_at_factory_bin_combine.py", line 94, in main
    esp32_at_combine_bin(args.module_name.upper(), args.flash_mode.upper(), args.flash_size.upper(), 
  File "/home/esp/project/esp-at/tools/esp_at_factory_bin_combine.py", line 47, in esp32_at_combine_bin
    with open(os.path.join(build_dir, bin_file), 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/esp/project/esp-at/build/customized_partitions/ble_data.bin'
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
idf.py build ret: 2
idf.py build failed
esp@esp:~/project/esp-at$ 

It seems that the ble_data.bin has not been output. Am I missing something?

Thx

ESP-Coco, 2021-02-26:
I found this error "No module named 'yaml'", just install through pip and retry.

ustccw, 2021-02-26:
@ldsheng As the doc said, you should install the xlrd and pyyaml module firstly. Maybe you can try with the following shell cmd:
```sh
# install esp-idf env firstly
cd esp-idf
./install.sh
. ./export.sh

# install esp-at necessary python module
pip install xlrd pyyaml
cd -

# build the esp-at project
./build.py build
```

ldsheng, 2021-02-26:
@ESP-Coco @ustccw Yes! After installing the pyyaml, I compiled esp-at successfully. Thank you! Now I start to restore my test environment. I'll let you know as soon as I get some results!

thx

ldsheng, 2021-03-01:
HI,

Since last Friday, after successfully compiled the commit 4c8bf77, I have run the test several times,  each test lasts more than 1 hour. The throughput is about 80 Kbytes per second. 

At first, I run the test for about 1 hour and then stop. Everything looks good. No errors occur.
Then, I extend the test. I found that errors occur when the test lasts more than 2 hours. This has happened 2 times. I'm not sure if this is still the bug or a problem with my router (Linksys WRT3200ACM). The snapshot when the error occurs is as below.
[image]
I'll keep on the test and update the results.

Anyway, the improvement is really impressive. Thank you!

ustccw, 2021-03-12:
@ldsheng Thank you for your patient reply, while i was busying addind IPv6 features and esp32c3 chip into esp-at in last two weeks.

Will look into this valuable issue next week, thanks again for your feedback.

ustccw, 2021-03-19:
@ldsheng could you please take a look, more details in README.md of the attachment.

[GHI457.zip](https://github.com/espressif/esp-at/files/6168992/GHI457.zip)

ustccw, 2021-03-19:
@ldsheng If anything unclear, please let me know.

ldsheng, 2021-03-19:
@ustccw I got your message. OK, I think I can do some tests next week. I'll let you know when I get some results.

ustccw, 2021-03-30:
@ldsheng Is it possible that there is a reply this week.

ldsheng, 2021-04-01:
Hi, I'm sorry for the delay but I don't think I can do the test this week.
Because my family is seriously ill since 14th March, I cannot go back to
work recently.

ustccw, 2021-04-01:
i am sorry to hear that, hope to get well soon~

caowent, 2021-04-01:
Maybe i will do the test later

发自我的iPhone

ustccw, 2022-05-27:
May I know if there is any update? If the issue still exists, please let us know.

caowent, 2022-05-27:
> May I know if there is any update? If the issue still exists, please let us know.

Sorry, I have no condition test now.