espressif/esp-at #458: Sending large file with many AT+CIPSEND=0,2048 fails. (same code works with AT 1.7)

jandrassy, 2020-08-09:
## Environment

- Development Kit/module:  Espressif WROOM-32 dev kit (ESP32,WROOM-32); esp8266  Wemos D1 mini 4MB flash; esp8266  Wemos D1 mini 1MB flash;
- host dev board: Arduino Mega; Arduino Zero
- esp-at version:  2.1.0 released binaries; v2.1.0.0-rc1_esp8266-30-g7f60470

## Intro

I made an Arduino networking library last year which uses the passive receive mode of the AT firmware. I published it only for AT 1.7 for esp8266 because at that time AT 2 passive mode was only in development. I followed and tested the partial implementations of the passive receive mode in AT 2. With release of AT 2.1 I finished the changes in my library to support AT 2. There are still problems with usability of UDP messages receiving in passive receive mode reported in a separate issue.

## Problem Description

One of the example sketches of my library is a web server listing directories and serving files from SD card. With AT firmware 1.7 it is able to download a 1 MB file without problems but AT 2.1 fails on the same esp8266 hardware setup and on esp32 too. The data are transferred with `AT+CIPSEND=0,2048`. After thousands of bytes transferred at one point AT 2.1 doesn't read all 2048 bytes sent, and the next `AT+CIPSEND` immediately fails with ERROR (no `>`). UART flow control doesn't help.

## Debug Logs

This is the tail of the debug log of the library (the AT log with DEBUG level on UART0 didn't print anything at this time):
```
esp> AT+CIPSEND=0,2048 ...sent
esp> OK ...ignored
esp> > ...matched
esp> Recv 2048 bytes ...matched
esp> SEND OK ...matched
esp> AT+CIPSEND=0,2048 ...sent
esp> OK ...ignored
esp> > ...matched
esp> Recv 251 bytes ...matched
esp> SEND OK ...matched
esp> AT+CIPSEND=0,2048 ...sent
esp> ERROR ...error
esp ERROR: expected > got ERROR
esp ERROR: CIPSEND failed at 61691
```

xcguang, 2020-08-19:
`SEND OK` is really sent successfully in v1.7, but after v2.0.0.0, it means just copy data into protocol buffer, maybe there is no buffer in the protocol stack.

jandrassy, 2020-08-19:
> `SEND OK` is really sent successfully in v1.7, but after v2.0.0.0, it means just copy data into protocol buffer, maybe there is no buffer in the protocol stack.

after 60 kB transferred to AT fw over slow 115200 baud it would suddenly have no free buffer? 
in other test it failed after 140 kB

ubichsel, 2020-08-24:
I had similar issues on my end. I could solve it by enable hardware flow control on UART (I use 5MBaud). In this way the UART signals to wait if buffer is full on ESP32 and continuous when free space is available.

jandrassy, 2020-08-24:
> I had similar issues on my end. I could solve it by enable hardware flow control on UART (I use 5MBaud). In this way the UART signals to wait if buffer is full on ESP32 and continuous when free space is available.

in his case UART flow control didn't help.

Maintraxx, 2020-08-31:
I have the same problem. Unfortunately I can't change the hardware anymore to connect hardware flow control (sold over 5000 times).
I saw, that normaly the uart interrupt is fast enough to receive all bytes. Threshold 64 bytes > received than about 67 bytes. The number of real received bytes are normaly under 70. Nevertheless runs the fifo sometimes over (128 byte). 

Why does the interrupt not execute in time? Is there any blocking function?

Maintraxx, 2020-09-01:
My Workaround for it: My Host splits the 2048 byte package into 128 byte chunks and added a delay of 3ms between all chunks.

jandrassy, 2021-07-10:
this is not fixed in 2.2

JAndrassy, 2022-05-26:
fixed in 2.4.0.0

closed 2022-05-26 by JAndrassy