espressif/esp-at #459: Immediately Disconnects After Receiving IP on Enterprise WiFi

legoabram, 2020-08-12:
## Environment

- Development Kit:      EVK-NINA-W102
- Module or chip used:  NINA-W102 (ESP32D0WDQ6)
- IDF version: v4.0.1-193-ge7ac221b4
- Build System: CMake
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows (Compiling using WSL)
- Power Supply: USB

## Problem Description

I'm attempting to connect to Enterprise WiFi (eduroam). I managed to get the certificates working, and the ESP authenticates successfully. It connects to the AP, and receives an IP address.  The IT guys at the university also confirm that it is authenticating correctly. However, once it does get its IP address, it immediately disconnects from the AP. I'm pretty sure this is on the ESP side at this point, especially with the `wifi:invalid amk` mentioned in the wifi debug log. But even then, I'm still stumped.

## Debug Logs

```
2.1.0
I (848) wifi:mode : sta (24:62:ab:b6:ef:d8) + softAP (24:62:ab:b6:ef:d9)
D (849) wifi:filter: set rx policy=1
D (849) wifi:connect status 0 -> 0
D (851) wifi:connect status 0 -> 0
D (851) wifi:Start wifi disconnect
D (851) wifi:connect status 0 -> 0
D (851) wifi:filter: set rx policy=8
D (851) wifi:filter: set rx policy=2
D (851) wifi:clear blacklist
D (852) wifi:clear rc list
I (852) wifi:mode : softAP (24:62:ab:b6:ef:d9)
max tx power=78,ret=0
I (855) wifi:set country: cc=US schan=1 nchan=11 policy=1

ready
AT+CWMODE

ERROR
AT+CWMODE=1
D (8255) wifi:filter: set rx policy=10
I (8256) wifi:mode : sta (24:62:ab:b6:ef:d8)
D (8257) wifi:filter: set rx policy=1
D (8258) wifi:connect status 0 -> 0

OK
AT+CWJEAP="eduroam",0,"someone@somewhere.edu",,,3
I (18307) wpa: WPA2: already disabled
D (18308) wifi:Start wifi disconnect
D (18308) wifi:connect status 0 -> 0
D (18309) wifi:filter: set rx policy=8
D (18309) wifi:clear blacklist
ca_len=4344
cert_len=1805
prvtkey_len=1675
I (18318) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

D (18319) wifi:Start wifi connect
D (18319) wifi:connect status 0 -> 0
D (18319) wifi:connect chan=0
D (18319) wifi:first chan=1
D (18319) wifi:connect status 0 -> 1
D (18319) wifi:filter: set rx policy=3
D (18320) wifi:clear scan ap list
D (18320) wifi:start scan: type=0x50f, priority=2, cb=0x4014acac, arg=0, ss_state=0x1, time=17493771, index=0
D (18321) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
D (18443) wifi:scan end: arg=0, status=0, ss_state=0x3
D (18443) wifi:perform scan: ss_state=0x9, chan<2,0>, dur<0,120>
D (18564) wifi:scan end: arg=0, status=0, ss_state=0x3
D (18564) wifi:perform scan: ss_state=0x9, chan<3,0>, dur<0,120>
D (18685) wifi:scan end: arg=0, status=0, ss_state=0x3
D (18685) wifi:perform scan: ss_state=0x9, chan<4,0>, dur<0,120>
D (18806) wifi:scan end: arg=0, status=0, ss_state=0x3
D (18806) wifi:perform scan: ss_state=0x9, chan<5,0>, dur<0,120>
D (18927) wifi:scan end: arg=0, status=0, ss_state=0x3
D (18927) wifi:perform scan: ss_state=0x9, chan<6,0>, dur<0,120>
D (18929) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (18930) wifi:profile match: ss_state=0x7
D (18931) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (18931) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (18931) wifi:sig_b=1, sig_g=0, sig_n=0, max_b=0, max_g=108, max_n=144
D (18932) wifi:profile match: ss_state=0x7
D (18932) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (18933) wifi:profile match: ss_state=0x7
D (18933) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (18934) wifi:profile match: ss_state=0x7
D (18996) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (18996) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (18996) wifi:sig_b=1, sig_g=0, sig_n=0, max_b=0, max_g=108, max_n=144
D (18997) wifi:profile match: ss_state=0x7
D (18998) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (18998) wifi:profile match: ss_state=0x7
D (19047) wifi:scan end: arg=0, status=0, ss_state=0x7
D (19048) wifi:find first mathched ssid, scan done
D (19048) wifi:filter: set rx policy=4
D (19048) wifi:first chan=1
D (19049) wifi:handoff_cb: status=0
D (19049) wifi:ap found, mac=00:f6:63:66:a7:41
D (19049) wifi:new_bss=0x3ffb78d0, cur_bss=0, new_chan=<6,0>, cur_chan=1
D (19050) wifi:filter: set rx policy=5
I (19050) wifi:new:<6,0>, old:<1,1>, ap:<255,255>, sta:<6,0>, prof:1
D (19051) wifi:connect_op: status=0, auth=4, cipher=3
D (19052) wifi:auth mode is not none
D (19634) wifi:connect_bss: auth=1, reconnect=0
I (19635) wifi:state: init -> auth (b0)
D (19635) wifi:start 1s AUTH timer
D (19635) wifi:clear scan ap list
D (19636) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (19637) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (19637) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (19637) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (19638) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (19638) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (19641) wifi:recv auth: seq=2, status=0
I (19641) wifi:state: auth -> assoc (0)
D (19641) wifi:restart connect 1s timer for assoc
D (19645) wifi:recv assoc: type=0x10
D (19645) wifi:filter: set rx policy=6
I (19645) wifi:state: assoc -> run (10)
I (19646) wpa: wpa2_task prio:2, stack:6144

D (19646) wifi:start 30s connect timer for 4 way handshake
I (19692) wpa: SSL: Need 3087 bytes more input data
I (19704) wpa: SSL: Need 2085 bytes more input data
D (19715) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

I (19721) wpa: SSL: Need 1083 bytes more input data
I (19740) wpa: SSL: Need 81 bytes more input data
D (21838) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (21838) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (21839) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (21839) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (21839) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (21840) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (21840) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (21841) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (21968) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

I (21970) wpa: >>>>>wpa2 FINISH

I (21976) wifi:connected with eduroam, aid = 1, channel 6, BW20, bssid = 00:f6:63:66:a7:41
I (21977) wifi:security: WPA2-Ent, phy: bgn, rssi: -74
D (21977) wifi:remove all except 00:f6:63:66:a7:41 from rc list
D (21977) wifi:clear blacklist
D (21982) wifi:filter: set rx policy=7
I (21982) wifi:pm start, type: 1

D (21982) wifi:Send sta connected event
D (21982) wifi:connect status 1 -> 5
D (21983) wifi:obss scan is disabled
D (21983) wifi:start obss scan: obss scan is stopped
WIFI CONNECTED
I (22070) wifi:AP's beacon interval = 102400 us, DTIM period = 3
D (22071) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (22378) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (22685) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (22992) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

I (23141) tcpip_adapter: sta ip: 144.35.32.59, mask: 255.255.254.0, gw: 144.35.32.1
BLUFI BLE is not connected yet
I (23142) wpa: WPA2 ENTERPRISE VERSION: [v2.0] disable

WIFI GOT IP

OK
D (23198) wifi:invalid akm
D (23198) wifi:Start wifi disconnect
I (23199) wifi:state: run -> init (0)
D (23199) wifi:connect status 5 -> 6
D (23199) wifi:stop beacon/connect timer, send diassoc(8)
D (23200) wifi:sta leave
I (23200) wifi:pm stop, total sleep time: lu us / lu us

D (23200) wifi:stop CSA timer
D (23201) wifi:remove 00:f6:63:66:a7:41 from rc list
I (23201) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
D (23202) wifi:filter: set rx policy=8
D (23203) wifi:Send disconnect event, reason=8, AP number=0
D (23203) wifi:connect status 6 -> 0
D (23204) wifi:filter: set rx policy=8
WIFI DISCONNECT
```

## Project Configuration

I disabled the Bluetooth since I didn't need it and wanted to save space, but it wouldn't compile due to a missing reference to `esp_blufi_send_wifi_conn_report` from `libesp32_at_core.a`. I got around this by adding this dummy function in `app_main.c`. I think this function should be added as a weak symbol if possible.

```
esp_err_t esp_blufi_send_wifi_conn_report(void*a, void*b, void*c, void*d) { return ESP_ERR_INVALID_STATE; }
```

This does reference my custom module folder which you can just replace with module_esp32_default.
[sdkconfig.txt](https://github.com/espressif/esp-at/files/5060037/sdkconfig.txt)

ustccw, 2020-08-24:
@legoabram Thanks feedback, will take a look.

ustccw, 2020-08-25:
About missing reference to `esp_blufi_send_wifi_conn_report`, esp-at fixed it internal, will auto-sync to GitHub after CI passed.

ustccw, 2020-08-25:
@legoabram About ESP32-AT disconnect from enterprise AP. I tested several times, I am afraid that can not reproduce.

[image]

What I suggest as the following:
- use the raw esp-at configuration as far as possible
such as use  `CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG=y` instead of `CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE=y`,for the new configuration can not get verified by test team.

- test with wpa2 enterprise raw demo
try with [https://github.com/espressif/esp-idf/tree/release/v4.0/examples/wifi/wpa2_enterprise](https://github.com/espressif/esp-idf/tree/release/v4.0/examples/wifi/wpa2_enterprise), see what happened.

closed 2020-08-26 by igrr