espressif/esp-at #460: AT command for read characteristic in server mode

gitsteven73, 2020-08-12:
Hello all:
I'm using AT command for ESP32 and create BLE GATT server OK, 
but for read / write characteristic , only :
AT+BLEGATTSSETATTR—GATTS Sets Characteristic  work fine (for write)
there is no read characteristic command for server mode ...
(of cause already try : AT+BLEGATTCRD : GATTC reads characteristics , but it for client mode)

please help , thanks a lot.

xiewenxiang, 2020-08-20:
@gitsteven73 

Yes, `AT+BLEGATTCRD ` is just for GATT client side and `AT+BLEGATTSSETATTR` just for Server side.

HectorJimenezMendez, 2020-10-13:
Hi, @xiewenxiang.

As long as `AT+BLEGATTCRD` cannot be used from server side... is there any AT command which lets the server read the value of a characteristic?

Thank you and best regards

rfrowe, 2021-02-19:
@HectorJimenezMendez #484

Luxin812, 2021-09-14:
@gitsteven73 @HectorJimenezMendez Thank you for using ESP-AT. At present, ESP-AT master has implemented AT+BLEGATTCRD command on the server side and AT+BLEGATTSSETATTR command on the client side. If you still need these functions, please test and develop based on ESP-AT master

AntonDrubetskiy, 2021-09-17:
@Luxin812 Hello!))
My ESP-AT firmware is in the master branch, however the AT + BLEGATTCRD command throws an error.
Here is the course of my actions:

AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEADVSTART

OK
+BLECONN:0,"62:7f:8f:7a:ef:9e"
AT+BLEGATTCRD=0,1,1

ERROR

The result of the AT + GMR command:

AT + GMR
AT version: 2.3.0.0-dev (5092dec - ESP32C3 - Sep 3 2021 12:00:11)
SDK version: v4.3-302-g47b96db12d
compile time (e5ca22e): Sep 16 2021 18:04:47
Bin version: 2.2.0 (MINI-1)

OK

Luxin812, 2021-09-23:
@AntonDrubetskiy  When the AT+BLEGATTCRD command is to be executed on the BLE server, the BLE client must actively establish a connection with the BLE server, and the BLE server must also actively send a connection action with the BLE client. You can try the following AT command sequence:
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
AT+BLEADVSTART

OK
+BLECONN:0,"62:7f:8f:7a:ef:9e"

**AT+BLECONN=0,"62:7f:8f:7a:ef:9e"
OK**

AT+BLEGATTCRD=0,1,1