espressif/esp-at #461: How do I make menuconfig w/o bluetooth - compile error

agatrost, 2020-08-19:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.0
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB|external 5V]

## Problem Description
I am trying to do a minimal build with the ESP-AT project, and I am having trouble removing the bluetooth portion.

//Detailed problem description goes here.
LD build/esp-at.elf
/home/agatrost/.esp32/Toolchain/xtensa-esp32-elf_8_2_0/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/agatrost/WorkDir/UTEC_Tstat/SandBox/esp32_master/components/at/lib/libesp32_at_core.a(at_wifiCmd.c.obj):(.literal.esp_at_wifi_event_handler+0x38): undefined reference to `esp_blufi_send_wifi_conn_report'
/home/agatrost/.esp32/Toolchain/xtensa-esp32-elf_8_2_0/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/agatrost/WorkDir/UTEC_Tstat/SandBox/esp32_master/components/at/lib/libesp32_at_core.a(at_wifiCmd.c.obj): in function `esp_at_wifi_event_handler':
(.text.esp_at_wifi_event_handler+0xb6): undefined reference to `esp_blufi_send_wifi_conn_report'
/home/agatrost/.esp32/Toolchain/xtensa-esp32-elf_8_2_0/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: (.text.esp_at_wifi_event_handler+0xca): undefined reference to `esp_blufi_send_wifi_conn_report'
/home/agatrost/.esp32/Toolchain/xtensa-esp32-elf_8_2_0/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: (.text.esp_at_wifi_event_handler+0x25e): undefined reference to `esp_blufi_send_wifi_conn_report'
collect2: error: ld returned 1 exit status
/home/agatrost/WorkDir/UTEC_Tstat/SandBox/esp32_master/esp-idf/make/project.mk:528: recipe for target '/home/agatrost/WorkDir/UTEC_Tstat/SandBox/esp32_master/build/esp-at.elf' failed
make: *** [/home/agatrost/WorkDir/UTEC_Tstat/SandBox/esp32_master/build/esp-at.elf] Error 1

### Expected Behavior
Compile and built
### Actual Behavior
Errors out with the linker
### Steps to repropduce

1. make menuconfig
2. remove anything to do with bluetooth, 
     Component config-->Bluetooth (verify not checked)
     AT-->
        [ ] AT ble command support. 
        [ ] AT ble hid command support.
        [ ] AT blufi command support.
        [ ] AT bt command support.            

// It helps if you attach a picture of your setup/wiring here.
This is a build issue and not a hardware issue.

### Code to reproduce this issue
Don't need to change anything in the project
```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
Build issue as shown above

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
     added .txt so it would upload
      [sdkconfig.txt](https://github.com/espressif/esp-at/files/5093385/sdkconfig.txt)

The whole issue started when the checkout that i had locked in to use for my project had broken hardware flow control.  So I am looking for a newer checkout, where it is fixed, and I decided to try to go with a release, and or the head.  But I am also needing a very minimal build on the AT project.

Thank you

legoabram, 2020-08-19:
This is something I mentioned briefly in my own issue report: #459 

I disabled the Bluetooth since I didn't need it and wanted to save space, but it wouldn't compile due to a missing reference to esp_blufi_send_wifi_conn_report from libesp32_at_core.a. I got around this by adding this dummy function in app_main.c. I think this function should be added as a weak symbol if possible.

```
esp_err_t esp_blufi_send_wifi_conn_report(void*a, void*b, void*c, void*d) { return ESP_ERR_INVALID_STATE; }
```

Just add this function at the end of app_main.c and it will compile just fine. 👍

ustccw, 2020-08-25:
About missing reference to `esp_blufi_send_wifi_conn_report`, esp-at fixed it internal, will auto-sync to GitHub after CI passed.

closed 2020-08-26 by igrr