espressif/esp-at #462: AT+BLEGATTSSRVCRE return ERROR

marcocossali, 2020-08-19:
## Environment

- Custom Board;
- Module chip: WROVER32-E;
- IDF version: V2.1.0.0;
- Build System: Make;
- Compiler version: V8.2.0;
- OS: Linux;
- PS: 3V3.

## Problem description:

Good morning,

I'm trying to enable the BLE function as Server.
I used the following command:

AT+BLEINIT=2 -> OK

AT+BLEADVSTART -> OK

AT+BLEGATTSSRVCRE -> ERROR

the command to create the service return the "ERROR" message.

I'have followed the guide "https://github.com/espressif/esp-at/blob/master/docs/en/Compile_and_Develop/How_to_customize_BLE_services.md" to generate the ble_data.bin and this is the log after 'make' command:

Python requirements from /home/marco/esp/esp-at/esp-idf/requirements.txt are satisfied.
building AT customized partition table module_config/module_esp32_default/at_customize.csv
building AT customized partition table succeed
generating AT customized partition bins to /home/marco/esp/esp-at/build/customized_partitions
generating ble_data.bin: python /home/marco/esp/esp-at/tools/BLEService.py -t /home/marco/esp/esp-at/build/customized_partitions/ble_data.bin /home/marco/esp/esp-at/components/customized_partitions/raw_data/ble_data/example.csv
AT customized partitions successfully generated: ble_data
generating server_cert.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/server_cert cert /home/marco/esp/esp-at/components/customized_partitions/raw_data/server_cert/server_cert.crt 
AT customized partitions successfully generated: server_cert
generating server_key.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/server_key key /home/marco/esp/esp-at/components/customized_partitions/raw_data/server_key/server.key 
AT customized partitions successfully generated: server_key
generating server_ca.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/server_ca ca /home/marco/esp/esp-at/components/customized_partitions/raw_data/server_ca/server_ca.crt 
AT customized partitions successfully generated: server_ca
generating client_cert.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/client_cert cert /home/marco/esp/esp-at/components/customized_partitions/raw_data/client_cert/client_cert_00.crt cert /home/marco/esp/esp-at/components/customized_partitions/raw_data/client_cert/client_cert_01.crt 
AT customized partitions successfully generated: client_cert
generating client_key.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/client_key key /home/marco/esp/esp-at/components/customized_partitions/raw_data/client_key/client_key_00.key key /home/marco/esp/esp-at/components/customized_partitions/raw_data/client_key/client_key_01.key 
AT customized partitions successfully generated: client_key
generating client_ca.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/client_ca ca /home/marco/esp/esp-at/components/customized_partitions/raw_data/client_ca/client_ca_00.crt ca /home/marco/esp/esp-at/components/customized_partitions/raw_data/client_ca/client_ca_01.crt 
AT customized partitions successfully generated: client_ca
generating factory_param.bin(WROVER-32): python /home/marco/esp/esp-at/tools/factory_param_generate.py -t /home/marco/esp/esp-at/build/customized_partitions/factory_param.bin /home/marco/esp/esp-at/components/customized_partitions/raw_data/factory_param/factory_param_data.csv
generate parameter bin: platform PLATFORM_ESP32, module name WROVER-32
AT customized partitions successfully generated: factory_param
generating mqtt_cert.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/mqtt_cert cert /home/marco/esp/esp-at/components/customized_partitions/raw_data/mqtt_cert/mqtt_client.crt 
AT customized partitions successfully generated: mqtt_cert
generating mqtt_key.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/mqtt_key key /home/marco/esp/esp-at/components/customized_partitions/raw_data/mqtt_key/mqtt_client.key 
AT customized partitions successfully generated: mqtt_key
generating mqtt_ca.bin: python /home/marco/esp/esp-at/tools/AtPKI.py generate_bin -b /home/marco/esp/esp-at/build/customized_partitions/mqtt_ca ca /home/marco/esp/esp-at/components/customized_partitions/raw_data/mqtt_ca/mqtt_ca.crt 
AT customized partitions successfully generated: mqtt_ca
AT customized bin generation finished

App "esp-at" version: ESP-AT
To flash all build output, run 'make flash' or:
python /home/marco/esp/esp-at/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 /home/marco/esp/esp-at/build/ota_data_initial.bin 0x1000 /home/marco/esp/esp-at/build/bootloader/bootloader.bin 0x20000 /home/marco/esp/esp-at/build/at_customize.bin 0x21000 /home/marco/esp/esp-at/build/customized_partitions/ble_data.bin 0x24000 /home/marco/esp/esp-at/build/customized_partitions/server_cert.bin 0x26000 /home/marco/esp/esp-at/build/customized_partitions/server_key.bin 0x28000 /home/marco/esp/esp-at/build/customized_partitions/server_ca.bin 0x2a000 /home/marco/esp/esp-at/build/customized_partitions/client_cert.bin 0x2c000 /home/marco/esp/esp-at/build/customized_partitions/client_key.bin 0x2e000 /home/marco/esp/esp-at/build/customized_partitions/client_ca.bin 0x30000 /home/marco/esp/esp-at/build/customized_partitions/factory_param.bin 0x37000 /home/marco/esp/esp-at/build/customized_partitions/mqtt_cert.bin 0x39000 /home/marco/esp/esp-at/build/customized_partitions/mqtt_key.bin 0x3B000 /home/marco/esp/esp-at/build/customized_partitions/mqtt_ca.bin 0xf000 /home/marco/esp/esp-at/build/phy_init_data.bin 0x100000 /home/marco/esp/esp-at/build/esp-at.bin 0x8000 /home/marco/esp/esp-at/build/partitions_at.bin

And after the flash command, the log is:

Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00010000 in 0.0 seconds (effective 4110.1 kbit/s)...
Hash of data verified.
Flash params set to 0x0230
Compressed 25632 bytes to 15297...
Wrote 25632 bytes (15297 compressed) at 0x00001000 in 1.4 seconds (effective 150.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 236...
[... 37 lines cut ...]
Hash of data verified.
Compressed 1398288 bytes to 895084...
Wrote 1398288 bytes (895084 compressed) at 0x00100000 in 79.6 seconds (effective 140.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 762.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

This is my service configuration in the "esp/esp-at/components/customized_partitions/raw_data/ble_data/example.csv" file:

index	uuid_len	uuid	perm	val_max_len	val_cur_len	value
0	16	0x2800	0x01	2	2	A002
1	16	0x2803	0x01	1	1	2
2	16	0xC300	0x01	1	1	30
3	16	0x2901	0x11	1	1	30

The ble_data.bin has been uploaded at the address 0x21000 but the command "AT+BLEGATTSSRVCRE" returns always ERROR (I've tried to use BLEGATTSSRVCRE before than BLEADVSTART, but the result don't change).

Any idea to resolve this issue?

Thank you very much,
Marco

xiewenxiang, 2020-08-20:
@marcocossali 

HI Marco,

Please provide your `ble_data.bin`.

marcocossali, 2020-08-20:
@xiewenxiang 

Thanks for your reply.
This is the content of the my ble_data.bin (I can't provide the .bin file directly):

```
À'{"Service": [{"index": 0, "uuid_len": 16, "uuid": "2800", "perm": 1, "val_max_len": 2, "val_cur_len": 2, "value": "A002"}, {"index": 1, "uuid_len": 16, "uuid": "2803", "perm": 1, "val_max_len": 1, "val_cur_len": 1, "value": "02"}, {"index": 2, "uuid_len": 16, "uuid": "C300", "perm": 1, "val_max_len": 1, "val_cur_len": 1, "value": "30"}, {"index": 3, "uuid_len": 16, "uuid": "2803", "perm": 17, "val_max_len": 1, "val_cur_len": 1, "value": "30"}]}
```

xiewenxiang, 2020-08-20:
@marcocossali 

I just want to know the hex value for the first 8 bytes of this bin. Could you provide it?

marcocossali, 2020-08-20:
@xiewenxiang 

Yes of course.. this is the content (in hex):

[image]

xiewenxiang, 2020-08-20:
The previous bytes are wrong. Could you use python 2 to create this bin?

marcocossali, 2020-08-20:
@xiewenxiang 

To generate the ble_data.bin I use simply the **make** command (and after **make flash** to flash the module). How could I modify python from version 3 to 2?

marcocossali, 2020-08-21:
@xiewenxiang 

Good morning,
I've tried to do this (inside the *esp_at/tools* folder):
```
python2 BLEService.py example.csv
```
but the the script returns the following message:
```
Traceback (most recent call last):
  File "BLEService.py", line 29, in <module>
    import yaml
ImportError: No module named yaml
```
(if I use `python BLEService.py example.csv` the script work fine!!)

If I try to install yaml with:
```
pip install pyyaml
```
the returned message is (as expected, because yaml is already installed):
```
Requirement already satisfied: pyyaml in /home/marco/.espressif/python_env/idf4.0_py3.6_env/lib/python3.6/site-packages (5.3.1)
```

xcguang, 2020-08-21:
python2 -m pip install pyyaml

xcguang, 2020-08-24:
Now, python3 should works well on the latest release/v2.1.0.0_esp32 and master branch.

marcocossali, 2020-08-24:
@xcguang 

I updated the esp-at repo at the new master branch:
```
marco@marco-VirtualBox:~/esp/esp-at$ git log --oneline
cb285c7 (HEAD -> master, origin/master, origin/HEAD) Merge branch 'bugfix/generate_ble_data_bin_error_by_python3' into 'master'
f8f49eb fix: generate ble_data.bin error by python3
a076d8e Merge branch 'docs/missing_chip_info' into 'master'
badd3f7 docs: missing the module info in ESP32-S2_AT_binaries.md
b5e1674 (tag: v2.1.0.0_esp32s2) Merge branch 'bugfix/ota_over_ssl_fail' into 'master'
...
```
but I have some problems when i try to run the `./install.sh` command.

The log error is the following (after the `./install.sh` command execution):
```
Installing collected packages: pygdbmi
  Attempting uninstall: pygdbmi
    Found existing installation: pygdbmi 0.10.0.0b0
    Uninstalling pygdbmi-0.10.0.0b0:
      Successfully uninstalled pygdbmi-0.10.0.0b0
ERROR: After October 2020 you may experience errors when installing or updating packages. This is because pip will change the way that it resolves dependency conflicts.

We recommend you use --use-feature=2020-resolver to test your packages with the new resolver before it becomes the default.

gdbgui 0.14.0.0 requires pygdbmi<0.11,>=0.10.0.0b0, but you'll have pygdbmi 0.9.0.2 which is incompatible.
Successfully installed pygdbmi-0.9.0.2
All done! You can now run:

  . ./export.sh
```

Then when I run the `. ./export.sh` the following error is printed:
```
Adding ESP-IDF tools to PATH...
Using Python interpreter in /home/marco/.espressif/python_env/idf4.3_py3.6_env/bin/python
Checking if Python packages are up to date...
The following Python requirements are not satisfied:
pygdbmi<=0.9.0.2
To install the missing packages, please run "/home/marco/esp/esp-at/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/marco/.espressif/python_env/idf4.3_py3.6_env
    Python interpreter used: /home/marco/.espressif/python_env/idf4.3_py3.6_env/bin/python
```

I tried to install *pygdbmi* version: 0.10.0.0b0, but when the `./install.sh` command is executed the *pygdbmi* packet is reinstalled with the version 0.9.0.2.

marcocossali, 2020-08-26:
@xcguang @xiewenxiang 

any tips to resolve this issue?

FayeY, 2021-06-21:
May I ask if this issue persists with the latest master?

Akashj35, 2024-03-29:
Is there any solution of this error. I am also facing the same issue but it is not constant it is occurs sometimes & randomly after hard reboot of ESP32 module.

JHedgesBartec, 2024-08-29:
@ustccw could you please take a look or provide a response on this?

I am also seeing the command producing ERROR consistently with modified GATTS data.

I simple test to reproduce this on master is to remove configs 22 onwards. After a clean build and flash with the factory image the command errors.

ustccw, 2024-08-30:
@Luxin812 PTAL

Luxin812, 2024-08-30:
> @ustccw could you please take a look or provide a response on this?
> 
> I am also seeing the command producing ERROR consistently with modified GATTS data.
> 
> I simple test to reproduce this on master is to remove configs 22 onwards. After a clean build and flash with the factory image the command errors.
> 
> Another example gatts_data.csv which causes the error:
> 
> ```
> # index,uuid_len,uuid,perm,max_len,cur_len,value
> 0,16,0x2800,0x01,2,2,BA1D
> 1,16,0x2803,0x01,1,1,02
> 2,16,0x1001,0x10,512,512,
> 3,16,0x2803,0x01,1,1,10
> 4,16,0x1002,0x01,1,1,00
> 5,16,0x2902,0x11,2,2,0000
> 6,16,0x2803,0x01,1,1,02
> ```

Hi,

Please delete cfg21, cfg22, and cfg23, not just cfg22. You can refer to the [document](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_customize_BLE_services.html) to understand the meaning of 0x2803.

Luxin812, 2024-08-30:
Please make the changes according to the following example and then test.
# index,uuid_len,uuid,perm,max_len,cur_len,value
0,16,0x2800,0x01,2,2,BA1D
1,16,0x2803,0x01,1,1,02
2,16,0x1001,0x10,512,512,
3,16,0x2803,0x01,1,1,10
4,16,0x1002,0x01,1,1,00
5,16,0x2902,0x11,2,2,0000