espressif/esp-at #463: esp-at工程是否支持ESP-WROOM-32D 模组

YzY19930301, 2020-08-26:
我使用的模组是从小米IOT平台申请的ESP-WROOM-32D 模组，现在想把AT指令和小米结合。我在编译这个工程并且烧录到ESP-WROOM-32D后，log串口显示不停在重启设备，不知道这个工程能否支持ESP-WROOM-32D 模组

xcguang, 2020-08-27:
支持乐鑫官方 ESP-WROOM-32D 模组

YzY19930301, 2020-08-27:
> 支持乐鑫官方 ESP-WROOM-32D 模组

请问从小米平台申请的ESP-WROOM-32D模组 与 官方有什么不同，工程烧录到以前购买的旧的WROOM32 是正常运行的，只有在ESP-WROOM-32D上运行一直在rebooting，是否有什么配置项需要修改

xcguang, 2020-08-27:
具体可以从小米处获取 datasheet，并对比差异

YzY19930301, 2020-08-27:
> 具体可以从小米处获取 datasheet，并对比差异；理论可以从 rebooting 信息中，看出点信息

rebooting中有这个错误信息，不太明白是什么原因：

I (667) boot: Loaded app from partition at offset 0x100000
I (667) boot: Disabling RNG early entropy source...
E (727) system_api: Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0x9e; calc_crc = 0x44
abort() was called at PC 0x400d193b on core 0

ELF file SHA256: 09e32225ecf4e28b

Backtrace: 0x4008d09d:0x3ffcd1d0 0x4008d409:0x3ffcd1f0 0x400d193b:0x3ffcd210 0x400d198d:0x3ffcd250 0x400de853:0x3ffcd280 0x400dbe22:0x3ffcd2a0 0x400968ba:0x3ffcd2c0 0x40090011:0x3ffcd2f0

ESP-Coco, 2020-08-27:
小米的定制 MAC 在 IDF 会判定为非法 MAC，从而 abort，所以需要先把 abort()
注释掉， 然后重新编译工程。
使用 make monitor 可以自动打印 Backtrace 对应的代码，一般处于这个位置 https://github.com/espressif/esp-idf/blob/release/v3.0/components/esp32/system_api.c#L141

closed 2021-03-22 by FayeY