espressif/esp-at #464: AT+BLEGATTSSRVCRE fails when attr table is large enough

HectorJimenezMendez, 2020-08-26:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    v4.0.1-193-ge7ac221b4
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
When creating GATTs Services via AT command (AT+BLEGATTSSRVCRE) I get "ERROR" in serial port. 

It happens when the service table contains at least 25 characteristics (no attributes) and it works when working with 24 charcateristics (needed hex edition at the header of ble_data.bin file to succesfully create services).

//Detailed problem description goes here.
I'm able to make the ESP32 module create GATTs services and characteristics using the following .csv file:
`index,uuid_len,uuid,perm,val_max_len,val_cur_len,value
0,16,0x2800,0x01,2,2,180a
1,16,0x2803,0x02,1,1,02
2,16,0x2a29,0x02,22,22,5a49562d4d45544552494e472d534f4c5554494f4e53
3,16,0x2803,0x02,1,1,02
4,16,0x2a24,0x02,5,5,57414c4c45
5,16,0x2803,0x02,1,1,02
6,16,0x2a25,0x02,10,10,30313233343536373839
7,16,0x2803,0x02,1,1,02
8,16,0x2a27,0x02,20,3,302e31
9,16,0x2803,0x02,1,1,02
10,16,0x2a26,0x02,20,3,302e31
11,16,0x2800,0x01,16,16,97f934e0985211e5887e0002a5d5c51b
15,16,0x2803,0x22,1,1,0a
16,128,0x9921086099d911e5863f0002a5d5c51b,0x22,1,1,30
18,16,0x2803,0x22,1,1,0a
19,128,0x658dabbcc56f33e67a7acec0c0804e21,0x22,1,1,30
24,16,0x2803,0x22,1,1,0a
25,128,0xc81dbb00985211e58b310002a5d5c51b,0x22,1,1,30
27,16,0x2803,0x22,1,1,0a
28,128,0xec0add08c79611e69d9dcec0c932ce01,0x22,1,1,30
30,16,0x2803,0x22,1,1,0a
31,128,0x827210c099d711e580dd0002a5d5c51b,0x22,1,1,30
32,16,0x2803,0x22,1,1,0a
33,128,0x590c4bd9b5e24a1a867e0002a5d5c51b,0x22,1,1,30
34,16,0x2803,0x22,1,1,0a
35,128,0xdee4f24099d811e5a1f20002a5d5c51b,0x22,18,1,30
36,16,0x2803,0x22,1,1,0a
37,128,0xae26cec099d911e582cd0002a5d5c51b,0x22,18,1,30
39,16,0x2803,0x10,1,1,08
40,128,0xf7c97b32b29411e896f8529269fb1459,0x10,1,1,31
42,16,0x2803,0x10,1,1,08
43,128,0xc8b7d0d047ac11e8842f0ed5f89f718b,0x10,1,1,30
57,16,0x2803,0x20,1,1,08
58,128,0x6d27ade0d1ae4dcbbf43c2a7d4829c19,0x20,16,16,30313233343536373839303132333435
60,16,0x2803,0x20,1,1,08
61,128,0x59faedd8270d46fea446d24cc099fe1f,0x20,16,16,30313233343536373839303132333435
63,16,0x2803,0x10,1,1,08
64,128,0x44fa6b43fc7e477490d03dc6d6ff61bc,0x10,10,1,30
66,16,0x2803,0x01,1,1,02
67,128,0xad54f949900f496fb6c43ab3520645a2,0x01,512,1,30
69,16,0x2803,0x02,1,1,02
70,128,0x12914eaa2a9843a29a8f1925d6a46a7a,0x02,512,1,30
72,16,0x2803,0x22,1,1,0a
73,128,0xc7dfc45050794e3b8f589fbfee1cdd89,0x22,16,1,30
75,16,0x2803,0x22,1,1,0a
76,128,0xcb13535865974631a03c5b929faf7e32,0x22,16,1,30
78,16,0x2800,0x01,16,16,f04254a099d911e5bc0c0002a5d5c51b
79,16,0x2803,0x01,1,1,08
80,128,0x2301a9e099da11e5ac080002a5d5c51b,0x10,1,1,30
82,16,0x2803,0x02,1,1,02
83,128,0x48860b6099db11e5ae840002a5d5c51b,0x02,64,11,412c322c302c302c302c30
,,,,,,

`

Note the index numbers are not in a sequence but it seems not to be important for BLEService.py tool as long as it creates a workable binary file.

But when I add another characteristic, then the module answers error when it tries to create the GATTs services:

`AT+BLEGATTSSRVCRE

ERROR
`

.csv file which causes the error:

`
index,uuid_len,uuid,perm,val_max_len,val_cur_len,value
0,16,0x2800,0x01,2,2,180a
1,16,0x2803,0x02,1,1,02
2,16,0x2a29,0x02,22,22,5a49562d4d45544552494e472d534f4c5554494f4e53
3,16,0x2803,0x02,1,1,02
4,16,0x2a24,0x02,5,5,57414c4c45
5,16,0x2803,0x02,1,1,02
6,16,0x2a25,0x02,10,10,30313233343536373839
7,16,0x2803,0x02,1,1,02
8,16,0x2a27,0x02,20,3,302e31
9,16,0x2803,0x02,1,1,02
10,16,0x2a26,0x02,20,3,302e31
11,16,0x2800,0x01,16,16,97f934e0985211e5887e0002a5d5c51b
15,16,0x2803,0x22,1,1,0a
16,128,0x9921086099d911e5863f0002a5d5c51b,0x22,1,1,30
18,16,0x2803,0x22,1,1,0a
19,128,0x658dabbcc56f33e67a7acec0c0804e21,0x22,1,1,30
24,16,0x2803,0x22,1,1,0a
25,128,0xc81dbb00985211e58b310002a5d5c51b,0x22,1,1,30
27,16,0x2803,0x22,1,1,0a
28,128,0xec0add08c79611e69d9dcec0c932ce01,0x22,1,1,30
30,16,0x2803,0x22,1,1,0a
31,128,0x827210c099d711e580dd0002a5d5c51b,0x22,1,1,30
32,16,0x2803,0x22,1,1,0a
33,128,0x590c4bd9b5e24a1a867e0002a5d5c51b,0x22,1,1,30
34,16,0x2803,0x22,1,1,0a
35,128,0xdee4f24099d811e5a1f20002a5d5c51b,0x22,18,1,30
36,16,0x2803,0x22,1,1,0a
37,128,0xae26cec099d911e582cd0002a5d5c51b,0x22,18,1,30
39,16,0x2803,0x10,1,1,08
40,128,0xf7c97b32b29411e896f8529269fb1459,0x10,1,1,31
42,16,0x2803,0x10,1,1,08
43,128,0xc8b7d0d047ac11e8842f0ed5f89f718b,0x10,1,1,30
45,16,0x2803,0x10,1,1,08
46,128,0xa4ae2fa099d911e5ac220002a5d5c51b,0x10,1,1,30
57,16,0x2803,0x20,1,1,08
58,128,0x6d27ade0d1ae4dcbbf43c2a7d4829c19,0x20,16,16,30313233343536373839303132333435
60,16,0x2803,0x20,1,1,08
61,128,0x59faedd8270d46fea446d24cc099fe1f,0x20,16,16,30313233343536373839303132333435
63,16,0x2803,0x10,1,1,08
64,128,0x44fa6b43fc7e477490d03dc6d6ff61bc,0x10,10,1,30
66,16,0x2803,0x01,1,1,02
67,128,0xad54f949900f496fb6c43ab3520645a2,0x01,512,1,30
69,16,0x2803,0x02,1,1,02
70,128,0x12914eaa2a9843a29a8f1925d6a46a7a,0x02,512,1,30
72,16,0x2803,0x22,1,1,0a
73,128,0xc7dfc45050794e3b8f589fbfee1cdd89,0x22,16,1,30
75,16,0x2803,0x22,1,1,0a
76,128,0xcb13535865974631a03c5b929faf7e32,0x22,16,1,30
78,16,0x2800,0x01,16,16,f04254a099d911e5bc0c0002a5d5c51b
79,16,0x2803,0x01,1,1,08
80,128,0x2301a9e099da11e5ac080002a5d5c51b,0x10,1,1,30
82,16,0x2803,0x02,1,1,02
83,128,0x48860b6099db11e5ae840002a5d5c51b,0x02,64,11,412c322c302c302c302c30
,,,,,,

`

### Expected Behavior
ESP32 module should create services 

### Actual Behavior
ESP32 module fails at creating Services

### Steps to repropduce
1. Generate bin file using BLEService.py tool and source file with 25 characteristics at least. hex edition might be needed.
2. flash generated ble_data.bin file into ble_data partition (0x21000 in my system)
3. Create GATTS services after init BLE:
AT+BLEINIT
AT+BLEGATTSSRVCRE
4.- Watch serial output to see the ERROR message

xiewenxiang, 2020-08-31:
I have test with your .csv file, it works fine.
```
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRV?
+BLEGATTSSRV:1,0,0x180A,1
+BLEGATTSSRV:2,0,0x97F934E0985211E5887E0002A5D5C51B,1
+BLEGATTSSRV:3,0,0xF04254A099D911E5BC0C0002A5D5C51B,1

OK
```

Please provide your `ble_data.bin`

xiewenxiang, 2020-08-31:
[ble_data.bin.zip](https://github.com/espressif/esp-at/files/5150189/ble_data.bin.zip)

xiewenxiang, 2020-08-31:
My guess is that the generated bin format is different because of a Python environment.

HectorJimenezMendez, 2020-08-31:
Hi, @xiewenxiang. 

dumping your ble_data.bin file to see the differences to our ble_data.bin file, I've noticed that you have generated that file with the source file that works also to us (first code). 

Characteristic with index 45,46 is not present in your file:

45,16,0x2803,0x10,1,1,08
46,128,0xa4ae2fa099d911e5ac220002a5d5c51b,0x10,1,1,30

This characteristic is the only difference between our working file and the one which doesn't create services.

Can you, please, try to generate ble_data.bin file from second part of code which includes 45,46 index characteristic?

csv file which fails: 
[ble_data_FAILS.csv.zip](https://github.com/espressif/esp-at/files/5150424/ble_data_FAILS.csv.zip)

On the other hand, your guess about python environment seems pretty accurate cause I cannot work with the bin file generated by BLEService.py tool. I must change the header of the file using a hex editor to make it work.
The generated file header looks like:
00000000  c3 b3 18 27 c2 95 7b 22  53 65 72 76 69 63 65 22  |...'..{"Service"|

Then I change it into: 
00000000  f3 18 27 95 7b 22 53 65  72 76 69 63 65 22 3a 20  |..'.{"Service": |

or to:
00000000 53 19 27 95 7b 22 53 65 72 76 69 63 65 22 3a 20 |..'.{"Service": |

and it works, but it doesn't work with the longer file (the one attached to this comment).

Can you, please try to generate the bin file using attached .csv file (ble_data_FAILS.csv)?

Thank you!

HectorJimenezMendez, 2020-08-31:
Hi, @xiewenxiang .

This is the bin file generated by BLSService.py tool: 
[generated_ble_data_FAILS.bin.zip](https://github.com/espressif/esp-at/files/5150552/generated_ble_data_FAILS.bin.zip)

And this is the file with the header changes (but still doesn't work):
[modified_ble_data_FAILS.bin.zip](https://github.com/espressif/esp-at/files/5150556/modified_ble_data_FAILS.bin.zip)

None of this .bin files are able to create gatts services, but they might be helpful for you to understand what the problem is

Thank you and kind regards

xiewenxiang, 2020-09-02:
Hi, @HectorJimenezMendez 

I test the source .csv file(second).

```
AT+BLEINIT=2

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRV?
+BLEGATTSSRV:1,0,0x180A,1
+BLEGATTSSRV:2,0,0x97F934E0985211E5887E0002A5D5C51B,1
+BLEGATTSSRV:3,0,0xF04254A099D911E5BC0C0002A5D5C51B,1

OK
AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0x2A29,0x02
+BLEGATTSCHAR:"char",1,2,0x2A24,0x02
+BLEGATTSCHAR:"char",1,3,0x2A25,0x02
+BLEGATTSCHAR:"char",1,4,0x2A27,0x02
+BLEGATTSCHAR:"char",1,5,0x2A26,0x02
+BLEGATTSCHAR:"char",2,1,0x9921086099D911E5863F0002A5D5C51B,0x0a
+BLEGATTSCHAR:"char",2,2,0x658DABBCC56F33E67A7ACEC0C0804E21,0x0a
+BLEGATTSCHAR:"char",2,3,0xC81DBB00985211E58B310002A5D5C51B,0x0a
+BLEGATTSCHAR:"char",2,4,0xEC0ADD08C79611E69D9DCEC0C932CE01,0x0a
+BLEGATTSCHAR:"char",2,5,0x827210C099D711E580DD0002A5D5C51B,0x0a
+BLEGATTSCHAR:"char",2,6,0x590C4BD9B5E24A1A867E0002A5D5C51B,0x0a
+BLEGATTSCHAR:"char",2,7,0xDEE4F24099D811E5A1F20002A5D5C51B,0x0a
+BLEGATTSCHAR:"char",2,8,0xAE26CEC099D911E582CD0002A5D5C51B,0x0a
+BLEGATTSCHAR:"char",2,9,0xF7C97B32B29411E896F8529269FB1459,0x08
+BLEGATTSCHAR:"char",2,10,0xC8B7D0D047AC11E8842F0ED5F89F718B,0x08
+BLEGATTSCHAR:"char",2,11,0xA4AE2FA099D911E5AC220002A5D5C51B,0x08
+BLEGATTSCHAR:"char",2,12,0x6D27ADE0D1AE4DCBBF43C2A7D4829C19,0x08
+BLEGATTSCHAR:"char",2,13,0x59FAEDD8270D46FEA446D24CC099FE1F,0x08
+BLEGATTSCHAR:"char",2,14,0x44FA6B43FC7E477490D03DC6D6FF61BC,0x08
+BLEGATTSCHAR:"char",2,15,0xAD54F949900F496FB6C43AB3520645A2,0x02
+BLEGATTSCHAR:"char",2,16,0x12914EAA2A9843A29A8F1925D6A46A7A,0x02
+BLEGATTSCHAR:"char",2,17,0xC7DFC45050794E3B8F589FBFEE1CDD89,0x0a
+BLEGATTSCHAR:"char",2,18,0xCB13535865974631A03C5B929FAF7E32,0x0a
+BLEGATTSCHAR:"char",3,1,0x2301A9E099DA11E5AC080002A5D5C51B,0x08
+BLEGATTSCHAR:"char",3,2,0x48860B6099DB11E5AE840002A5D5C51B,0x02

OK
```

xiewenxiang, 2020-09-02:
[ble_data.bin.zip](https://github.com/espressif/esp-at/files/5159532/ble_data.bin.zip)

xiewenxiang, 2020-09-02:
[image]

xiewenxiang, 2020-09-02:
[image]

xiewenxiang, 2020-09-02:
@HectorJimenezMendez 

For now, that's the difference between the old and new Python versions.

xcguang, 2020-09-02:
You can try the latest master branch, it should work well.

HectorJimenezMendez, 2020-09-02:
Hi, @xiewenxiang.

I tried to manually change the header of our ble_data.bin generated file (magic number and length) to the values present in your posted file and it worked, ESP32 was able to create GATT services. Thank you very much for your help.

@xcguang, I downloaded the latest master branch and tried to generate another ble_data.bin file with some more characteristics and descriptors (87 seven lines in .csv file). BLEService.py tool generated a working file which i didn't need to change the header. So thank you as well, now we are working with all our desired services without any problem.

In my opinion, this issue is resolved and it can be closed.

Thank you both for your support and kind regards

closed 2020-09-02 by xcguang