espressif/esp-at #465: AT+CIPCLOSE uses RST instead of FIN

Maintraxx, 2020-08-27:
Because the firmware (AT 1.7.4) contains so many annoying errors, I wanted to try this firmware.

I have implemented a webserver on my host system, which closes the connection with Close after completion. 
Unfortunately this does not work anymore, because the browser tells me that the connection was closed with a RST instead of a FIN.

I close the connection this way:

`  35.322 [ESP8266 ] [RX] DATA[+SYSRAM:45588,39128]
  35.323 [ESP8266 ] [RX] DATA[OK]
  35.421 [ESP8266 ] [RX] DATA[+LINK_CONN:0,0,"TCP",1,"192.168.2.10",59657,80]
  35.430 [ESP8266 ] AT CMD[AT+CIPSEND=0,145] STATEOK[0x80] STATEERR[0x84] TIMEOUT[15000ms]
  35.432 [ESP8266 ] [TX] LEN[18]
  35.435 [ESP8266 ] [RX] DATA[OK]
  35.436 [ESP8266 ] [RX] DATA[>]
  35.438 [ESP8266 ] [TX] LEN[145]
  35.442 [ESP8266 ] [RX] DATA[Recv 145 bytes]
  35.450 [ESP8266 ] [RX] DATA[SEND OK]
  35.454 [ESP8266 ] AT CMD[AT+CIPSEND=0,289] STATEOK[0x80] STATEERR[0x84] TIMEOUT[15000ms]
  35.455 [ESP8266 ] [TX] LEN[18]
  35.458 [ESP8266 ] [RX] DATA[OK]
  35.459 [ESP8266 ] [RX] DATA[>]
  35.461 [ESP8266 ] [TX] LEN[289]
  35.466 [ESP8266 ] [RX] DATA[Recv 289 bytes]
  35.473 [ESP8266 ] [RX] DATA[SEND OK]
  35.476 [ESP8266 ] AT CMD[AT+CIPCLOSE=0] STATEOK[0x70] STATEERR[0x70] TIMEOUT[2500ms]
  35.478 [ESP8266 ] [TX] LEN[15]
  35.491 [ESP8266 ] [RX] DATA[0,CLOSED]
  35.494 [ESP8266 ] [RX] DATA[OK]
`

With the old V1.7.4 At Firmware this works.

ustccw, 2020-08-28:
@Maintraxx Could you please providing the tcp packets and AT log, AT commands you input, version by `AT+GMR`.

Maintraxx, 2020-08-28:
Yes, here is a real example:

**Version**
AT+GMR
AT version:2.2.0.0-dev(7963ea8 - ESP8266 - Aug 25 2020 07:31:48)
SDK version:v3.3-2-gb6e861a7
compile time(cf63f23):Aug 28 2020 11:08:11
Bin version:2.0.0(Unknown)_

**AT-Log**
```
+LINK_CONN:0,0,"TCP",1,"192.168.2.10",58214,80
+IPD:0,534
AT+CIPRECVDATA=0,534
+CIPRECVDATA,534:"payload"
OK
AT+CIPSEND=0,145
OK
>
"payload"
Recv 145 bytes
SEND OK
AT+CIPSEND=0,256
OK
>
"payload"
Recv 256 bytes
SEND OK
AT+CIPSEND=0,35
OK
>
"payload"
Recv 35 bytes
SEND OK
AT+CIPCLOSE=0
0,CLOSED
OK

```

**Transmitted Payload**
[image]

**TCP Packets**
[image]

**RST Packet**
[image]

Maintraxx, 2020-08-31:
Additional information: It seems, that this only occures for small packages. (~ < 512Byte). If i transfer larger sites, everythink works as expected.

ustccw, 2020-08-31:
will take a look this week

ustccw, 2020-09-07:
@Maintraxx I tried several time, i am afraid that can not reproduce it.

@SUNLUXIN Could you please help to reproduce it.

Maintraxx, 2020-09-07:
Maybe it depends from "SO_LINGER"?
https://github.com/espressif/esp-at/blob/master/docs/en/AT_Command_Set/TCP-IP_AT_Commands.md#cmd-TCPOPT

I tried several settings, but the result is always the same.

jandrassy, 2020-09-07:
the AT firmware didn't get ACK for the last packet so it issued a RST

ustccw, 2020-10-09:
`SO_LINGER` is off by default. It will not close the connection use RST instead of FIN.

have you set this configuration?

Maintraxx, 2020-10-09:
No, i don't use it. I made some test with the linger option, but it makes no differences regarding this behaviour.

ustccw, 2020-10-09:
what is the esp-at firmware version, have you tried the latest version?

tested with the following version, and i am sorry that can not reproduce it.
```
AT+GMR
AT version:2.2.0.0-dev(ca41ec4 - ESP32 - Sep 16 2020 11:27:04)
SDK version:v4.0.1-193-ge7ac221b4
compile time(ab79540):Oct  9 2020 10:28:49
Bin version:2.1.0(WROOM-32)

OK
```

Maintraxx, 2020-10-09:
I'm working on a ESP8286. The version is:
`AT+GMR
AT version:2.2.0.0-dev(7963ea8 - ESP8266 - Aug 25 2020 07:31:48)
SDK version:v3.3-2-gb6e861a7
compile time(cf63f23):Aug 28 2020 11:08:11
Bin version:2.0.0(Unknown)`

ustccw, 2020-10-09:
I tested the ESP8266 firmware, it looks like work well.
```
ready
AT+CWMODE=1

OK

AT+CWJAP="huawei_chenwu",""
WIFI CONNECTED
WIFI GOT IP

OK
AT+GMR
AT version:2.2.0.0-dev(ca41ec4 - ESP8266 - Sep 16 2020 10:34:08)
SDK version:v3.3-25-g5c773fc8
compile time(ab79540):Oct  9 2020 19:54:17
Bin version:2.0.0(WROOM-02)

OK
AT+CIPMUX=1

OK
AT+CIPMODE=1

ERROR
AT+CIPRECVMODE=1

OK
AT+CIPSTA?
+CIPSTA:ip:"192.168.110.238"
+CIPSTA:gateway:"192.168.110.1"
+CIPSTA:netmask:"255.255.255.0"

OK
AT+CIPSTART=0,"TCP","192.168.110.6",8080
0,CONNECT

OK

+IPD,0,5
AT+CIPRECVDATA?

ERROR
AT+CIPRECVDATA=0,5
+CIPRECVDATA:5,hello
OK

+IPD,0,549
AT+CIPRECVDATA=0,549
+CIPRECVDATA:549,
...
OK
AT+CIPSEND=0,145

OK

>

Recv 145 bytes

SEND OK
AT+CIPSEND=0,256

OK

>

Recv 256 bytes

SEND OK
AT+CIPSEND=0,35

OK

>

Recv 35 bytes

SEND OK
AT+CIPCLOSE=0
0,CLOSED

OK
AT+GMR
AT version:2.2.0.0-dev(ca41ec4 - ESP8266 - Sep 16 2020 10:34:08)
SDK version:v3.3-25-g5c773fc8
compile time(ab79540):Oct  9 2020 19:54:17
Bin version:2.0.0(WROOM-02)

OK
AT+SYSRAM?
+SYSRAM:46324,39748
OK

```

ustccw, 2020-10-09:
Could you please try with this firmware: http://download.espressif.com/esp_share/github/esp-at/issues/target.bin, which i test ok.

Maintraxx, 2020-10-11:
The size of your Bin-File is 2MB, but my ESP8285 only has 1MB capacity.

ustccw, 2020-10-12:
@Maintraxx Please try with the firmware here: https://github.com/espressif/esp-at/suites/1256130336/artifacts/19184934

Maintraxx, 2020-10-12:
Thank you.
With this version, the uart did not work as expected. After the boot, the ESP-Tx-Line goes low and no communication is possible:
[image]
[image]
This problem has been reported in #445

Maintraxx, 2021-06-10:
Problems seems to be gone with the actual version.

closed 2021-06-10 by Maintraxx