espressif/esp-at #466: ESP32-S2-WROOM UART1 & UART0 pins definition

LlamSulfate, 2020-08-28:
Hi all,

I have a question regarding the UART1 and UART0 pin defitions mentioned in the ESP32S2 AT documentation.

On [this documentation](https://github.com/espressif/esp-at/blob/master/docs/en/Get_Started/Hardware_connection.md) i see that U1TX is on GPIO17, U1RTS is on GPIO19, U1CTS is on GPIO20, U1RX is on GPIO21, U0TX is on GPIO43 and U0RX is on GPIO44.

However on the ESP32-S2-WROOM [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf) i see that U1RX is mapped only on GPIO18 instead.

Since i still don't have an ESP32-S2-WROOM in my hands to check it myself, which is the right GPIO for U1RX? is the ESP32S2 AT documentation wrong?

Additionally, in [this part of the documentation](https://github.com/espressif/esp-at/blob/master/docs/en/Compile_and_Develop/How_to_clone_project_and_compile_it.rst#esp32s2-platform) it is mentioned that U0TX is mapped on GPIO1 and U0RX on GPIO3. Is this a possible error in the documentation?

Thanks in advance for your answers.

ustccw, 2020-08-31:
@LlamSulfate 
1. ESP32-S2-WROOM UART0 GPIOs should be GPIO43 for U0TXD and GPIO44 for U0RXD.
It's really a mistake in our documentation, will fix it, thank you.

2. ESP32-S2 has a GPIO matrix, U1RXD can be set to GPIO21, other than GPIO18 by default.
Please refer to [esp32-s2 technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf).

LlamSulfate, 2020-08-31:
> @LlamSulfate
>
> 1. ESP32-S2-WROOM UART0 GPIOs should be GPIO43 for U0TXD and GPIO44 for U0RXD.
>    It's really a mistake in our documentation, will fix it, thank you.
> 2. ESP32-S2 has a GPIO matrix, [...]

Thank you very much for your reply, i missed that part of the technical reference.

In this case, does the bootloader use this configuration of U1RXD as well or the default pin for U1RXD?

ustccw, 2020-09-01:
bootloader keeps the raw configuration of UART. 
`esp-at` changes UART1 RXD pin in the begin of the application code, you can find it  [here](https://github.com/espressif/esp-at/blob/master/main/interface/uart/at_uart_task.c#L371).

The reason for the change is that `GPIO18` pin has other special uses, it used to control the light in some `DevKits` and control the download mode at the same time.

LlamSulfate, 2020-09-01:
Alright, thanks for the answer. I think the issue can be closed now

closed 2020-09-01 by LlamSulfate