espressif/esp-at #468: branch 'master' compile problem

Takeya-Yuki, 2020-08-30:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ v] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ v] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ v] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above ## Environment

- Development Kit:      [ESP32S2-WROOM|ESP32S2-WROVER]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32S2-WROOM|ESP32S2-WROVER]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.
modify makefile to PLATFORM_ESP32S2
make menuconfig
cat sdkconfig|grep IDF_TARGET

### Expected Behavior
IDF_TARGET=esp32s2

### Actual Behavior
IDF_TARGET=esp32

### Steps to repropduce

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Takeya-Yuki, 2020-08-30:
And All the docs/*.md documents comes "HTTP 404 Not Found" in master branch

xcguang, 2020-08-30:
1. the latest docs can be accessed from https://docs.espressif.com/projects/esp-at/en/latest/
2. ./build.py build to compile the project on the latest master.

Takeya-Yuki, 2020-09-01:
build success via build.py HEAD master , but flash failed by idf.py

[image]
[image]

xcguang, 2020-09-02:
Can you try the example in the idf or change download baud rate to 115200? It seems there is something wrong with UART communication .

closed 2021-03-22 by FayeY