espressif/esp-at #471: LAN8720 and AT ETH command

pixsysedp, 2020-09-14:
I compiled firmware from https://github.com/espressif/esp-at/commit/fb54a93f52d27808baaa4ca166d4dfc76261977c to implement ETH support on my custom board ESP32 based.

I already tested the HW with an Arduino firmware and hardware is on, the Ethernet PHY work correctly.
My board is based on Olimex ESP32-EVB ( https://www.olimex.com/Products/IoT/ESP32/ESP32-EVB/open-source-hardware ) if schematic is necessary.

I enabled LAN8720 ( CONFIG_PHY_LAN8720=y ) ETH Support and a Debug Log Level:

`
CONFIG_PHY_LAN8720=y
CONFIG_ETH_USE_ESP32_EMAC=y
CONFIG_ETH_PHY_INTERFACE_RMII=y
CONFIG_ETH_RMII_CLK_INPUT=y
CONFIG_ETH_RMII_CLK_IN_GPIO=0
CONFIG_ETH_DMA_BUFFER_SIZE=512
CONFIG_ETH_DMA_RX_BUFFER_NUM=10
CONFIG_ETH_DMA_TX_BUFFER_NUM=10
CONFIG_AT_ETHERNET_SUPPORT=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
`

After boot standard AT commands reply correctly, ETH related commands respond with ERROR.
Someone can help me to investigate problem?

this is my BOOT and reply to eth command:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1284
load:0x40078000,len:12560
load:0x40080400,len:3496
entry 0x4008063c
[0;32mI (586) cpu_start: Pro cpu up.[0m
[0;32mI (586) cpu_start: Application information:[0m
[0;32mI (586) cpu_start: Project name:     esp-at[0m
[0;32mI (588) cpu_start: App version:      ESP-AT[0m
[0;32mI (593) cpu_start: Compile time:     Sep 14 2020 10:32:06[0m
[0;32mI (599) cpu_start: ELF file SHA256:  10f48307d87cd829...[0m
[0;32mI (605) cpu_start: ESP-IDF:          v4.0.1-193-ge7ac221b4[0m
[0;32mI (611) cpu_start: Single core mode[0m
D (616) memory_layout: Checking 9 reserved memory ranges:[0m
D (621) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0[0m
D (627) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10[0m
D (634) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388[0m
D (640) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20[0m
D (647) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c[0m
D (653) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffcac28[0m
D (659) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440[0m
D (666) memory_layout: Reserved memory range 0x40070000 - 0x40078000[0m
D (672) memory_layout: Reserved memory range 0x40080000 - 0x4009cfe4[0m
D (679) memory_layout: Building list of available memory regions:[0m
D (685) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000[0m
D (691) memory_layout: Available memory region 0x3ffb6388 - 0x3ffb8000[0m
D (698) memory_layout: Available memory region 0x3ffb9a20 - 0x3ffbdb28[0m
D (705) memory_layout: Available memory region 0x3ffbdb5c - 0x3ffbdb60[0m
D (711) memory_layout: Available memory region 0x3ffcac28 - 0x3ffcc000[0m
D (718) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000[0m
D (724) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000[0m
D (731) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000[0m
D (738) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000[0m
D (744) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000[0m
D (751) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000[0m
D (757) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000[0m
D (764) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000[0m
D (771) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000[0m
D (777) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000[0m
[... 237 lines cut ...]

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

pixsysedp, 2020-09-14:
This is my sdkconfig file.
[sdkconfig.zip](https://github.com/espressif/esp-at/files/5219164/sdkconfig.zip)

xcguang, 2020-10-12:
@ustccw Please help to confirm this

katodo, 2020-11-30:
is there any news?
In boot sequence could I see any messages about LAN8720?

iw2lsi, 2020-11-30:
Hello katodo,

  for me is working...

AT+CIPETHMAC?
+CIPETHMAC:"1c:5e:09:xx:xx:xx"

during the boot phase, I see:

I (2398) tcpip_adapter: sta ip: 192.168.1.159, mask: 255.255.255.0, gw: 192.168.1.254

I'm working on OLIMEX ESP32-GATEWAY w/ idf4

   Giampaolo

katodo, 2020-12-15:
can you share your SDK configuration with me?

iw2lsi, 2020-12-15:
Hi,

I've just emailed it to you...

katodo, 2020-12-15:
Tnx iw2lsi !

In the afternoon I started with a fresh build:
- rm -rf ~/esp-at
- git clone --recursive https://github.com/espressif/esp-at.git
- cd esp-at
- I changed factory_param_data.csv to :
  PLATFORM_ESP32,WROOM-32,,0xfcfc,2,1,78,1,1,13,CN,115200,1,3,-1,-1,-1,-1
 
- ./build.py menuconfig
> Platform name:  PLATFORM_ESP32 
> Module name: WROOM-32
> Enable silence mode to remove some logs and reduce the firmware size? Yes

In menuconfig menu I changed:
- Bootloader config  -> Bootloader log verbosity (Error)
- Bootloader log verbosity (Error) -> Bootloader log verbosity (Error)
- Optimization Level (Release (-Os))
- Component config > Ethernet > Support ESP32 internal EMAC controller -> PHY interface (Reduced Media Independent Interface (RMII))
- Component config > Ethernet > Support ESP32 internal EMAC controller -> RMII clock mode (Input RMII clock from external)
- Component config > AT -> AT ethernet support -> Ethernet PHY (Microchip LAN8720 PHY)
- Removed "AT OTA command support" to save flash space.

At exit from menuconfig:
- ./build.py build

Buid hang with this output:

`
/home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ../components/at/lib/libesp32_at_core_silence.a(at_ethCmd.c.obj):(.literal.esp_at_eth_cmd_regist+0xc): undefined reference to `esp_at_get_eth_default_config'
/home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ../components/at/lib/libesp32_at_core_silence.a(at_ethCmd.c.obj): in function `esp_at_eth_cmd_regist':
(.text.esp_at_eth_cmd_regist+0x5): undefined reference to `esp_at_get_eth_default_config'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
idf.py build ret: 2
idf.py build failed
`

[complete output is here](https://pastebin.com/85q9AQbB)

katodo, 2020-12-15:
I removed esp-at and xtensa-esp32-elf and started with another fresh build with attached config.
[sdkconfig.zip](https://github.com/espressif/esp-at/files/5696943/sdkconfig.zip)

If I enable  **Component config -> AT ->  AT ethernet support** command "./build.py build" fail with this error:

'(.text.esp_at_eth_cmd_regist+0x5): undefined reference to `esp_at_get_eth_default_config'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
idf.py build ret: 2
idf.py build failed'

but with normal make build is ok:
CC build/xtensa/eri.o
CC build/xtensa/debug_helpers.o
CC build/xtensa/trax.o
AS build/xtensa/debug_helpers_asm.o
AR build/xtensa/libxtensa.a
Generating esp32.project.ld
LD build/esp-at.elf
esptool.py v2.8
CC phy_init_data.o
BIN phy_init_data.bin
To flash all build output, run 'make flash' or:
python /home/ut1/esp/esp-at/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 /home/ut1/esp/esp-at/build/ota_data_initial.bin 0x1000 /home/ut1/esp/esp-at/build/bootloader/bootloader.bin 0x20000 /home/ut1/esp/esp-at/build/at_customize.bin 0x21000 /home/ut1/esp/esp-at/build/customized_partitions/ble_data.bin 0x24000 /home/ut1/esp/esp-at/build/customized_partitions/server_cert.bin 0x26000 /home/ut1/esp/esp-at/build/customized_partitions/server_key.bin 0x28000 /home/ut1/esp/esp-at/build/customized_partitions/server_ca.bin 0x2a000 /home/ut1/esp/esp-at/build/customized_partitions/client_cert.bin 0x2c000 /home/ut1/esp/esp-at/build/customized_partitions/client_key.bin 0x2e000 /home/ut1/esp/esp-at/build/customized_partitions/client_ca.bin 0x30000 /home/ut1/esp/esp-at/build/customized_partitions/factory_param.bin 0x37000 /home/ut1/esp/esp-at/build/customized_partitions/mqtt_cert.bin 0x39000 /home/ut1/esp/esp-at/build/customized_partitions/mqtt_key.bin 0x3B000 /home/ut1/esp/esp-at/build/customized_partitions/mqtt_ca.bin 0xf000 /home/ut1/esp/esp-at/build/phy_init_data.bin 0x100000 /home/ut1/esp/esp-at/build/esp-at.bin 0x8000 /home/ut1/esp/esp-at/build/partitions_at.bin

iw2lsi, 2020-12-15:
Hi,

   are you getting warnings regarding the esp-idf banch ?

         Giampaolo

katodo, 2020-12-15:
yes... '( 

this is my log:
 
_Makefile:28: There is no module_config/module_wroom-32,Using module_config/module_esp32_default
Toolchain path: /home/ut1/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp-2020r3
Expected to see version: esp-2019r2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.4.0
Expected to see version(s): 8.2.0_

**Compiler version is not supported: 8.4.0** -> this was my last attempt.
**Toolchain version is not supported: esp-2020r3** -> the toolchain is downloaded automatically with the build.

iw2lsi, 2020-12-15:
ok... let esp-at download the correct esp-idf branch into the esp-at
folder... and set the IDF_PATH variable accordingly.

This should fix the warning and hopefully you'll be able to compile...

     Giampaolo

katodo, 2020-12-16:
I do it, but with normal "make" command I get this error with automatic downloaded esp-idf:
WARNING: Toolchain version is not supported: esp-2020r1
Expected to see version: esp-2019r2

katodo, 2020-12-16:
someone can try to build attached configuration?
[sdkconfig_20201216_1527.zip](https://github.com/espressif/esp-at/files/5703279/sdkconfig_20201216_1527.zip)

I complete removed any esp/idf file in my  system.
I downloaded a fresh git with **git clone --recursive https://github.com/espressif/esp-at.git**.
I changed  second raw of **components/customized_partitions/raw_data/factory_param/factory_param_data.csv**  to:
`PLATFORM_ESP32,WROOM-32,,0xfcfc,2,1,78,1,1,13,CN,115200,1,3,-1,-1,-1,-1`

esp-idf is automatically downloaded with "./build.py menuconfig"... during menuconfig I generated attached sdkconfig file.

During build I retrieve this error:

```
[2/8] Performing build step for 'bootloader'
ninja: no work to do.
[3/6] Linking CXX executable esp-at.elf
FAILED: esp-at.elf 
: && /home/ut1/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/esp-at.elf.dir/project_elf_src.c.obj  -o esp-at.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/bt/libbt.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/main/libmain.a esp-idf/spi_driver/libspi_driver.a -Wl,--cref -Wl,--Map=/home/ut1/esp/esp-at/build/esp-at.map -fno-rtti -fno-lto esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/bt/libbt.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/protocomm/libprotocomm.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/protobuf-c/libprotobuf-c.a ../components/at/lib/libesp32_at_core_silence.a ../components/at/lib/libesp32_at_core_silence.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/bt/libbt.a -L/home/ut1/esp/esp-at/esp-idf/components/bt/controller/lib -lbtdm_app esp-idf/mdns/libmdns.a esp-idf/console/libconsole.a esp-idf/json/libjson.a esp-idf/mqtt/libmqtt.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp-tls/libesp-tls.a esp-idf/nghttp/libnghttp.a esp-idf/spiffs/libspiffs.a esp-idf/openssl/libopenssl.a esp-idf/fatfs/libfatfs.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a -L /home/ut1/esp/esp-at/components/esp-wolfssl/wolfssl/lib/esp32 -lwolfssl_nano esp-idf/cxx/libcx [... cut ...] eference to `esp_at_get_eth_default_config'
/home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ../components/at/lib/libesp32_at_core_silence.a(at_ethCmd.c.obj): in function `esp_at_eth_cmd_regist':
(.text.esp_at_eth_cmd_regist+0x5): undefined reference to `esp_at_get_eth_default_config'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
idf.py build ret: 2
idf.py build failed
```