espressif/esp-at #472: build.py is not building project

Piruthivi, 2020-09-18:
I have downloaded esp-at master. I already use esp-idf, which have paths defined in system variables.
Now am trying to build esp-at project by using **build.py** in windows command prompt, but it shows following error

_platform_name=ESP32,module_name=WROVER-32
old commit:5c0719aab633b69f1396230a1aa4c66987df1327

checkout commit:e7ac221b42517df8196c8b88cc9ac8dfb63db0f6
Please wait for the update to complete, which will take some time
fatal: reference is not a tree: e7ac221b42517df8196c8b88cc9ac8dfb63db0f6
M       build.py
Your branch is up to date with 'origin/master'.
Update completed
'export' is not recognized as an internal or external command,
operable program or batch file.
'cp' is not recognized as an internal or external command,
operable program or batch file._

i am not sure how to pass this error. need some help

work4embedded, 2020-10-02:
do you have any luck with this issue?
I am also facing same problem

ustccw, 2020-10-09:
Please follow the guide: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html#

xcguang, 2020-10-12:
Windows OS？

Piruthivi, 2020-10-12:
yes i am using Windows 10 OS

xcguang, 2020-10-12:
Please compile it on linux or the Windows Subsystem for Linux (WSL).

Piruthivi, 2020-10-12:
but it was working really fine in windows before without any error or flaw

felipepsilv, 2021-07-15:
Hi, I am with the same issue on Linux.

I was using a old version of esp at firmware. I've removed the old folder with rm -r /esp-at and cloned it again.

after that when I type ./build.py menuconfig I choose the module ESP32(1), module type WROOM-32(1) and silence mode off (0).

The error occur when it is cloning 'esp-idf'

Maybe someone forgot to push his commit?

Can someone help me with this?

[image]

ustccw, 2022-05-27:
@Piruthivi @felipepsilv @work4embedded esp-at v2.4 and master branch simplfied the compile workflow. just run `python build.py install` and `python build.py build` to build the esp-at project, could you please have a try?  
see this doc for more details:
https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html