espressif/esp-at #473: cannot enable flow control with AT+UART_{CUR,DEF}

niziak, 2020-10-01:
It is not possible to change flow control runtime using command: `AT+UART_CUR` nor `AT+UART_DEF`.
Flow control change works only when set by `CONFIG_AT_UART_DEFAULT_FLOW_CONTROL` or loaded from NVS during software startup.

Background:
 - IDF: `uart_set_hw_flow_ctrl()` is setting UART perihery flow control (but not configuring alternative pin function). I think this is a bug in [ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK)
 - IDF: `uart_param_config()` is configuring everything correctly.
 
During AT software startup UART configuration structure is initialized with compiled defaults. Then it tries to load UART settings from NVS and initializes UART correctly using `uart_param_config()` call.

During run time, when user is trying to enable flow control with command `AT+UART=115200,8,1,0,3`, function `at_setupCmdUart()` is changing uart paramaters by calling one by one:
```c
uart_wait_tx_done(esp_at_uart_port,portMAX_DELAY);
uart_set_baudrate(esp_at_uart_port,uart_config.baudrate);
uart_set_word_length(esp_at_uart_port,uart_config.data_bits);
uart_set_stop_bits(esp_at_uart_port,uart_config.stop_bits);
uart_set_parity(esp_at_uart_port,uart_config.parity);
uart_set_hw_flow_ctrl(esp_at_uart_port,uart_config.flow_control 120);
```
but no one from above calls sets PINs to alternate function.

ustccw, 2020-10-09:
@niziak In fact, esp-at try to load UART settings from factory parameter partition, rather than NVS. it is defined in `module_config/module_esp32_default/at_customize.csv`.

UART flow control pins defined in `components/customized_partitions/raw_data/factory_param/factory_param_data.csv`.

Please take a look at:
- [How to Set AT Port Pin](https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_set_AT_port_pin.html 
)
- [How To Create Factory Parameter Bin](https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_create_factory_parameter_bin.html)

niziak, 2020-10-09:
Super, I know that. You are proposing workaround what is unacceptable in some scenarios where main bug still exits.
Perhaps I don't want to enable flow control during compile time and lose connectivity where CTS line is not used. 
 want to enable HW flow control during device operation using  `AT+UART_CUR` command. **Now it is not possible.**

ustccw, 2020-10-09:
@niziak esp-at `CAN` enable or disable flow control in runtime by `AT+UART_CUR`.
esp-at `CAN NOT` change cts pin and rts pin in runtime. 

user can try to re-define `AT+UART_CUR` command to implement it. The source code is in main/interface/uart/at_uart_task.c

FayeY, 2021-03-22:
Closing as this question appears to be answered. Please feel free to reopen if this is still an issue. Thanks.

closed 2021-03-22 by FayeY

niziak, 2021-03-22:
Thanks to Open Source  code everything can be changed. The background of this issue was that it is a bug  because it not behave as documented. And that is all.

Maintraxx, 2021-06-08:
Thank you @niziak !
I've been struggling with this for a few days now, I keep losing data on the UART. In the meantime I suspect the hardware flow control. Your observation applies exactly to me. 
Due to compatibility with older firmware versions, I activate the FlowControl only after startup. This works fine with the NONOS version. Now this no longer works.
This is definitely a bug and I have no understanding at all why this bug was closed.

Translated with www.DeepL.com/Translator (free version)