espressif/esp-at #475: build.py is not building the project

work4embedded, 2020-10-02:
## Environment

- Development Kit:      [ESP-32S NodeMCU]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1.1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v2.1.0.0_esp32s2-84-g5ddd24f
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- Power Supply:         [US]

## Problem Description

I have cloned the git repository and tried to build it.
>build.py menuconfig
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP8266
3. PLATFORM_ESP32S2
choose(range[1,3]):1

Module name:
1. WROOM-32
2. WROVER-32
3. PICO-D4
4. SOLO-1
5. ESP32-D2WD (description: 2MB flash, No OTA)
choose(range[1,5]):1

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):0
platform_name=ESP32,module_name=WROOM-32
'"cd esp-idf; git rev-parse HEAD"' is not recognized as an internal or external command,
operable program or batch file.
Traceback (most recent call last):
  File "D:\Work\Projects\dummy\esp-at\build.py", line 349, in <module>
    main()
  File "D:\Work\Projects\dummy\esp-at\build.py", line 343, in main
    auto_update_idf(platform_name, module_name)
  File "D:\Work\Projects\dummy\esp-at\build.py", line 129, in auto_update_idf
    rev_parse_head = subprocess.check_output(['cd esp-idf; git rev-parse HEAD'], shell = True).decode(encoding="utf-8").strip()
  File "C:\Python38\lib\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Python38\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['cd esp-idf; git rev-parse HEAD']' returned non-zero exit status 1.

i am not sure how to pass this error. need some help

borrosjim, 2020-10-03:
Hi, the problem here is that the script is using unix like commands so in order to make it work on windows you'll have to adapt the commands in build.py. My approach was to install WSL with ubuntu for windows 10 and after installing all the dependencies it worked (you'need python, pip, flex, curss, bison, gperf...). 
Bear in mind that if you are using an evaluation board like ESP32_Devkit_C which has the CP2102 as a serial interface, WSL 2 won't work because of compatibility issues so I recommend you to use WSL1 instead. 
I hope this helps,
Borja

work4embedded, 2020-10-04:
Hello Borja,

Thanks for your feedback.
Do you have any user manual or document to install all the dependencies?
Since i am new to this environment it would be very much helpful.

Thanks in advance.

borrosjim, 2020-10-05:
Once you have WSL installed and running you have to set up ESP-IDF toolchain 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#install-prerequisites
And after that just navigate to esp-idf folder and execute install.sh and export.sh.
Once you,ve done with all of this you are ready to go.
Borja

closed 2021-04-14 by FayeY