espressif/esp-at #476: [ESP32 & ESP8266] Uploading SSL certificates on the fly.

LaBus3, 2020-10-05:
Hello,
I used AT+SYSFLASH to upload bins of ca, client certificate and key (generated by AtPKI.py) to their respective partitions over the factory bin provided in the release.
Still the command AT+CIPSSLCCONF=2,0,0 gives NO CA FOUND error. Am I missing something? The certs and keys that I use are standard pem encoded ones that I test with other ME modules.

Upload commands I used : 
AT+SYSFLASH=1,"client_cert",0,1040
AT+SYSFLASH=2,"client_key",0,1744
AT+SYSFLASH=2,"client_ca",0,1040

NB: Is it possible to purchase WROOM modules with AT firmware built in? If so what firmware will it contain, the one in the releases section here?

ustccw, 2020-10-09:
@LaBus3 It looks like esp-at can not find the `client_ca` partition. 
please query the partition by `AT+SYSFLASH?`

I am not sure the esp-at firmware version about `the factory bin provided in the release`.

About `purchase WROOM modules  with AT firmware built in`, as far as i know, it is.
I suggest you confirm with the sales again.

LaBus3, 2020-10-10:
> @LaBus3 It looks like esp-at can not find the `client_ca` partition.
> please query the partition by `AT+SYSFLASH?`
>
> I am not sure the esp-at firmware version about `the factory bin provided in the [...]

Hi, thank you for the reply.
This is the output of AT+SYSFLASH? after uploading using the previously mentioned commands.
```+SYSFLASH:"factory_param",64,8,0xf1000,0x1000
+SYSFLASH:"server_cert",64,2,0xf2000,0x2000
+SYSFLASH:"server_key",64,3,0xf4000,0x2000
+SYSFLASH:"server_ca",64,4,0xf6000,0x2000
+SYSFLASH:"client_cert",64,5,0xf8000,0x2000
+SYSFLASH:"client_key",64,6,0xfa000,0x2000
+SYSFLASH:"client_ca",64,7,0xfc000,0x2000
+SYSFLASH:"wpa2_cert",64,8,0xfe000,0x2000
+SYSFLASH:"wpa2_key",64,9,0x100000,0x2000
+SYSFLASH:"wpa2_ca",64,10,0x102000,0x2000
+SYSFLASH:"mqtt_cert",64,11,0x104000,0x2000
+SYSFLASH:"mqtt_key",64,12,0x106000,0x2000
+SYSFLASH:"mqtt_ca",64,13,0x108000,0x2000

OK
```

Also the factory image i mentioned is the file under ESP8266-IDF-AT_V2.1.0.0.zip\ESP8266-IDF-AT_V2.1.0.0\ESP8266-IDF-AT_V2.1\factory\factory_WROOM-02.bin (ESP8266)
&
ESP32-WROOM-32_AT_Bin_V2.1.0.0.zip\ESP32-WROOM-32_AT_Bin_V2.1.0.0\ESP32-WROOM-32_AT_Bin_V2.1\factory\factory_WROOM-32.bin (ESP32)

ustccw, 2020-10-12:
1. please erase the partition data before write.

2. please provide the complete log, AT command, input certificate file.

LaBus3, 2020-10-12:
> 
> 
>     1. please erase the partition data before write.
> 
>     2. please provide the complete log, AT command, input certificate file.

Log : https://gist.github.com/LaBus3/7cb7d5a77095d651ade5ef0eef546b07
there are some complications in uploading the certificates, ill do so if its mandatory.

BertLammers, 2020-12-05:
@LaBus3 : Did you manage to upload and use the certificates?

ustccw, 2020-12-14:
@LaBus3 I am afraid can not reproduce this issue.  
Could you please test with the following firmware which added some log. Please provide the UART0 log and UART1 log if possible.

[target.zip](https://github.com/espressif/esp-at/files/5687999/target.zip)

At the same time, could you please provide the `client.crt`, `client.key`, and `server.crt` you mentioned before.

LaBus3, 2020-12-19:
> @LaBus3 I am afraid can not reproduce this issue.
> Could you please test with the following firmware which added some log. Please provide the UART0 log and UART1 log if possible. [...]

Sorry for the delayed response. I will reply with the result as soon as possible.

ustccw, 2021-01-12:
@LaBus3 Just a kindly remind: have you tested this issue recently.

closed 2021-03-22 by FayeY