espressif/esp-at #477: AT+BTA2DPINIT command for setting the role

MGA97, 2020-10-06:
Hi,
I have a problem with the AT+BTA2DPINIT command, I am trying to init the a2dp profile in this way:
`AT+BTA2DPINIT=1,1` (1=role of sink, 1=init the profile) but this sends me an ERROR
I tried with `AT+BTA2DPINIT=1` and this is ok, it sends me an OK and if I query `AT+BTA2DPINIT?` it sends me:

`+BTA2DPINIT:1`

`OK`

but how can I set the role?

thank you.

MGA97, 2020-10-07:
I tried with `AT+BTA2DPINIT=2 ` and it is OK, it connects and sets the role of sink.
why is this happening? 

thanks.

xcguang, 2020-10-12:
@xiewenxiang Please help to confirm this.

Luxin812, 2022-04-19:
@MGA97 There are some changes to the AT+BTA2DPINIT command. If you are interested, you can read the relevant documentation for this command (https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/AT_Command_Set/BT_AT_Commands.html#at-bta2dpinit-classic-bluetooth-a2dp）.