espressif/esp-at #479: [ESP8266] Command "AT+CWJAP?" only response OK

leungngok, 2020-10-10:
I use "AT+CWJAP?" to query WiFi status of ESP8266 (has connected to WiFi), but it only responses OK without any information.

The firmware is V2.0.0.1_esp8266

ustccw, 2020-10-10:
@leungngok Please try with the latest version, it should get fixed.

leungngok, 2020-10-10:
> @leungngok Please try with the latest version, it should get fixed.

Sorry, the firmware is V2.1.0.0_esp8266

ustccw, 2020-10-10:
@leungngok I tested with the firmware here https://docs.espressif.com/projects/esp-at/en/latest/AT_Binary_Lists/ESP8266_AT_binaries.html
It seems to be working well.
```
ready
AT+CWMODE=1

OK
AT+CWJAP="BL_841R","1234567890"
WIFI CONNECTED
WIFI GOT IP

OK
AT+CWJAP?
+CWJAP:"BL_841R","44:33:4c:90:96:9a",11,-55,0,0,0,0

OK
AT+GMR
AT version:2.1.0.0(ded5b14 - Jul 24 2020 12:01:05)
SDK version:v3.3-2-gb6e861a
compile time(36bb200):Jul 28 2020 08:30:34
Bin version:2.1.0(WROOM-02)

OK
```

leungngok, 2020-10-10:
@ustccw The UART console output shows it is connected, but responses NO AP after "AT+CWJAP?"

```
wifi:connected with HUAWEI_316, aid = 49, channel 10, HT20, bssid = 5c:78:f8:96:80:53
WIFI CONNECTED
tcpip_adapter: sta ip: 192.168.8.112, mask: 255.255.255.0, gw: 192.168.8.1

AT+CWJAP?
No AP

OK
AT+GMR
AT version:2.1.1.0-dev(24b71e4 - Aug 25 2020 08:41:31)
SDK version:v3.2-611-g741bbfad
compile time(741bbfad):Oct 10 2020 18:23:14

OK
```

jandrassy, 2020-10-10:
@leungngok, you use static IP config?

leungngok, 2020-10-10:
> @leungngok, you use static IP config?

No, connect router by command AT+CWJAP="ssid","pwd"

jandrassy, 2020-10-10:
> > @leungngok, you use static IP config?
> 
> No, connect router by command AT+CWJAP="ssid","pwd"

you don't have "WIFI GOT IP" in output

ustccw, 2020-10-12:
@leungngok It looks like a DHCP issue. Could you please try the demo https://github.com/espressif/ESP8266_RTOS_SDK/tree/release/v3.3/examples/wifi/simple_wifi

leungngok, 2020-10-12:
@ustccw 

At the beginning, I did not find the esp-at release for ESP8266, so I moved the AT component into ESP8266_RTOS_SDK project, which caused something wrong.

Today I directly compile the esp-at ESP8266 release, it can output "WIFI GOT IP" and "AT+CWJAP?" works well.

Thank you for your support!

closed 2020-10-12 by leungngok