espressif/esp-at #481: AT+CIPSEND, SEND OK sometimes received after endpoint response

mjessome, 2020-10-16:
## Environment

- Development Kit: ESP-WROVER-KIT
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used: ESP32-WROOM-32
- IDF version: v4.0.1-193-ge7ac221b4
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: macOS
- Power Supply: USB

esp-at commit `4335e4cbc7b2b122a493495f80a466babb4fed24`
Default sdkconfig with log verbosity set to `info`.

## Problem Description

I have a very simple HTTP client that sends data over UART using TCP/IP commands (CIPSTART/CIPSEND) using SSL.
Sometimes when I send `AT+CIPSEND=<length>`, I receive a response through `+IPD,<len>:` from the server before the esp-at firmware has returned `SEND OK`.

### Expected Behavior
If there is a CIPSEND in progress, any incoming data over that connection should be returned _after_ the `SEND OK` response.

### Actual Behavior
I often receive `+IPD` response data before `SEND OK` has been received.

### Steps to reproduce
This reproduces maybe 20% of the time when I use the script provided below.

1. Make a connection to a server with `AT+CIPSTART=.....`
2. Send some data down, which expects a response (in this case, a GET request)

### Code to reproduce this issue
Here is a simple python script that I use to demonstrate the issue, it is hard-coded with an image URL that I have seen the problem with a lot.

https://gist.github.com/mjessome/8f60059405944d4c5078dc3522a01581

To run:
```
python3 OoO-Demo.py <serial port> <ssid> <pass>
```

## Debug Logs

Logs from python script:
```
---> Draining output
> AT+CWMODE=1

< AT+CWMODE=1
< 
< OK
--> Connect to WIFI
> AT+CWJAP="xxxxx","yyyyy"

< AT+CWJAP="xxxxx","yyyyy"
< WIFI DISCONNECT
< CLOSED
< WIFI CONNECTED
< WIFI GOT IP
--> Disable SSL
> AT+CIPSSLCCONF=0

< 
< OK
--> Open Connection
> AT+CIPSTART="SSL","pyxis.nymag.com",443

< AT+CIPSSLCCONF=0
< 
< OK
< AT+CIPSTART="SSL","pyxis.nymag.com",443
< CONNECT
--> Send
> AT+CIPSEND=156

< 
< OK
--> Wait For >
---> Send Header
> GET /v1/imgs/2f3/e6b/4f8d37ff4555c6aedae051ff3080cf9b9f-05-animations-01.2x.rhorizontal.w1100.jpg HTTP/1.1
Host: pyxis.nymag.com
User-Agent: Custom-UA

< 
< Recv 156 bytes
< 
< +IPD,283:HTTP/1.1 200 OK
< Connection: keep-alive
< Content-Length: 340694
< Last-Modified: Mon, 05 Oct 2020 16:02:35 GMT
< ETag: "7e412d8b8e3bb8745c846275b923a462"
< Content-Type: image/jpeg
< Cache-Control: max-age=86400
< Accept-Ranges: bytes
< Date: Fri, 16 Oct 2020 15:54:53 GMT
< Age: 112892
< 
< 
< 
< SEND OK
```

Notice how the `SEND OK` is received _after_ the +IPD response from the server

## Other items if possible
- SDKConfig is default with logs increased, but can be found here https://gist.github.com/mjessome/8f60059405944d4c5078dc3522a01581

ustccw, 2021-01-12:
@mjessome sorry for the late reply.  
i am afraid esp-at maybe will not be designed in this way, for `SEND OK` is `TCP TX` and `+IPD` is `TCP RX`.  
they are parallel, otherwise, the throughput of esp-at will cut down a half.

emm, if you insist on doing this, it is suggested to pre-process the data on uart tx and uart rx.

uart rx: https://github.com/espressif/esp-at/blob/master/main/interface/uart/at_uart_task.c#L117
uart tx: https://github.com/espressif/esp-at/blob/master/main/interface/uart/at_uart_task.c#L109

if something not clear, please let me know.

FayeY, 2021-03-22:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2021-03-22 by FayeY