espressif/esp-at #483: Build failed when enabled ethernet

jitenshap, 2020-10-23:
I'm trying to build a custom AT command firmware to enable Ethernet.
I set it like this in menuconfig.
Component Config->AT->AT Ethernet Support [*]
If you do not check AT ethernet support, the build will succeed, but if you check it, the build will not be possible.

esp-at repository version:
~/esp-at$ git log -n 1 --format=%H
a1f3fc0c45d3935e99e52c44bd98b2eb5bd63c95

Build logs:
module_name WROVER-32
platform_name=ESP32,module_name=WROVER-32
Checking Python dependencies...
Python requirements from /home/ubuntu/esp-at/esp-idf/requirements.txt are satisfied.

...

[100%] Linking CXX executable esp-at.elf
/home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ../components/at/lib/libesp32_at_core_silence.a(at_ethCmd.c.obj):(.literal.esp_at_eth_cmd_regist+0xc): undefined reference to `esp_at_get_eth_default_config'
/home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ../components/at/lib/libesp32_at_core_silence.a(at_ethCmd.c.obj): in function `esp_at_eth_cmd_regist':
(.text.esp_at_eth_cmd_regist+0x5): undefined reference to `esp_at_get_eth_default_config'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/esp-at.elf.dir/build.make:269: esp-at.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1779: CMakeFiles/esp-at.elf.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2

stokersoft, 2020-11-06:
Hi,

Just went through this - it was a little tough to get this compiled, but i think you can manage it by doing the following:
Make sure these are set in your environment:
ESP_AT_PROJECT_PLATFORM=PLATFORM_ESP32
ESP_AT_MODULE_NAME = Wrover-32
PLATFORM_NAME = PLATFORM_ESP32
And do a "make clean" and then build again with make

jitenshap, 2020-11-13:
@stokersoft 
>Make sure these are set in your environment:
Do you mean?

jitenshap, 2020-11-16:
@songjiaxiu-espressif 
Can you clone the latest repository and enabled Ethernet for WROVER?
Is it possible to fix this issue?

ESP-Coco, 2020-11-16:
@jitenshap Hi, I have tested the latest master branch and it's no problem.

jitenshap, 2020-11-17:
@ESP-Coco 
Are you using build.py?
Or are you using other make commands or idf.py to build?

ESP-Coco, 2020-11-17:
@jitenshap Sorry, you are right. There are some problems with Cmake, and GNU make is normal. We will fix soon,

katodo, 2020-12-16:
I'm in same @jitenshap situation, see my other post about my issue #471 ( I could  close #471 )

This is a "summary" of all test I repeated today... all procedures are done in a clean Ubuntu system:
- I complete removed any esp-at and esp-idf file in my system.
- I downloaded a fresh git with git clone --recursive https://github.com/espressif/esp-at.git
- I changed the second line of _components/customized_partitions/raw_data/factory_param/factory_param_data.csv_ to:
  `PLATFORM_ESP32,WROOM-32,,0xfcfc,2,1,78,1,1,13,CN,115200,1,3,-1,-1,-1,-1`

esp-idf is automatically downloaded with "./build.py menuconfig" command, I don't choose any commit.

I choose this option at first execution:
```
Platform name: PLATFORM_ESP32
Module name: WROOM-32
Enable silence mode to remove some logs and reduce the firmware size? Yes
```
During menuconfig I generated attached sdkconfig file:
[sdkconfig_20201216_1527.zip](https://github.com/espressif/esp-at/files/5703754/sdkconfig_20201216_1527.zip)

During build I retrieve this error:

```
ld -T esp32.rom.newlib-funcs.ld -T esp32.rom.newlib-nano.ld -Wl,--gc-sections ../esp-idf/components/xtensa/esp32/libhal.a -L /home/ut1/esp/esp-at/build/esp-idf/esp32 -T esp32_out.ld -u app_main -L /home/ut1/esp/esp-at/build/esp-idf/esp32/ld -T esp32.project.ld -L /home/ut1/esp/esp-at/esp-idf/components/esp32/ld -T esp32.peripherals.ld -u call_user_start_cpu0 -u ld_include_panic_highint_hdl -u esp_app_desc -u vfs_include_syscalls_impl -L /home/ut1/esp/esp-at/esp-idf/components/esp_wifi/lib_esp32 -lgcov -lc_nano -lm -lgcc -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl && :
/home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ../components/at/lib/libesp32_at_core.a(at_ethCmd.c.obj):(.literal.esp_at_eth_cmd_regist+0xc): undefined reference to `esp_at_get_eth_default_config'
/home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ../components/at/lib/libesp32_at_core.a(at_ethCmd.c.obj): in function `esp_at_eth_cmd_regist':
(.text.esp_at_eth_cmd_regist+0x5): undefined reference to `esp_at_get_eth_default_config'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
idf.py build ret: 2
idf.py build failed
```

### xtensa-esp32-elf-gcc --version --verbose output is:
```
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/ut1/esp/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2020r1' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2020r1) 
COLLECT_GCC_OPTIONS='--version' '-v'
 /home/ut1/esp/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/cc1 -quiet -v -iprefix /home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/ help-dummy -quiet -dumpbase help-dummy -auxbase help-dummy -version --version -o /tmp/cc1dZcfH.s
GNU C17 (crosstool-NG esp-2020r1) version 8.2.0 (xtensa-esp32-elf)
	compiled by GNU C version 4.9.2, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version isl-0.19-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
COLLECT_GCC_OPTIONS='--version' '-v'
 /home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/as --traditional-format --version -o /tmp/ccuaYud3.o /tmp/cc1dZcfH.s
GNU assembler (crosstool-NG esp-2020r1) 2.31.1
Copyright (C) 2018 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `xtensa-esp32-elf'.
COMPILER_PATH=/home/ut1/esp/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/:/home/ut1/esp/xtensa-esp32-elf/bin/../libexec/gcc/:/home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/
[... cut ...]  -lc -lnosys -lc -lgcc /home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/crtend.o /home/ut1/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/crtn.o
GNU ld (crosstool-NG esp-2020r1) 2.31.1
Copyright (C) 2018 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
COLLECT_GCC_OPTIONS='--version' '-v'
```

katodo, 2020-12-17:
I switched to xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz
 keept from https://github.com/espressif/crosstool-NG/releases/tag/esp-2019r2 

but, enabling AT Ethernet support "./build.py build" fail with .text.esp_at_eth_cmd_regist+0x5): undefined reference to `esp_at_get_eth_default_config'

ustccw, 2020-12-18:
@katodo @jitenshap Here is a workaround for AT ethernet compile.
Please modify: https://github.com/espressif/esp-at/blob/master/components/at/CMakeLists.txt#L3
to 
```
    bootloader_support app_update openssl wpa_supplicant spi_flash main)
```

That is, appending the `libmain.a` to at core require_components, so that `libesp*at.a` can link to implement of `esp_at_get_eth_default_config`, which defined in libmain.a.

At the same time, `esp-at` will simply and adjust the directory structure to facilitate reasonable code management.

katodo, 2020-12-18:
Ok, now build end succesfully! Now I need to do some test on my PHY :)

katodo, 2020-12-18:
I just tried the new firmware but after booting I have these messages:

`max tx power=78,ret=0

[0;31mE (1188) lan8720: lan8720_pwrctl(332): power up failed[0m
[0;31mE (1188) lan8720: lan8720_init(368): power control failed[0m
[0;31mE (1189) esp_eth: esp_eth_driver_install(197): init phy failed[0m
regist ethernet cmd fail`

no response to AT command after cmd fail messages.

iw2lsi, 2020-12-18:
Hi,

   see https://github.com/espressif/esp-idf/issues/5425

  you have to review the lan8720_pwrctl()

Giampaolo

katodo, 2020-12-18:
I don't understand, after a clean and rebuild no more lan8720 related messages in logs... I'm confused...

katodo, 2020-12-21:
ok, problem was in esp-at\esp-idf\components\esp_eth\include\esp_eth_phy.h

I neet to change .phy_addr from 1 to 0 in #define ETH_PHY_DEFAULT_CONFIG()...

ustccw, 2021-01-11:
@katodo @jitenshap esp-at had updated the code structure, and fixed the ethernet build failed issue.

please `git pull` the latest code, and test again.

jitenshap, 2021-01-13:
@ustccw 
It works!
Thank you.

closed 2021-01-13 by jitenshap