espressif/esp-at #484: Read characteristic from server side

HectorJimenezMendez, 2020-10-26:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
     v4.0.1-193-ge7ac221b4
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

Is there any AT Command which lets the server read the value of a certain characteristic?

It's needed for our project to be able to read a characteristic value from server side. We need to know from server side the value of a characteristic which is defined in build time and it's not gonna be changed until ESP32 module FW is updated.

### Expected Behavior
AT command for reading characteristic value from server side

### Actual Behavior
No AT command for this purpose

xiewenxiang, 2020-11-10:
This feature is not supported at the moment.

HectorJimenezMendez, 2020-11-11:
Thank you, @xiewenxiang .

Is there any plan of including this feature in future versions of esp-at?

Kind regards!

xcguang, 2020-11-11:
If everything goes well, it is planned to be completed by the end of next month.

HectorJimenezMendez, 2020-11-11:
Thank you, @xcguang .

We will consider whether waiting for that version or trying another method to read that value instead.

Kind regards.

HectorJimenezMendez, 2021-01-12:
Hi, @xcguang.

When is it planned next esp-at version to be released?
Is it going to include any AT command to read a characteristic value from server side?

Thank you!

xcguang, 2021-01-13:
@HectorJimenezMendez We are doing with it. And it will be at GitHub next week.

ESP-AT release is about in 2020.04.

HectorJimenezMendez, 2021-01-13:
thank you, @xcguang

Luxin812, 2021-09-14:
@HectorJimenezMendez Thank you for using ESP-AT. At present, ESP-AT master has implemented AT+BLEGATTCRD command on the server side and AT+BLEGATTSSETATTR command on the client side. If you still need these functions, please test and develop based on ESP-AT master