espressif/esp-at #485: NOTIFY not in HEX and cannot be stopped

Bmooij, 2020-10-27:
## Environment

- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): 
   v4.0.1-193-ge7ac221b4
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    Windows
- Power Supply:         USB external 5V

## Problem Description

The command `AT+BLEGATTCWR=0,4,2,1,2` starts notifying, but the values are not in HEX. And cannot stop the notified messages.

### Expected Behavior

I would expect `+NOTIFY:0,4,2,20,<HEX>`

### Actual Behavior

Actual Behavior  `+NOTIFY:0,4,2,20,<RAW Bytes>`

### Steps to repropduce

1. `AT+BLEINIT=1`
2. `AT+BLECONN=0,"<remote_address>"`
3. `AT+BLEGATTCWR=0,4,2,1,2`

// It helps if you attach a picture of your setup/wiring here.
Bluetooth Characteristics for 0,4,2
+BLEGATTCCHAR:"char",0,4,2,0x<UUID>,0x34
+BLEGATTCCHAR:"desc",0,4,2,1,0x2902

Bmooij, 2020-10-27:
Writing `{0x01, 0x00}` should start the notifieer and writing `{0x00, 0x00}` should stop the notifieer

peterantypas, 2021-04-19:
Is this still open? Does anyone know if it affects the ESP32-WROOM as well?