espressif/esp-at #486: ova image hash

valoni, 2020-11-01:
i tried to download OVA image but since i don't did i download well or not
we need a little help 

http://download.espressif.com/esp_share/env/ubuntu1804.ova
- size of image and hash of image (to be sure we downloaded properly)

example 
Size : 5.27 GB
HASH : MD5 - xxxxxxx
            SHA1 - xxxxxxx
            SHA256 - xxxxxx

to help us to be sure we downloaded properly file

ustccw, 2021-01-12:
@valoni sorry for late reply.
Here is a list of `md5sum`:
- [ubuntu1804.ova](http://download.espressif.com/esp_share/env/ubuntu1804.ova): `4847e100346df2c21a478925e0933617`
- [virtualbox-5.2.16.exe](http://download.espressif.com/esp_share/env/virtualbox-5.2.16.exe):`b21175b80bbbaed2e72ac62807b05b8f`
- [virtual-box-plug](http://download.espressif.com/esp_share/env/virtual-box-plug):`76ccda092e0e2f1775a4050f1ece3db7`

valoni, 2021-01-12:
thanks for info and just downloaded : 
----------------------------------------------------------------------------
Test of : 12.01.2021
----------------------------------------------------------------------------
Here is a list of md5sum:
----------------------------------------------------------------------------
2020.03.12 09:36 - created image info
----------------------------------------------------------------------------
http://download.espressif.com/esp_share/env/ubuntu1804.ova (5.32 GB)
ubuntu1804.ova: 4847e100346df2c21a478925e0933617 (OK)
----------------------------------------------------------------------------
2019.12.18 03:24  - virtual box info
----------------------------------------------------------------------------
http://download.espressif.com/esp_share/env/virtualbox-5.2.16.exe (108 MB)
virtualbox-5.2.16.exe:b21175b80bbbaed2e72ac62807b05b8f (OK)
----------------------------------------------------------------------------
2019.12.18 03:23  - virtual box plugins
----------------------------------------------------------------------------
http://download.espressif.com/esp_share/env/virtual-box-plug.tar (52.8 MB)
----------------------------------------------------------------------------
it show to me as MD5 -> AAC23AAF9DDDE9FAE689F90BD4FEFDB9 is not 76ccda092e0e2f1775a4050f1ece3db7

[image]

ustccw, 2021-01-13:
@valoni Hi, i checked it again, it looks nothing different. Could you please check it again?

[image]

valoni, 2021-01-13:
found issues of this file

if you download with wget it download as you said but if you (without .tar extension throught wget) and if you download from browser it download something another things  (.tar extension)  ..

wget is MD5 sum as you described (18.6 MB)
browser (.tar file) is as it showed to me from browser (52.8 MB) have other MD5 hash

thanks for clarify it.

closed 2021-01-13 by valoni

ustccw, 2021-01-14:
just a little curious, i can not get any resources if appended the `.tar` to the end of URL.
```c
chenwu@espressif:$ wget http://download.espressif.com/esp_share/env/virtual-box-plug.tar
--2021-01-14 10:34:28--  http://download.espressif.com/esp_share/env/virtual-box-plug.tar
Resolving download.espressif.com (download.espressif.com)... 119.9.116.168
Connecting to download.espressif.com (download.espressif.com)|119.9.116.168|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-01-14 10:34:28 ERROR 404: Not Found.
```

valoni, 2021-01-14:
> just a little curious, i can not get any resources if appended the `.tar` to the end of URL (same situation to my chrome browser).
> 
> ```c
> chenwu@espressif:$ wget http://download.espressif.com/esp_share/env/virtual-box-plug.tar
> --2021-01-14 10:34:28--  http://download.espressif.com/esp_share/env/virtual-box-plug.tar
> Resolving download.espressif.com (download.espressif.com)... 119.9.116.168
> Connecting to download.espressif.com (download.espressif.com)|119.9.116.168|:80... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2021-01-14 10:34:28 ERROR 404: Not Found.
> ```

no no you don't need to create .TAR extension,
it created by iteself when on browser wh you tried to open link

http://download.espressif.com/esp_share/env/virtual-box-plug

ustccw, 2021-01-14:
much clear, thanks for your explanation!