espressif/esp-at #488: ESP-32 (Wrover 32) and ethernet. Incoming tcp packets are corrupted, the string "deleted" is injected in the IPD response.

stokersoft, 2020-11-06:
Hi,

We send http requests, that we craft ourselves via TCP over Ethernet cable to an external server.

See the log from starup of module here.
[startuplog.txt](https://github.com/espressif/esp-at/files/5501524/startuplog.txt)

## Environment

- Branch: 2.1.0.0 latest code (also tried current latest master branch, same problem) in https://github.com/espressif/esp-at/
- Development Kit:      ESP32-Wrover-Kit
- Kit version v2
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.0.1-193-ge7ac221b4
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0:
- Operating System:     Linux, Ubuntu 19
- Power Supply:         USB
- Other: Ethernet, ESP32-Ethernet-Kit-A-V12, with ESP32-WROVER-E module
- Uart for AT commands: (set through factory_param_data.csv)
RX = IO4
TX = IO2

## Problem Description

With AT commands over the uart connected to pin 2 and 4 we send commands to connect and send a http request to a remote server, but get the the string "deleted" injected into the IPD response. The received text is extended with the lenght of deleted, so the actual text is IPD reported byte count + strlen("deleted") - so no response text is overwritten, but the byte count is not updated to reflect that "deleted" is inserted. The string "deleted" comes at random places in the http response. 

### Expected Behavior
Expected IPD with the correct byte count and actual data sent from the server. We verified that the data sent from the server does not contain "deleted", also the fact that IPD byte count is wrong implies that this corruption happens inside the module code.

### Actual Behavior
Get a IPD response with data with string "deleted" inserted into the text and byt count from IPD is not reflected.

### Steps to reproduce
AT+CIPSTART=1,"TCP","up.stokercloud.dk",80
AT+CIPSEND=1,51
Then sending 51 bytes... http header

Then we get this response:
"Recv 51 bytes

SEND OK

+IPD,1,172,"46.4.92.181",80:HTTP/1.1 200 OK
Server: nginx
Date: Fri, 06 Nov 2020 10:56:57 GMT
Content-Type: text/html;charsdelete
et=UTF-8
Transfer-Encoding: chunked
Connection: close

5
a=58

0

1,CLOSED"

See how text "delete" is inserted in the middle of the string charset - nothing is overwritten but its appended...

### Code to reproduce this issue

No code - just AT commands sent to the module.

## Debug Logs

See above.

## Other items if possible

stokersoft, 2020-11-09:
It would be nice if some developer can comment on this.

But we found a workaround, here are a few points
- Moved the uart pins for AT commands from 2 and 4 to 32 and 33
- Now string "delete" is not appearing on the AT command uart
- It still comes on the COM4 uart (thrugh the usb device) and thats fine
- It only comes on connection:close responses, where the server closes the connection

I found this in esp-at.bin
udp create %d socket fail
 delete
 %d,CLOSED

The text "delete" thats the one.

Can anyone confirm if this is a unintenional debug message placed by accident by some developer with access to the core files?

Problem is solved from our side, we can continue, but i think you should look into this.

Regards Jens

vcmania, 2020-12-03:
I used ESP8266(ESP-12).
Me too same problem.
I hope patch this problem.

ustccw, 2021-01-12:
@stokersoft @vcmania thanks for your observation.  
By default, UART0 and UART1 will [use different GPIO](https://docs.espressif.com/projects/esp-at/en/latest/Get_Started/Hardware_connection.html). `+IPD` print and `delete` print are from different UART port output.  
that is to say, they are independent and will be no a data injected by peer.

Now, if config UART0 and UART1 to the same GPIO, it is possible to have a data injected by each other, for the state and buffer of the opposite end will not be known between two UARTs.

emm, esp-at will try the best to make two UARTs work on same GPIO without the data injected.

vcmania, 2021-01-15:
@ustccw 
my project must use UART0.
version 2.0 is not have a problem.
message remove in libespXXXX_at_core.a file.

xcguang, 2021-01-18:
Maybe the old version has no these log. you can `./build.py SILENCE=1 build` to disable log

vcmania, 2021-01-22:
@xcguang 
I make use SILENCE but recieved 'delete'.

My environment.
fireman@PC:~/esp/v2.1.0.0_esp8266$ git describe --tags
v2.1.0.0_esp8266-7-g3355cd8

fireman@PC:~/esp/v2.1.0.0_esp8266$ git branch
* release/v2.1.0.0_esp8266

fireman@PC:~/esp/v2.1.0.0_esp8266$ xtensa-lx106-elf-gcc --version
xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-100-ge567ec7) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

eps-at.map into used 'libesp8266_at_core_silence.a'
fireman@PC:~/esp/v2.1.0.0_esp8266$ cat build/esp-at.map | grep -m10 libesp8266_at_core
/home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_custom_partitions.c.obj)
/home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_wifiCmd.c.obj)
/home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_baseCmd.c.obj)
                              /home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_wifiCmd.c.obj) (at_sys_store_mode_is_flash)
/home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_nvm.c.obj)
                              /home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_wifiCmd.c.obj) (at_nvm_ap_type_config_get)
/home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_wpsCmd.c.obj)
/home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_mqtt_cmd.c.obj)
/home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_smartconfigCmd.c.obj)
/home/fireman/esp/v2.1.0.0_esp8266/components/at/lib/libesp8266_at_core_silence.a(at_ipCmd.c.obj)

vcmania, 2021-03-08:
my solution

make menuconfig
> Component config > ESP8266-specific > UART for console output > None

then disabled uart_putc() in esp-idf/components/esp8266/source/ets_printf.c

FayeY, 2021-03-22:
Closing as this question appears to be answered. If the problem is still unsolved, please feel free to reopen it. Thanks.

closed 2021-03-22 by FayeY