espressif/esp-at #489: Build issue for AT through UART

kkrischker, 2020-11-12:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- ESP-AT version: v2.1.0.0_esp32s2-122-g0c15ea3
- Build System:         CMake
- Compiler version : 8.4.0
- Operating System:     Linux
- Power Supply:        USB

## Problem Description

Hello

If I build the standard configuration (AT through UART) I run into compile errors in the file "main/interface/hspi/at_hspi_task_esp32.c":

In file included from ../esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:17,
                 from ../esp-idf/components/soc/include/soc/gpio_periph.h:18,
                 from ../esp-idf/components/driver/include/driver/gpio.h:23,
                 from ../main/interface/hspi/at_hspi_task_esp32.c:37:
../main/interface/hspi/at_hspi_task_esp32.c: In function 'pull_high_cb':
../main/interface/hspi/at_hspi_task_esp32.c:44:38: error: 'CONFIG_AT_HSPI_HANDSHAKE_PIN' undeclared (first use in this function); did you mean 'CONFIG_PTHREAD_STACK_MIN'?
 #define SPI_SLAVE_HANDSHARK_GPIO     CONFIG_AT_HSPI_HANDSHAKE_PIN
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
 and so on.

After looking in the code of "at_hspi_task_esp32.c", I could work around the issue by adding
"#ifdef CONFIG_AT_BASE_ON_HSPI" in line 37 and a "#endif" at the end of the file.

ESP-Coco, 2020-11-13:
@kkrischker Thanks. This is indeed a problem introduced by our latest commit, and we will fix this problem as soon as possible.

ustccw, 2021-01-12:
@kkrischker Sorry for the late reply. esp-at had fixed it one month ago, please try with the latest version.

closed 2021-04-14 by FayeY